mongoose-os-apps/shelly-homekit #186: Shelly HT / Temperature AddOn

schemhad, 2020-10-12:
It would be very nice if we could support Shelly HT. Maybe this could be similar to the Power Measuring?

andyblac, 2020-10-12:
> It would be very nice if we could support Shelly HT. Maybe this could be similar to the Power Measuring?

is HT bluetooth ?

schemhad, 2020-10-12:
It's an Shelly1 with Humidity and Temperature Sensor. 

https://shelly.cloud/knowledge-base/devices/shelly-ht/

The Sensor could also bought seperate. 

https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312

andyblac, 2020-10-12:
ok, wifi, I can't see that battery lasting a week on HomeKit, as it will drain battery as it keep checking connectivity.

schemhad, 2020-10-12:
But the AddOn Module could run on every Shelly 1 & PM or the retail H&T Module on USB.

andyblac, 2020-10-12:
> But the AddOn Module could run on every Shelly 1 & PM or the retail H&T Module on USB.

ye I was talking about H&T on battery.

rojer, 2020-10-12:
yes, this can be done

rojer, 2020-10-12:
HAP has accessories for temperature and humidity sensors, shouldn't be hard to add.
ideally, we'd auto-detect the sensor on startup so no UI changes would be necessary.

MilanBacaCz, 2020-10-12:
is there any documentation how to auto detect the temperature and other sensors?

Ludelag, 2020-10-14:
> yes, this can be done

Dear @rojer does it means will be done it?

rojer, 2020-10-14:
eventually yes

Ludelag, 2020-10-14:
> eventually yes

that will be nice

rojer, 2020-10-14:
looked into this. addon provides a 1-wire bus with output on gpio0 and input on gpio3 (uart0 rx).
output is connected to input (via isolator).
so this can be used for 1-wire sensors or simply as an extra input.

Pixel-Chris, 2020-10-21:
Will this be usable for the second sensor for the opened position of the garage door opener with a Shelly 1?  (Then I don’t need to power a Shelly 2.5 with an extra power supply)

rojer, 2020-10-21:
@Pixel-Chris this is indeed possible. the add-on contains circuitry that provides proper isolation for the input.

Pixel-Chris, 2020-10-21:
Ok 
I must order that addon 😉

Ludelag, 2020-10-22:
I will for sure also order one and be ready for tasting

Ludelag, 2020-11-18:
Thank you @andyblac, You already linked my issues in here nevertheless I will write them again to make it easier to read directly:
@rojer and @andyblac I just wonder if there is a support to use the Shelly 1 together with Shelly H & T as thermostat which could appear as a thermostat in the Home app? It will enable controling the heating by schedules using automation. Thank you.
[image]

Ludelag, 2020-11-18:
planning to add Shelly-Homekit the Add-on of shelly?

https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312

Thank you.
[image]
[image]

SilvioMigu, 2020-12-18:
Hello

Temperature Sensor AddOn for Shelly 1/1PM, do you think you can do it soon?
Thanks

rojer, 2020-12-18:
sorry, no ETA.

SilvioMigu, 2020-12-18:
Sorry, but what do you mean "no ETA"?

andyblac, 2020-12-18:
"Estimated Time of Arrival"

SilvioMigu, 2020-12-18:
This of being new, in this things, is what happens ...
We are always learning.

andyblac, 2020-12-18:
> This of being new, in this things, is what happens ...
> We are always learning.

no one knows everything, the world would be a boring place to live.

SilvioMigu, 2020-12-18:
I also like to share what I'm doing so that I can help each other ...

I have three temperature sensors here in my water cylinder.
Maybe there is a better way to use shellys in this area

My problem here, is that the temperature sensor Addon from shelly, has 3 wires and the sensors that I have in the cylinder, are 2 wires.
I'm still not sure how to connect the sensor ...
[image]
[image]

Pixel-Chris, 2021-03-11:
Some free timeslots left for this?

rojer, 2021-03-11:
been thinking about this lately, actually. shouldn't b too difficult, it might just happen.

SilvioMigu, 2021-03-12:
I am looking forward to seeing the final result...

Ludelag, 2021-03-13:
What Will be interesting is to be able to make a thermostat with Shelly 1 and add on

bjornand1, 2021-03-24:
I would be very interested in support for the Temp Sensor Addon - the plan is to use the sensor to measure the water temperature in our fish tank and turn on/off the aquarium lights based on the temp.

Baddy-HIP, 2021-09-03:
Hello to all and thanks for the great opportunity to integrate Shelly into HomeKit!!!

When will the addon be integrated? I am looking for a solution to control my sauna and some terrarium. Since I already control everything with Shelly switches, it would be great if I could measure and control the temperature as well.

bjornand1, 2021-09-03:
Hi, @Baddy-HIP ´, notice that this plugin, https://github.com/metbosch/homebridge-http-temperature#readme, actually works very well for the temp sensor!

Baddy-HIP, 2021-09-03:
Thank you for your information!
I tried using home bridge and  had many issues. So I would prefer a solution to implement it directly in HomeKit

Pixel-Chris, 2021-09-03:
I would use it for a second sensor for the garage door mode.

timoschilling, 2022-03-17:
I removed Temperature AddOn from title because it's supported now.

markirb, 2022-06-17:
I just noticed, in the meantime there was this commit for the driver from Allterco :smile: so the way to supporting HT got shorter
https://github.com/mongoose-os-libs/dht/commit/35a6ffc29152c7944b2552150f2039f44c142cfa

vrpt2022, 2022-07-14:
Good morning,
I would like to know if support for add-on and Shelly 1 is just for sensor DS18B20 or DHT22 is also supported. I tried the shelly 1 with add-on and DHT22 (AM2302 sensor) and neither through Web interface/home app I'm able to see any information regarding the add-on/sensor. (I can see both, humidity and temperature through native Shelly firmware...)

In the case of not being supported at the moment, do you think would it be possible to give support?

Thanks in advance for your help.

markirb, 2022-07-16:
It is not supported currently, as written in the documentation. Otherwise this issue would not exist anymore :)

I have a HT at home now but little time...

vrpt2022, 2022-07-26:
> It is not supported currently, as written in the documentation. Otherwise this issue would not exist anymore :)
> 
> I have a HT at home now but little time...

Thanks for the answer.... waiting for it :D

vrpt2022, 2022-08-24:
Good afternoon,

Any update regarding this?

Thank you

markirb, 2022-09-06:
sorry, not from me so far

vrpt2022, 2022-10-21:
Good morning,

Any update regarding the H&T sensor?

Thank you

markirb, 2022-10-22:
Mhh, did some lines already, but not there yet.
Have to finish my tax returns first unfortunately after that I home I find some time.

vrpt2022, 2023-01-28:
Good afternoon,
I’ve seen the PR advancements, but there is any update regarding this? 
Thank you!

markirb, 2023-01-28:
Not yet, the two things mentioned in the PR still exist, and I haven't found a way around the first one. So If anyone can help, it would be much appreciated

vrpt2022, 2023-02-05:
Hi,

apologies if this is no appropriate, I have absolutely no idea about coding, but found in this project a reference about integration of DHT sensors in HomeKit, could this be of any help?

https://github.com/RavenSystem/esp-homekit-devices

https://github.com/RavenSystem/esp-homekit-devices/wiki/Temperature-and-Humidity-Sensors

markirb, 2023-02-05:
There is no general issue, generally the implementation works already. 

But somehow it has hiccups on startup. This needs an oscilloscope and time to debug...

timoschilling, 2023-03-01:
@pfeifix I think you misunderstood some things. What you wrote sound like you have a Shelly 1 or 1PM and now you expect that you can use it with a DHT? If so, we don’t support the DHT at the moment only the DS18B20. Where did you read that we support DHT, give me a link please. The second thing is this issue is about the [Shelly HT](https://www.shelly.cloud/en-de/products/product-overview/shelly-h-and-t-white) and not about the DHT sensor. Support for the DHT sensor is in progress #1110 and the thermostat support you ask for is in #895 requested.

pfeifix, 2023-03-01:
@timoschilling: Thanks for the update! 
Indeed I meant the DS18B20 (not the DHT) which is what I have bought.
So you say the pure temperature sensor accessory is already built-in in the latest firmeware of the Shelly 1? That would be great.

I am looking forward to the release of the thermostat support as tracked by https://github.com/mongoose-os-apps/shelly-homekit/issues/895)](https://github.com/mongoose-os-apps/shelly-homekit/issues/895).

vrpt2022, 2023-08-03:
Good evening!

Any update?

Thanks!

closed 2024-07-22 by markirb