mongoose-os-apps/shelly-homekit #187: After Update HomeKit Automation Devices gone (only Shelly 2.5 as Shutter)

Rappelsack, 2020-10-13:
Hi there.

Yesterday I updated all my Shelly to the new 2.3.0.
After the update all Shelly 2.5 are deleted out of the HK automation.

I have a lot of automations as for the Shelly 1 (at XX:XX switch on shelly....).
Also for the 2.5 (shutter; at XX:XX close the shutter in bedroom).

After the update all Shelly 1 still normal.
All shutter Shelly 2.5 were deselected (in the automation I hat to set all devices again. All the shelly 2.5 automations were without devices).

Maybe this is a known issue?

Montardit, 2020-10-13:
I have the same problem on 1 from 4 shelly 2.5 as rolling shutter with 2.3.0. Web UI Ok but no response on homekit.

Rappelsack, 2020-10-13:
That is not the same problem. Maybe u misunderstood something in my problem.
My problem is that all automations with the shelly 2.5 the devices are deleted in this automation after updating 2.3.0.

andyblac, 2020-10-13:
> That is not the same problem. Maybe u misunderstood something in my problem.
> My problem is that all automations with the shelly 2.5 the devices are deleted in this automation after updating 2.3.0.
> 
> I do not have a "no response" issue.

what version where you on before ?

Rappelsack, 2020-10-13:
I updated from 2.2.1 via the GUI to 2.3.0.

andyblac, 2020-10-13:
> I updated from 2.2.1 via the GUI to 2.3.0.

ok and you changed setting to Roller Shutter ?, if so unfortunately I think that correct behaviour, as the shelly is then put in Bridge mode, HomeKit will think it a new accessory and thus remove any old automations from old accessory.

@rojer will be able to confirm this.

Rappelsack, 2020-10-13:
No. All was set. With 2.2.1 the were roller shutter. I just updated to 2.3.0. Not more, not less.
And than in all automations the shutter are gone "Automation at 10:00pm - no device selected"

andyblac, 2020-10-13:
> No. All was set. With 2.2.1 the were roller shutter. I just updated to 2.3.0. Not more, not less.
> And than in all automations the shutter are gone "Automation at 10:00pm - no device selected"

ok, maybe some code, was changed and resulted in the same thing, HomeKit thinks it is a new accessory, maybe due to Garage Door type been added.

Rappelsack, 2020-10-13:
ok. Thats possible. 
Now I set them all new.

andyblac, 2020-10-13:
I'll close this, reopen if you need too.

closed 2020-10-13 by andyblac

rojer, 2020-10-13:
my apologies, i think https://github.com/mongoose-os-apps/shelly-homekit/commit/407bb9b7040ef728419f43659dbfb585f20fbcd6 is to blame.
you see, when initially adding window covering / roller shutter support i made a mistake by giving them the same IID range as stateless switch - a copy-paste error, forgot to update the number.
i discovered this when adding garage door opener and just fixed, not thinking much of it.
but actually, it causes renumbering to happen on update and all existing roller shutter devices to seemingly disappear.
sorry, yes, you will have to set up the devices. it won't happen again.