mongoose-os-apps/shelly-homekit #189: Garagedoor setup

tohmc, 2020-10-13:
Hi,

Is it possible to add a mode that activates two outputs on Shelly 2.5?
One output to open the garage and one to close the Garage?
My garage have separate inputs to open and close.

Thanks
Tohmc

rojer, 2020-10-13:
yes, it's possible.

tohmc, 2020-10-13:
Can you explain how to do it please ?
👍

rojer, 2020-10-13:
well, what i meant was - it's _feasible_. needs some code changes to make it _possible_. :)

tohmc, 2020-10-13:
Do you think you can add this feature?
It would be very nice ;-)

rojer, 2020-10-13:
please test this firmware: http://rojer.me/files/shelly/2.3.1-beta2/

set "output mode" to "dual", connect output 1 to close input of the controller, output 2 to open.
let me know if it works.

tohmc, 2020-10-13:
Thanks!

Great work!
I will try on weekend!

rojer, 2020-10-13:
btw, what type of a controller do you have?

rojer, 2020-10-18:
released 2.3.1 with this change, let me know if it doesn't work.

closed 2020-10-18 by rojer

tohmc, 2020-10-19:
Sorry I had no time this weekend to Test.

Thanks for the great work.
If it doesn’t work I will inform you.

Regards

tohmc, 2020-10-19:
I have an marantec Highline 15
It’s close to the comfort series