mongoose-os-apps/shelly-homekit #192: Shelly 1L support?

Tomer51m, 2020-10-14:
Hi guys! 
I there a rough ETA for Shelly 1L support for HomeKit?
Thanks!

Tomer51m, 2020-10-14:
Not yet. planning to buy a few next month. will let you know.
I have another question, but I will open a new issue so others can find it quickly.

rojer, 2020-10-14:
so, it's not yet released, expected in mid November. i think it will be easy to add support, but i need one to test. feel like getting one for me to work with?

andyblac, 2020-10-14:
what will people use them for ? as there only 5amp.

rojer, 2020-10-14:
that's plenty for any light, pretty much. led lights use milliamps.

Tomer51m, 2020-10-19:
> so, it's not yet released, expected in mid November. i think it will be easy to add support, but i need one to test. feel like getting one for me to work with?

Sure! I will get a few next month. I have my current order stuck in customs 🤦‍♂️. Once its released I will buy few L1’s to try.

rojer, 2020-10-19:
ok, i'm sufficiently curious that i pre-ordered a couple for myself. will take a look when they arrive.

andyblac, 2020-10-19:
> ok, i'm sufficiently curious that i pre-ordered a couple for myself. will take a look when they arrive.

last time i ordered "in stock" items direct from shelly they took over a month to arrive, that why i do not purchase from shelly.cloud.

pivale, 2020-10-20:
for portugal, there are several electronic stores that have next day delivery. I either order from https://mauser.pt/ or https://www.castroelectronica.pt/ depending on stock availability.

Pixel-Chris, 2020-10-22:
> what will people use them for ? as they're only 4amp. ok for single use devices like lights, tv, etc

You need no NEUTRAL wire! that‘s the benefit

andyblac, 2020-10-22:
ye i know.

andyblac, 2020-11-04:
rumours are Shelly are working on a 2L 😄

rojer, 2020-11-25:
please help test 2.6.0-beta - https://github.com/mongoose-os-apps/shelly-homekit/issues/291

closed 2020-11-25 by rojer