mongoose-os-apps/shelly-homekit #194: shelly 2.5 cycle switch

kx3kx3kx3, 2020-10-14:
it's possible to reproduce this with homekit 
[image]

rojer, 2020-10-14:
i guess. does it mean tapping either tile in homekit does this?

andyblac, 2020-10-14:
> i guess. does it mean tapping either tile in homekit does this?

i think only for physical switch.

rojer, 2020-10-14:
either way with the way code is structured now, it would need some ugly hacks, as currently switches and their inputs are completely independent.
not impossible, but not trivial. let's see if there's more demand for this.

kx3kx3kx3, 2020-10-14:
> > i guess. does it mean tapping either tile in homekit does this?
> 
> i think only for physical switch. for single switch to use on 2 channels

yes only for physical push button

rojer, 2020-10-14:
with just the physical button, it's a bit simpler but still, kinda meh. will leave this one to gather more support.

riccardog821, 2020-11-30:
Up 
I have installed HomeKit firmware on a Shelly 2.5 that was set as cycle switch controlling 3 lights with 2 control buttons. Now there is a way about how configure the Shelly so that pressing first time the button “channel 1” will be turned on and on a second push the “channel2” will be on?

kx3kx3kx3, 2020-11-30:
I found how to do it but it is an old firmware that is no longer supported

riccardog821, 2020-11-30:
Oh what a sad news :( so no any other option to make it work as per the original firmware but also on HomeKit?

lucaspinelli85, 2021-03-17:
I think it can be useful to many

lucaspinelli85, 2021-06-30:
In this case the relay would no longer be directly connected to the button and would only work via the wifi network. I don't want to avoid disconnection problems, if there is no wifi network nothing would work. It would be much better if the function is integrated into the device, so it works even if it is not connected in wifi. Can you put it in the 3.0 firmware?

lucaspinelli85, 2021-12-29:
when will the cycle switch function be implemented?[image]

lucaspinelli85, 2022-01-05:
Please @rojer can you implement this feature?
In two rooms I only have space for one button for two lamps, I can't add another button ... Currently I have the original firmware and I would like to transform it HomeKit like all the other shellys I have!
Thank you!

robfero, 2023-01-27:
Please implement this. Your firmware is solving a lot of issues I had with homebridge. This is the only thing that I'm missing.

antodan91, 2023-03-21:
I follow this discussion and I hope that this feature will be implemented. @rojer do you think that it is possibile? Many thanks for your firmware.

Tomvol-1, 2023-05-21:
Ciao , è possibile implementare questa opzione?

Grazie