mongoose-os-apps/shelly-homekit #195: GarageDoor wiring

49adrien, 2020-10-14:
Hello,

On this picture, the wiring is not really clear, 12VDC is supported ? 110-230VAC ? 24-60VDC ?

[image]

Thanks

Whiterabbit64, 2020-10-15:
You must use a low DC voltage power supply in this configuration. DON’T use AC.

49adrien, 2020-10-15:
Can you explain me why ?

elgalera, 2020-10-15:
You can use 12Vdc or a voltage from 24Vdc to 60Vdc. Remember set the jumper according to input voltage.

The magnetic contact sensor only can operates with DC voltage. You can not use AC with this type of sensor.

You can find the correct configuration of SW in the following link:
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Garage-Door-Opener-Settings

49adrien, 2020-10-15:
This contact sensor support AC ?

[](https://www.aliexpress.com/item/4000890440801.html)

Whiterabbit64, 2020-10-15:
It’s dangerous to power a such device with « high voltage ». If there’s an issue with it you could get hurt through the door which is probably metallic.

49adrien, 2020-10-15:
I have found this product :

[https://fr.aliexpress.com/item/32966599687.html](https://fr.aliexpress.com/item/32966599687.html)

Whiterabbit64, 2020-10-15:
Do what you want, but I’ll never take a such risk. It’s better to work in low voltage and to add a small 24 DC power supply which costs few Euros.

49adrien, 2020-10-15:
Ok thanks.

49adrien, 2020-10-15:
Ok thanks.

closed 2020-10-15 by 49adrien