mongoose-os-apps/shelly-homekit #199: Add option to reset AutoOff Delay every time an ON command is received

jsiegenthaler, 2020-10-16:
Hi

I use this great firmware in as a hallway light switch. 
I would like the option of resetting the auto-off timer every time the switch is gets a turn ON command.
ie: auto-off 5 minutes after Last trigger (and not after First trigger)

Use case: 
A persons walks down the hall and triggers the motion sensor. The auto-off timer starts with 5min to run.
2 minutes pass. The auto-off still has 3 minutes to run.
Another person triggers the motion sensor.
The auto-off is reset and now has 5 minutes to run.
The light stays on the whole time

rojer, 2020-10-16:
this makes total sense, i don't think we even need an option, it should just be like that.

andyblac, 2020-10-16:
This would only work in detached mode

rojer, 2020-10-16:
how so?

andyblac, 2020-10-16:
> how so?

ATM we don’t support sensors do we ?

andyblac, 2020-10-16:
Oh I get it you mean wired sensors ?

rojer, 2020-10-16:
basically, restart auto-off timer any time we get an on command, via whatever means (hap, web ui). this makes sense.

andyblac, 2020-10-16:
Can’t be switch as next press would be OFF in momentary, and same with rocker

rojer, 2020-10-16:
but could be with automation.
although it's a good point, @jsiegenthaler did you mean triggering with a physical input?

jsiegenthaler, 2020-10-17:
@rojer:

No, not physical.
 I mean reset the auto-off timer over the Homekit system via a MotionSensor telling it to go ON again when new motion is detected.

My real world use case in my apartment is as follows:

Kitchen with Philips Hue motion sensor
Kitchen Lights are fluorescent, switched by Shelly1 using wall switch connected to SW input, SW input in Edge mode.
Either motion sensor, Homekit app, or wall switch can turn lights on and off. This keeps the entire family happy.

When person A enters the kitchen, the lights come on, and delay timer starts
Person A leaves and person B goes into the kitchen, delay timer is still counting down, and the chance exists that the lights will go off while person B is cooking. That creates family friction..

The only way I have to prevent this at the moment is to disable the Shelly1 autooff, and use a Homebridge DelaySwitch plugin.
The DelaySwitch has it's delay timer reset at every MotionSensor trigger from the Hue motion sensor. 
Thus we achieve the goal of "delay starts after Last motion was sensed", and not "after First motion was sensed".

I was hoping the delay timer reset could be built in to the Shelly1 (mongoose)
That would allow me to save a switch and an automation, and increase reliability, as sometimes the delay timer automation does not fire for unknown Apple reasons.

PS: I am not familiar with the HAP protocol, but I assume that a "turn a light ON" command still gets sent, regardless of the light's current power-state status.

rojer, 2020-10-17:
ok, done, will be in the next release.

closed 2020-10-17 by rojer

lucaspinelli85, 2020-11-16:
I don't find this function in the last release

andyblac, 2020-11-16:
> I don't find this function in the last release

iirc it is just enabled by default. no toggle to turn on / off

lucaspinelli85, 2020-11-16:
> > I don't find this function in the last release
> 
> iirc it is just enabled by default. no toggle to turn on / off

I don't understand, if my toggle put on the switch, then start the auto off, but while auto off running the toggle go off, the auto off stop and light off.
What I need is this: toggle go on, light on, when toggle go to off I need the switch keep on for "x" time (auto off after...).
How can I do this?
Thanks

andyblac, 2020-11-16:
> I don't understand, if my toggle put on the switch, then start the auto off, but while auto off running the toggle go off, the auto off stop and light off.
> What I need is this: toggle go on, light [...]

you can't, why turn it off, why not just let auto off turn it off, thats the purpose of it.

lucaspinelli85, 2020-11-16:
In specific this is my situation: I have a rain sensor (toggle), when toggle is active it stop irrigation and grass cutting. When toggle disactivated (sun!) it immediately don't stop more the irrigation and cutting, but the grass is still wet! I want the toggle is for 24 more hours activated after off, so dries grass!
In original firmware there is "Activation switch" mode that used for motion sensor, any inputs turn ON an resets auto off timer

andyblac, 2020-11-16:
> In specific this is my situation: I have a rain sensor (toggle), when toggle is active it stop irrigation and grass cutting. When toggle disactivated (sun!) it immediately don't stop more the [...]

ok, so you need a new Switch Mode, "Sensor" -> On Only. so the switch on the device only actives with ON. iirc stock shelly has this mode, @rojer could you add this option ?

andyblac, 2020-11-16:
@lucaspinelli85 create a new ticket for this.

andyblac, 2020-11-16:
ie 
[image]

lucaspinelli85, 2020-11-16:
Almost... or rather, in the original firmware auto off start when input on, and it's perfect for motion sensor. In my case I want the auto off start when toggle go off, practically shelly keep on the output after switch off (otherwise, after "24h" from start raining it go to off tough it's still raining..). I hope you understand me...

lucaspinelli85, 2020-11-16:
You cloud add both, the mode "sensor" at ON (restart auto off when input) and at OFF (keep on for...)...

andyblac, 2020-11-16:
anyway open a new ticket.

lucaspinelli85, 2020-11-16:
> anyway open a new ticket.

Sorry, how can do a new ticket? You mean new issues or new pull request? Thx

andyblac, 2020-11-16:
goto to "Issues" click on "New"