mongoose-os-apps/shelly-homekit #200: Roller shutter automation fails

otiman, 2020-10-16:
Hello Rojer, first many thanks for your work! 

With fw 2.3.0 i have the following problem: i created an automation for my 11 shelly 2.5 to close at sunset. If a roller is already closed at this time, it will open, even if the automation says close. The already closed shutters go open and the others go close. I have deleted the automation and created it again but without any change. 

Maybe you can solve this in an upcoming Update. Thanks!

rojer, 2020-10-16:
yep, already fixed on master, will be in the next release.

rojer, 2020-10-16:
actually, fixed for garage door but not yet for the roller shutter.

rojer, 2020-10-17:
will be in the next release

closed 2020-10-17 by rojer

Montardit, 2020-10-17:
> Hello Rojer, first many thanks for your work!
>
>
>
> With fw 2.3.0 i have the following problem: i created an automation for my 11 shelly 2.5 to close at sunset. If a roller is already closed at this [...]

I detected the same problem. Try to set it to 1% or 99%. It's nearly the same and it works well on automations till next update.