mongoose-os-apps/shelly-homekit #201: Request: Add a documentation snippet about How a reflashed Shelly works directly with Homekit

jsiegenthaler, 2020-10-16:
I'm a techie guy (an engineer) and was wondering how on earth a mongoose-flashed Shelly1 can communicate directly to Homekit?

So I reflashed one of my many Shelly1s (which previously connected to Homekit via Homebridge and a Shelly1 plugin)  and loaded mongoose 2.3.0
I was highly delighted by the functions but noted some weird behaviour as well (I'll log bugs when I can reproduce the behaviour).

But still perplexed as to how it connects to Homekit, given that the Shelly1 is Wifi and not Zigbee.
Then I reviewed the Homekit config.
And found the secret: the Shelly1 appears as it's own Bridge.

Which makes me ask the question: is every mongoose-flashed Shelly an Accessory and a Bridge at the same time?
So if I reflash 10 x Shelly1s, will I have 10 extra Bridges added?
And if so, could you add a brief snippet to the Wiki describing this? Doesn't need much.

rojer, 2020-10-16:
and to answer your question specifically - yes, you may end up with 10 bridge accessories in your home app (depending on the configuration of each device - as i said, we only create a bridge accessory if we have to), but in practice it won't mean much except an additional tile.

rojer, 2020-10-17:
added https://github.com/mongoose-os-apps/shelly-homekit/wiki/Internals#hap-details

closed 2020-10-17 by rojer