mongoose-os-apps/shelly-homekit #203: HomeKit actions lead to out-of-sync behavior for physical switch in toggle-mode

philipptrenz, 2020-10-17:
Hey there, 

thanks for this great project!

I noticed an issue when using a Shelly 1 on firmware version 2.3.0 in input mode _toggle_:

When the physical switch is used first and then the output state gets changed via HomeKit or the web interface, the physical switch has to be used twice to change the output state. Strangely, when updating the web interface, the correct output state gets shown. So some internal state seems to get out of sync.

I hope this description helps to track the issue down.

rojer, 2020-10-17:
hm, no, not quite. can you clarify what the current and expected behavior is?

philipptrenz, 2020-10-17:
Oh okay, I‘ll try:

My expected behavior of toggle-mode is, that on each use of the wall switch the Shelly output changes its state (connected light turns from on to off or other way round).

The current behavior is, when I turn on the light via the wall switch and turn it off using HomeKit, I have to use the wall switch two times until the lamp turns on again. Same for turning off using the wall switch and afterwards on via HomeKit, it needs two wall switch operations to turn off again.

When I turn the light on and off again (or other way round) via HomeKit, the wall switch will change the relay state of the Shelly on the first usage.

rojer, 2020-10-17:
what you are describing sounds more like "Edge" mode - flip state on every _change_ of the switch. "Toggle" mode mirrors the _state_ of the physical switch, i.e. on or off, so naturally it may go out of sync with reality if state is changed via other means.

philipptrenz, 2020-10-17:
Oh alright, then I misunderstood the documentation. 
Thanks for your quick reply and explanations!

closed 2020-10-17 by philipptrenz