mongoose-os-apps/shelly-homekit #204: Manual percentage adjustment in Roller shutter mode

heili1094, 2020-10-17:
Hy, the Software for Homekit is WOW!!!
I face one issue which is only cosmetic 😊
Possible to change the percentage which displays  in HomeKit App? F.i. 37% roller shutter position is in real 15% so possible to off set manual?

Thanks 
BR René

rojer, 2020-10-17:
yeah, this one may not be as easy to tackle as simply add an offset. while working on #5 i deliberately put off positioning accuracy because of the differences. for example, it's been mentioned that some shutters have non-linear dependency between the amount of movement time and position.
so let's start gathering data. for your shutters, can you measure actual position in 10% increments? as in - set position to 10% in homekit, measure actual position, set to 20%, measure, etc up to 90%.
also, don't do it sequenctially 10-12-30-... (becasue error will accumulate), do 0-10-0-20-0-30-..., up to 90 (0 being fully closed position).

heili1094, 2020-10-18:
Hy,

stupid question- how should i measure? With measure tape? 
Thanks 

Sent from 

rojer, 2020-10-18:
pretty much, yeah.

heili1094, 2020-10-18:
https://www.icloud.com/iclouddrive/0I7PrVtKQHjyausFshEOycgpg

Hy, enclosed the link and hope it helps to find a solution 

Br René 

Sent from 

rojer, 2020-10-18:
thank you!
so, it looks like about 1/4 of movement from closed actually goes to open gaps between the slats.
when the curtain starts moving, from 30% on, we have the following values for number of cm per 10%:
13, 13, 14, 17, 19, 22. this non-linearity towards the end is a bit puzzling.
and the end position is not that far off from the 90% - it's around 80% by my quick calculation.

heili1094, 2020-10-19:
Hy, so and Solution for this „cosmetic“ issue? 
Thanks

rojer, 2020-10-19:
i'll need some more data from users to decide what to do.

heili1094, 2020-10-19:
Ok Thanks

juansf99, 2020-10-22:
Hello,
For a few weeks I have installed the firmware on a door blind.
The installed firmware version is 2.3.0
The measurements have been made following the pattern: 0-10-0-20-0-30-0-40 ...

% | cm
10% | 4.5
20% | 18
30% | 32
40% | 50
50% | 69
60% | 89
70% | 111
80% | 135
90% | 162
100% | 194

As discussed above, a non-linearity is observed in the path of the blind. At the beginning it goes up a little and at the end of the route it goes up very fast.
I hope that this data can be of help to be able to add some parameter in the web interface and correct this behavior.

Thanks,
JASF

rojer, 2020-10-22:
@juansf99 thank you!
what i'm thinking here is maybe we'll be able to offer a number of curves for the user to choose, linear being the default and then various options. let's collect some data to know which options to offer.

ken-sa, 2020-10-25:
hi rojer, 

thank you for your great work. 
i have the same problem. 

Her the mesaurment from my roller.  

10% | almost close 
20% | [image]
30% | 6
40% | 26
50% | 46.5
60% | 71
70% | 97
80% | 122.5
90% | 158
100% | 216

tomorrow i will give you a feedback from the second one.

domke24, 2020-10-27:
Hi Rojer,

at first: the HomeKit Firmware is amazing! Thanks for that!

I have the same problem. I use a shelly for one of my windows. The problem is that when it´s fully closed the home app thinks it is nearly 25% opened. So in the home app the roller shutter are always open and never closed.

I would be perfect if you can find a solution.

Thanks in advance!

ken-sa, 2020-10-29:
I tried it today with the stock firmware and it worked. After that I loaded the HomeKit firmware and calibrated it again. In the attachment you can find the log, maybe it can help. 
[shelly_calibration.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5461299/shelly_calibration.txt)

chstoll, 2020-10-30:
I really don't know if there is an issue here. I have roller shutter modules from fibaro (z-wave) and some really old marmitek X10 and they all calibrate in a similar way (just like the shellys with rojers homekit firmware). What do you expect 50% should be? In fact, the picture above seems exactly 50%, because the motor has to run quite a bit to get from closed (with open slots) to fully closed. So, knowing that just take 60% to get the shutter to an "optical" 50%.

ken-sa, 2020-10-30:
If I want to reach the optical 50% I have to open the shutters to 74%, but that is not the point. We want to improve the firmware that works very well and bring it to another level. This is not a criticism from my side but rather a suggestion to get better. Please do not misunderstand.

rojer, 2020-10-30:
50% in home app should mean exactly half open, half closed. meanwhile, let's continue collecting data.

yuvalbeck, 2020-11-02:
[Shutter.pdf](https://github.com/mongoose-os-apps/shelly-homekit/files/5473288/Shutter.pdf)
Hi Rojer, I had some time I put tapes every few cm on the window and filmed it - attached the numbers.
Not 100% accurate but maybe could help.

thatslolo, 2020-11-04:
I spotted a problem on my blinds that there on different positions when set to the same, depending from which state they started to move to the desired position. It's obviously the same problem. I'll try to document it asap.

andyblac, 2020-11-04:
sounds to me we need 3 positions for calibration FULLY OPEN, CLOSED, and MOTOR REST.

OPEN = shade at open position
CLOSED = shade at visible closed position
MOTOR REST = let the shade motor go to fully closed posotion, so all slats compress into the fully closed position.

@rojer would this be posible ?

thatslolo, 2020-11-04:
I don't think that can be represented in HomeKit tbh.
But it could be used to determine the actual position of the shutter when set to a specific open state in percentage.

andyblac, 2020-11-04:
why not, 1% could be CLOSE, and 0% could be REST.

thatslolo, 2020-11-04:
And why not make it configurable as an advanced option, so you can set the lower 10% for just changing the state of slats e.g. Just an idea. :)

andyblac, 2020-11-04:
> And why not make it configurable as an advanced option, so you can set the lower 10% for just changing the state of slats e.g. Just an idea. :)

thats why i suggested in a aditional 'Roller Slats" option.

pivale, 2020-11-06:
> why not, 1% could be CLOSE, and 0% could be REST. so if 10% is called for you actually get 10% etc. if in REST and you call 1% the shade open to CLOSED position.
> 
> i doubt hardly anyone will acually have a real world use for 1%, and even if they did the differance between 1% and 2% would hardly anything to complain about.

I actually use a lot the "only slats" open in my shutters :)
In the summer to prevent the house from overheating but still have some light...
Or to have some light in the rooms when my kids have their afternoon naps...

I like the idea of 0-10% being for slats and the rest for normal opening.

To configure this correctly, I would suggest that before calibration, the user sets the shutters to only slats open and start from there. That could give you an educated guess of what should be the slats time (i suspect really fast).

andyblac, 2020-11-06:
> I actually use a lot the "only slats" open in my shutters :)
> In the summer to prevent the house from overheating but still have some light...
> Or to have some light in the rooms when my kids have [...]

what i was saying is do you do a REAL 1% open, (forget about open slats), i was saying that we could have 1% as "open slats", and 2% as a REAL 1%, so in home kit when you ask for 1% it just opens slats.

pivale, 2020-11-06:
> i can't see how you could have 10% to open slats and also have a REAL 50% open????

You can if you have a non linear counting of the percentage.
You could have something like:
0-all closed
10-slats fully open/shutter down to the ground
50-shutter half open
100-shutter fully open

Still, I would be happy with something like 10% for slats, even if that meant that "real half open" would be 45%... I can deal with a 5% error 😉

andyblac, 2020-11-06:
> You can if you have a non linear counting of the percentage.
> You could have something like:
> 0-all closed
> 10-slats fully open/shutter down to the ground
> 50-shutter half open
> 100-shutter fully open [...]

thats my point i very must doubt HomeKit slider willl support none linear percentage. thats why sugested 1% as "open slats"

so in calibration you would have 3 postions 100% = open, 1% = Open Slats, 0% = Fully closed.

pivale, 2020-11-09:
Hey,

Here is my data, I have a "small" window and a "big" window.
Although I set it to round numbers, in the end home app reported a different percentage. The values here are the ones reported by home app after stopping.

Small
at 6% the shutter is touching the ground but slats are all open
0-9% - 2cm
0-21% - 12cm
0-30% - 21cm
0-40% - 32cm,
0-50% - 44cm
0-59% - 55cm
0-69% - 69cm
0-79% - 84cm
0-89% - 101cm
0-100% - 120cm

Big
at 8% the shutter is touching the ground but slats are all open
0-9% - 1cm
0-19% - 15cm
0-29% - 31cm
0-39% - 49,5cm,
0-50% - 68cm
0-59% - 90cm
0-69% - 113,5cm
0-79% - 139,5cm
0-89% - 167cm
0-100% - 210cm

DarkAppleLead, 2020-11-12:
Hey rojer, first of all thank you for your great project! 

I have a problem with one of my shutters. It says my window is fully open, but it isn‘t 😅
[image]

CyberMrProper, 2020-11-15:
Hi @rojer, here are my numbers:

10% - 0 cm : all time is spent opening the slats, no visual window is opened
20% - 13 cm
30% - 28 cm
40% - 46 cm
50% - 65 cm
60% - 87 cm
70% - 109 cm
80% - 135 cm
90% - 159 cm
100% - 192 cm

Thanks!

funkosgr1410, 2020-11-15:
Hello Rojer,

Many thanks for this software which is really awesome.
With regards to calibration, I use a shelly 2.5 which controls multiple shutters (it has been installed on a shutter central switch that controls all shutters).
This means that calibration is not applicable as each window has its own position for the same time that switch is open.
As such calibration now fails and there is no alternative to set manually a timer (e.g. 26 seconds) for open close time.
This is not related to this specific software as calibration was also failing with original shelly software.
How likely could it be to develop an override option that could use open/close based on a predefined time interval?

For the record, calibrating shelly 2.5 devices that are individually installed on each shutter works perfectly and thanks again for your excellent development.

Thanks!

greensouth, 2020-12-10:
For this issue, please take a look at the approach made by Jose, the author of HAA firmware about how to set a modifier for Window Covering:
[https://github.com/RavenSystem/esp-homekit-devices/wiki/Window-Covering](url)

`HomeKit Position = motorPosition / (1 + ((100 - motorPosition) * correctionValue * 0.0002))`

In there, you set a correction value and see if it visually matches with the slider position. It's just a question of trying some values until you fine-tune the visual aspect.

Lausbuab1, 2020-12-19:
Hi Rojer,

as also mentioned above, your work is awesome! Thank you for that.

Sadly, I also have the same issue with my shutter and shelly 2.5, that the shutter is about 50% closed and the app says, it's 80 percent closed.

Is it possible to add a manual calibration? Like entering the percent if the we can control it by ourselves?

janigro, 2021-01-06:
> come to think of it, non-linearity is expected - there's less curtain to lift, so more of the energy goes to movement rather than counteracting gravity, so it goes faster per unit time.

Hi Rojer, even if the motor manages to keep a constant speed at all times, the radius of the rolled shutter must be taken into account. In the attached picture, you can see that one full turn of the roller lifts 4 slats at first, while the same full turn at the end will lift 12 slats! That's 3 times more per turn of the roller. This heavily depends on the shutter type. The typical shutter in Spain (like the one attached in the picture), has a very thick slat (around 10mm + the spacing while rolled). However, a cloth blinder has probably 2mm at most. The first case will be much less accurate than the second.

I think I can work out a formula that takes into account the slat thickness and the initial radius of the roller (i.e., the roller before rolling any slats).

I have not received my Shelly 2.5 yet, but I'll try to figure it out when I get my hands on it. I'll report back.

[image]

Marcusg562, 2021-01-24:
Thanks @andyblac haven‘t seen it... but yeah, I have the same issue.

greensouth, 2021-04-02:
no improvements on this?

WaW1190, 2021-04-15:
Hi, first of all I want to say a HUGE thanks for this, this made managing my smart home so much more easy.

I do have the same issue, running FW 2.8.2.

50% opening in HomeKit or the Web interface is roughly 75% closed. Everything else works flawlessly

jtolsa, 2021-09-15:
HI Rojer,

Many thanks for the software, it's just awesome.

Below my measurements. Would be great to these curves somehow implemented in the code... doesn't need to be to the mm but a rough approximation would be great!

10% - 1.0 cm
20% - 11.5 cm
30% - 24.0 cm
40% - 37.5 cm
50% - 53.0 cm
60% - 69.0 cm
70% - 87.0 cm
80% - 105.0 cm
90% - 125.5 cm
100% - 153.0 cm

Many thanks again!

Geert92, 2021-10-30:
Hi I’m new to this. 
I started using the shelly 2.5 for my roller shades (not blinds). It seems to work sort off,
When calibrating it does not go down. So I xannot fully controll it at all. 
Can someone advise me on what to do?
Thanks 🙏

jekruege, 2021-11-01:
Hello, I have problems with the calibration, too. 
It goes down 1 second und goes up 1 second and repeat it 50 times. Can I calibrate manual?

Geert92, 2021-11-01:
> Hi I’m new to this.
> I started using the shelly 2.5 for my roller shades (not blinds). It seems to work sort off,
> When calibrating it does not go down. So I xannot fully controll it at all.
> Can someone advise me on what to do?
> Thanks 🙏

Small side note. Im using a motor that has a limiter build in, so it cuts off at to min and maximum hight.

marvinrose, 2022-04-06:
> I tried it today with the stock firmware and it worked. After that I loaded the HomeKit firmware and calibrated it again. In the attachment you can find the log, maybe it can help. [shelly_calibration.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5461299/shelly_calibration.txt)
> 
> And here is the result, after the calibration. [image] [image]

Same for my installation. Did you adjust anything else so that 50% is really 50% open?

greensouth, 2022-05-18:
Hi! No further news on this? Latest post can't be more explanatory :P

marcus-aa, 2022-06-19:
OK, would be good to improve on the non-linearity calibration.

There are mainly 2 parts of this problem.

1: many shutters are adjusted so the end position is quite a lot further than when the slats are in the lowest position. The motor turns a bit longer until it reaches it's limit switch. So when opening, it first turns without moving the slats correspondingly. This can be around 10% in my test installation. So at 10%, we are still fully closed. 

Now, the motor starts to lift the slats, but many of them are made so they will first separate slightly to let light in. Now we can be at 15%, and we see light through them, but the lowest one is still all the way down. This position is often wanted in automations and it would be good to be able to calibrate so that all of them are at the same % in this position.

This is a fixed offset that could be entered. "seconds to transparancy position, set that to 1% and then start counting normally as it starts to lift.

So, 0% - fully closed, 1% lifted to see-through position but bottom slat still in lowest position. 2% - bottom slats starts lifting.

2: The next problem is the increasing size of the roll that oulls up the slats.

Many shutters work on a 3" / 76-80mm pipe (but there are also other sizes). So, the first turn will then pull up 78x3.14 (pi) = 245mm of the shutter. 
Then a layer of the slats are added, with some extra space as they are not optimally formed around themselves (assumed to 25% here). 
So the second layer of 12mm slats will have a diameter of 78 + 2x 12 x 1,25 = 108mm, and lifts 108x3.14 = 340mm

The third layer will lift (78 + 4 x 12 x 1.25 ) x pi = 433mm
4th : 528 mm

and so on. 

to fully compensate for this we would need to input the base pipe diameter, the slat thickness and the length of the shutter. 

But, the pipe dia is just an offset, pi is of course constant, and the length is relative to the travel time.

So, the actual influence is the slat thickness. It decides the slope of the curves that show the increasing change of position as a function of turns, time, length.

[image]

ingski, 2022-11-30:
Hi, just let me thank you for this great work. I have a bunch of Shelly devices in our home, all with your firmware and they work like a charm. This includes one roller shutter with a 2.5 shelly.

However - today I flashed a Shelly 2.5 that is attached to three roller shutters that are controlled simultaneously and the calibration turned out to not discover full open or closed state even close :-(

Depending on where the shutters are before I calibrate it kind of expands from there:

- calibration started with all shutters fully open results in a "closed" position that covers about 1/3 of the windows while "open" is just fine
- calibration started with all shutters fully closed results in a "closed" position that is at about 98% closed while open just leaves about 1/3 of the windows open

Actually instead of the automatic calibration a timer would be great, since all shutters are equipped with an automated stopping switch at either end...

Any suggestions how to solve my issue?

Ingo

[image]
[image]
[image]

desek90, 2023-06-11:
Hello, is there any progress on this topic?

LukasFoukal, 2024-07-09:
Hello, first of all thank you for this amazing OSS project!

I have been using this software for some time with the PlugS and RGBW2, but with the support of 2PM I have begun using it with blinds as well.

I understand that blinds are not roller shutters and therefore there is some specificities that are similar to the issue discussed here.

I am talking about achieving this flat position, which allows light in without heating the room too much. As a percentage of the whole travel this is around 2% open.

[image]

So I would also welcome some non-linearity option to achieve greater accuracy in the angle of the blinds, since right now with the course control in 1% increments it is very difficult to achieve.

markirb, 2024-07-09:
I think there are several pull requests addressing this. most notably #1033 
that needs to be rebased and tested