mongoose-os-apps/shelly-homekit #207: Garage opener status not working in HomeKit automation

mac-alx, 2020-10-18:
Hello,
the new garage opener function is great!

When creating an automation with shortcuts the automation will not work when pulling a request on the door status. 

Example:
12:00
If garage door is open
  Then
      Light on

The automation will work if you start it manually, but not automatically at a specified time. 

The automation will terminate as soon as it comes to a request on the Shelly. 

My IOS is the latest iOS 14 and Shelly FW is 2.3.1. 

Many greetings
Marc

rojer, 2020-10-18:
i tested this and could not reproduce.
i created automation that tied a light to status of garage door. i used "at night" instead of specific time, that's the only difference.
but otherwise, it worked - when door opens, light turns on (via automation), when door closes, light turns off.

rojer, 2020-10-18:
using ipad mini with ios 13 as hub.

mac-alx, 2020-10-18:
My hub is an Apple TV 4K. 

I do not have an automation on an open event but time triggered. 

Example which works:
18:00

Shortcut

light 1 on
If garage door current state is open
   Light 2 on
End if

Result: light 1 turns on, light 2 not. The script ends as soon as you put a request on garage door current state. 

Did you do the automation with or without shortcuts? I only have the problems when using shortcuts 

(See pic, sorry but my iOS is German)

[image]

mac-alx, 2020-10-18:
The automation above will work if you start it manually, but not at 18:00

mac-alx, 2020-10-18:
I just testet your automation:
When creating it without the Shortcut script ist works. 

(What I want to have at the end is an alert at 8 pm via pushcut if the door is open.  => the light was just to test the function)

rojer, 2020-10-18:
this seems like an ios bug. it really makes no difference to the device whether there's a shortcut in home app or not.

mac-alx, 2020-10-18:
Could you recreate that behaviour? Just to know if it is (another) iOS 14 bug?

Right now I do not have that with other devices. The same automation works eg with an eve door sensor.

mac-alx, 2020-10-20:
Update:
Did an update to tvos 14.2 Beta. 
HomeKit works more stabile concerning time triggered automations but still the same behaviour when starting an automation which checks the state of the garage door.

Pixel-Chris, 2020-10-22:
I have the same problem, but only with the garage door.
But when I click on „test this automation“ it works.
I configured an other Shelly to detached and use the double press function for my automation.
If garaged door is closed
  open it
Else
 close it.

„Test Automation“ works. 

The double click is recognised in HomeKit (I can see it), but the automation is not started !?

andyblac, 2020-10-22:
are you doing it with "Convert to shortcut"?

andyblac, 2020-10-22:
i have same issue y Alarm system, it is not a firmware issue, it is a HomeKit issue, try seting up with the a 3rd party app and 2 Events.

see here.
https://www.reddit.com/r/HomeKit/comments/jf9yrt/stateless_buttons_and_security_state/

Pixel-Chris, 2020-10-22:
Ok I will test it later! Thanks for that information

andyblac, 2020-10-22:
i also use "a convert to shortcut" with IF's for my shades, that one works no problem, it appears some event triggers in HomeKit are not working correctly

andyblac, 2020-10-22:
just for info:
[image]
[image]

mac-alx, 2020-10-22:
The idea of Andyblac got me to get it work!
First I made an automation in the eve app.

trigger: timestamp e.g. 20:00
condition: garage door is open
scene: any scene, does not matter which one

The switch to the HomeKit app and open the automation created in the eve app
Now you can see the if-statement "20:00 only if garage door is open"

Go into the action menu and deactivate the scene and active "convert to shortcut".
Here I inserted the url  of a pushcut notification (URL function with the pushcut adress) and the function "get contents of url".

Now as the if-rule is no more in the shortcut but in the trigger-rule of the automation everything works and I get a notification at 20:00 if the garage door is open.

Thanks a lot for the idea!

andyblac, 2020-10-30:
i'll close this now, reopen if you think you still have an issue.

closed 2020-10-30 by andyblac