mongoose-os-apps/shelly-homekit #210: Flash script does not work

Drugantix, 2020-10-20:
i get this when i run the script :

alessio@MBP-di-Alessio tools % ./flash_shelly.py -a        
Failed to lookup version information
alessio@MBP-di-Alessio tools % ./flash_shelly.py -la
Failed to lookup version information
alessio@MBP-di-Alessio tools %

andyblac, 2020-10-20:
same issue reported here https://github.com/mongoose-os-apps/shelly-homekit/issues/149#issuecomment-709910514 please try this command.

andyblac, 2020-10-20:
i'll add script to wiki and write it up thx, please report if it does indeed fix your issue.

Drugantix, 2020-10-20:
I solved it in the way you suggested but I used Install Certificates.command present in the installation package of python3. Thank you.

closed 2020-10-20 by andyblac