mongoose-os-apps/shelly-homekit #214: WC - Automations with Time Schedule

noXi0815, 2020-10-22:
Hi,

Thank you for this great project. I have updated from 2.2.0 to 2.3.1. Now my Roller Shutters arenˋt closing in automations with time schedules. When I set an automation for 8pm for complete closing it closes just 1% (sometimes 0% and sometimes it is closing completly). Itˋs only in homekit time automations and only for full closing. When i close the roller shutter manual it is working fine.

best regards

rojer, 2020-10-22:
please enable debug log in system settings, save it immediately after the automation fails and post here

noXi0815, 2020-10-22:
60956857 mgos_sys_config.c:232   Loading conf2.json
60980614 mgos_sys_config.c:232   Loading conf3.json
61116794 mgos_sys_config.c:174   Saved to conf9.json
61141671 HAPPlatformTCPStrea:274 0x3fff8964 192.168.188.60:53729 Accepted HAP connection, ns 3/5/9 ts 0x3fff6d44
62401410 HAPPlatformTCPStrea:104 0x3fff8964 192.168.188.60:53729 HAP connection closed, ns 3/5/9 i 2
62413322 HAPPlatformTCPStrea:301 HAPPlatformTCPStreamClose ts 0x3fff6d44 nc 0x0 3/4/9
62458583 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.188.62:51674
63985560 HAPPlatformTCPStrea:120 0x3fff688c 192.168.188.65:51993 Incoming HAP connection, ns 3/4/9
64201230 HAPPlatformTCPStrea:274 0x3fff5bfc 192.168.188.65:51989 Accepted HAP connection, ns 3/5/9 ts 0x3fff6914
65437739 HAPPlatformTCPStrea:104 0x3fff5bfc 192.168.188.65:51989 HAP connection closed, ns 3/5/9 i 2
65450696 HAPPlatformTCPStrea:301 HAPPlatformTCPStreamClose ts 0x3fff6914 nc 0x0 3/4/9
66470884 shelly_main.cpp:482     Up 66.46, HAP 3/4/9 ns 4, RAM: 16280/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
67236654 HAPPlatformTCPStrea:274 0x3fff6574 192.168.188.60:53730 Accepted HAP connection, ns 2/5/9 ts 0x3fff6914
68470315 HAPPlatformTCPStrea:104 0x3fff6574 192.168.188.60:53730 HAP connection closed, ns 2/5/9 i 2
68483241 HAPPlatformTCPStrea:301 HAPPlatformTCPStreamClose ts 0x3fff6914 nc 0x0 2/4/9
70269801 HAPPlatformTCPStrea:274 0x3fff71fc 192.168.188.65:51991 Accepted HAP connection, ns 1/5/9 ts 0x3fff6914
71497876 HAPPlatformTCPStrea:104 0x3fff71fc 192.168.188.65:51991 HAP connection closed, ns 1/5/9 i 2
71510796 HAPPlatformTCPStrea:301 HAPPlatformTCPStreamClose ts 0x3fff6914 nc 0x0 1/4/9
71880072 HAPPlatformTCPStrea:120 0x3fff6c9c 192.168.188.65:51994 Incoming HAP connection, ns 1/4/9
73296790 HAPPlatformTCPStrea:274 0x3fff688c 192.168.188.65:51993 Accepted HAP connection, ns 1/5/9 ts 0x3fff6914
74507780 HAPPlatformTCPStrea:104 0x3fff688c 192.168.188.65:51993 HAP connection closed, ns 1/5/9 i 2
74519689 HAPPlatformTCPStrea:301 HAPPlatformTCPStreamClose ts 0x3fff6914 nc 0x0 1/4/9
76337188 HAPPlatformTCPStrea:274 0x3fff6c9c 192.168.188.65:51994 Accepted HAP connection, ns 0/5/9 ts 0x3fff67d4
77545919 HAPPlatformTCPStrea:104 0x3fff6c9c 192.168.188.65:51994 HAP connection closed, ns 0/5/9 i 2
77558845 HAPPlatformTCPStrea:301 HAPPlatformTCPStreamClose ts 0x3fff67d4 nc 0x0 0/4/9
78583112 shelly_main.cpp:482     Up 78.57, HAP 0/4/9 ns 4, RAM: 17056/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
78729499 HAPPlatformTCPStrea:120 0x3fff688c 192.168.188.60:53734 Incoming HAP connection, ns 0/4/9
79421675 HAPPlatformTCPStrea:274 0x3fff688c 192.168.188.60:53734 Accepted HAP connection, ns 0/5/9 ts 0x3fff5e9c
84121930 HAPPlatformTCPStrea:120 0x3fff5bfc 192.168.188.65:51995 Incoming HAP connection, ns 0/5/9
84137280 HAPPlatformTCPStrea:274 0x3fff5bfc 192.168.188.65:51995 Accepted HAP connection, ns 0/6/9 ts 0x3fff6914
87624165 shelly_main.cpp:482     Up 87.61, HAP 0/6/9 ns 6, RAM: 17128/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
93105457 mgos_http_server.c:187  0x3fff87a4 GET /
93141903 mgos_http_server.c:180  0x3fff6804 HTTP connection from 192.168.188.62:51695
93161228 mgos_http_server.c:180  0x3fff7244 HTTP connection from 192.168.188.62:51696
93173844 mgos_http_server.c:187  0x3fff87a4 GET /axios.min.js.gz
93195812 mgos_http_server.c:187  0x3fff6804 GET /logo.png
93211575 mgos_http_server.c:187  0x3fff7244 GET /axios.min.js.gz
93240432 mgos_http_server.c:180  0x3fff7c84 HTTP connection from 192.168.188.62:51697
93252284 mgos_http_server.c:187  0x3fff7c84 GET /logo.png
93298346 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.188.62:51674
95624029 shelly_main.cpp:482     Up 95.61, HAP 0/6/9 ns 6, RAM: 16444/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
103633828 shelly_main.cpp:482     Up 103.61, HAP 0/6/9 ns 6, RAM: 17796/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
111624094 shelly_main.cpp:482     Up 111.61, HAP 0/6/9 ns 6, RAM: 17096/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
119624161 shelly_main.cpp:482     Up 119.61, HAP 0/6/9 ns 6, RAM: 17096/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
127624057 shelly_main.cpp:482     Up 127.61, HAP 0/6/9 ns 6, RAM: 18588/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
134144307 mgos_http_server.c:180  0x3fff6d9c HTTP connection from 192.168.188.62:51700
135634035 shelly_main.cpp:482     Up 135.61, HAP 0/6/9 ns 6, RAM: 19444/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
140936950 mgos_http_server.c:180  0x3fff6804 HTTP connection from 192.168.188.62:51701
143634888 shelly_main.cpp:482     Up 143.61, HAP 0/6/9 ns 6, RAM: 19252/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
146607639 mgos_http_server.c:180  0x3fff6804 HTTP connection from 192.168.188.62:51702
151633921 shelly_main.cpp:482     Up 151.61, HAP 0/6/9 ns 6, RAM: 19444/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
156552743 mgos_http_server.c:180  0x3fff6d9c HTTP connection from 192.168.188.62:51703
159634080 shelly_main.cpp:482     Up 159.61, HAP 0/6/9 ns 6, RAM: 19444/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
159659970 mgos_http_server.c:180  0x3fff6d9c HTTP connection from 192.168.188.62:51704
163959912 mgos_http_server.c:180  0x3fff6d9c HTTP connection from 192.168.188.62:51705
167633852 shelly_main.cpp:482     Up 167.61, HAP 0/6/9 ns 6, RAM: 19252/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:100.00
172446803 mgos_http_server.c:180  0x3fff6d9c HTTP connection from 192.168.188.62:51706
[... cut ...]
215073758 mgos_sys_config.c:232   Loading conf3.json
215210059 mgos_sys_config.c:174   Saved to conf9.json
215228350 shelly_hap_window_c:305 WC 1: State: stop -> stopping (24 -> 25)
215244558 shelly_hap_window_c:305 WC 1: State: stopping -> idle (25 -> 0)
215624294 shelly_main.cpp:482     Up 215.61, HAP 0/6/9 ns 6, RAM: 18760/38712; 4.1: c:1 mp:75.07 mt_ms:22390 cp:98.31
216028162 mgos_http_server.c:180  0x3fff6804 HTTP connection from 192.168.188.62:51714
220040673 mgos_http_server.c:180  0x3fff6804 HTTP connection from 192.168.188.62:51715

Rappelsack, 2020-10-22:
Same here. 
They go down 1% (15min after sunset). Sometimes they do nothing.

Rappelsack, 2020-10-23:
Any news here?

Acemach1ne, 2020-10-23:
At the moment I'm back to 2.2.1 and all works great. Waiting for a fix now 😀

rojer, 2020-10-23:
```
214342008 shelly_hap_window_c:325 WC 1: Tgt pos 100.00 -> 0.00 (HAP)
...
214820810 shelly_hap_window_c:325 WC 1: Tgt pos 0.00 -> 98.20 (HAPalt)
```

ok, what i see here is that we get two independent commands to close, one and then 0.5 seconds later another.
because of [this](https://github.com/mongoose-os-apps/shelly-homekit/commit/67c4c2c6e59c4b28285ceefc331770e2a159d718) the change made in response to #181 and introduced in 2.3.0.
hm. i wonder what we should do. #181 makes sense, but then we have this thing... i need to think.

noXi0815, 2020-10-23:
Is this "open stop close stop" (#181) thing on the tile click, that what homekit is expecting? Before this i have used homebridge with original shelly firmware and there was the behavior "open close open close". (Isn't there a specification for that from apple?)

Rappelsack, 2020-10-23:
With 2.3.0 everything works fine I guess

rojer, 2020-10-23:
alright, i think i have a solution that will keep the behavior introduced for #181 but also allow automations to work.
please try https://rojer.me/files/shelly/2.4.0-beta3/ and let me know how it goes.

rojer, 2020-10-23:
@Rappelsack @Acemach1ne please try https://rojer.me/files/shelly/2.4.0-beta3/

Rappelsack, 2020-10-23:
@rojer it works! 😊

Rappelsack, 2020-10-23:
Is it recommended to install on all my Shelly 2.5 the beta that all my automation will work again?

rojer, 2020-10-23:
yeah, should be fine. this will become 2.4 very soon, all indications are good so far.

Rappelsack, 2020-10-23:
Awesome job! Many thanks!!

andyblac, 2021-01-25:
can this be closed ??

closed 2021-03-03 by andyblac