mongoose-os-apps/shelly-homekit #217: Shelly-HomeKit and REST API

socialsubs, 2020-10-23:
Hi,
I would like to know if there is a way to change the state of a Shelly1-HomeKit switch via an http command just like possible with the original firmware.
I have a Shelly Door/Window sensor and I would like to control the lighting of a lamp when the sensor detects the opening of the front door.

Thank you very much for you fantastic job,

rojer, 2020-10-23:
yes. the firmware uses [Mongoose OS RPC mechanism](https://mongoose-os.com/docs/mongoose-os/userguide/rpc.md) and there is REST API for that (along with UART, WebSocket, MQTT and others).
for HTTP, here's an example:

```
$ curl http://shelly1-test.local/rpc/Shelly.SetSwitch -d '{"id": 1, "state": true}'
null
```

id is id of the switch (1 or 2 for devices with 2 switches). `null` means success, in case of error you'll get an error code and message:

```
[rojer@nbd ~/allterco/shelly-homekit master]$ curl http://shelly1-test.local/rpc/Shelly.SetSwitch -d '{"id": 123, "state": true}'
{"code":400,"message":"component not found"}
```

socialsubs, 2020-10-23:
Thank you rojer, 
but its not clear to me which string I should put in the Shelly app.
i.e. if the IP of my shelly 1pm is 192.168.1.48 and the id is shelly1pm-C123AB, what string i need to put in the shelly app?

rojer, 2020-10-23:
192.168.1.48 should work.

andyblac, 2020-10-23:
I have added this info to Wiki, as i think it could also benifit others https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#api

socialsubs, 2020-10-23:
To better clarify...
With the original firmware I can control the switching of the lights through a command like this:

http://192.168.1.48/relay/0?turn=on

I can put this command directly into the "Actions" section of the  Door/Windows sensor settings.

Now, how can I format and send the http command you suggest without spaces and quotes?

rojer, 2020-10-23:
no, not yet. we only accept POST requests for RPCs with arguments, and it'd have to be a GET request.
we should have something like this, but not yet.

andyblac, 2020-10-23:
@socialsubs what your after is this issue https://github.com/mongoose-os-apps/shelly-homekit/issues/70, add your support to it, i think this one can be closed

socialsubs, 2020-10-23:
Ok, now it is clear...

Thanks!

rojer, 2020-10-23:
implementing GET-accessible API is easier than implementing web hooks (actions on other devices), i think these should be separate. let's keep this one for that purpose.

andyblac, 2020-10-23:
ok.

rojer, 2020-10-27:
done. with https://github.com/mongoose-os-libs/rpc-common/commit/31045e6a7248c75d838fd76632a32b89416a1c40 the following two callas are equivalent:

```
$ curl http://shelly1-test.local/rpc/Shelly.SetSwitch -d '{"id": 1, "state": true}' 
$ curl 'http://shelly1-test.local/rpc/Shelly.SetSwitch?id=1&state=true'
```

will ship in the next release.

closed 2020-10-27 by rojer

andyblac, 2020-10-27:
updated Wiki API.

socialsubs, 2020-10-28:
Wow, great news rojer!!

Thank you very much.

ggallegosm, 2020-12-17:
Great post. I couldn't find it in the Wiki API, but here I am able now to have homekit and also http commands at the same time.

blaxxor, 2020-12-21:
> To better clarify...
> With the original firmware I can control the switching of the lights through a command like this:
>
> http://192.168.1.48/relay/0?turn=on
>
> I can put this command directly into the [...]

I'm having trouble understanding the Wiki regarding http-commands.
The original firmware supported switching the relay trough http-command and since installing the mongoose-os on my Shelly 1 that stopped working. And I'm, really missing it :) 

It seems as if there should be support for it now but I cant get it to work.

I'm using a Shelly 1 as a garagedoor opener and want to be able to toggle the switch trough a http-command.
(I'm using a Wifi-switch which sends a http command to open the garagedoor from inside my house) 
I don't know how the http url should look like since the old url doesnt work after updating firmware. (http://192.168.xxx.xxx/relay/0?turn=on)

andyblac, 2020-12-21:
are you on 2.7.0beta? if so API has chnaged you need to use the updated url cmd

https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#API

blaxxor, 2020-12-21:
> are you on 2.7.0beta? if so API has changed you need to use the updated url cmd
> 
> https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#API

Seems to be using 2.6.1 and there is no available update when using the "Check" button on the device.
If i update to the beta. Will I be able to toggle the switch trough a http-command as with the stock firmware?

If so, where do I get the beta? :)

andyblac, 2020-12-21:
> Seems to be using 2.6.1 and there is no available update when using the "Check" button on the device.
> If i update to the beta. Will I be able to toggle the switch trough a http-command as with the [...]

here https://github.com/mongoose-os-apps/shelly-homekit/issues/330

blaxxor, 2020-12-21:
> > > are you on 2.7.0beta? if so API has changed you need to use the updated url cmd
> > > https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#API
> > 
> > 
> > Seems to be using 2.6.1 and there is no available update when using the "Check" button on the device.
> > If i update to the beta. Will I be able to toggle the switch trough a http-command as with the stock firmware?
> > If so, where do I get the beta? :)
> 
> yes, here #330

Thanks for your quick help. I will give it a try :)

blaxxor, 2020-12-21:
I might be completely dumb here but I can't get it working calling a http command trough my browser.

I updated to 2.7.0 beta 2 without problem. I checked the wiki how the url should look like but I can't get it to work.

Do you have an example how the url should look like for it to work?
Stock Shelly FW for calling trough a browser looks like this: http://192.168.x.x/relay/0?turn=on

I kinda need a similar solution for my wifi-switch to work. I can't call a script or similar in the switch. It only works with a simple url.

Regards

andyblac, 2020-12-21:
> I might be completely dumb here but I can't get it working calling a http command trough my browser.
>
> I updated to 2.7.0 beta 2 without problem. I checked the wiki how the url should look like but I [...]

i don't have a roller setup, but for normal switch mode, it would be like this
`http://shelly1-test.local/rpc/Shelly.SetState?id=1&type=0&state=%7b%22state%22%3atrue%7d`

ID=1  (for relay1)
TYPE=0 (for switch)

blaxxor, 2020-12-21:
I only get: {"code":-103,"message":"component not found"} when trying.
Am I missing some dependencies? 

It's a Shelly 1 set to "Garage Door Opener". I don't know if the TYPE=0 is right when it's a garagedoor or not.
I tried changing to 1 and 2 but unfortunately it makes no difference.

andyblac, 2020-12-21:
> I only get: {"code":-103,"message":"component not found"} when trying.
> Am I missing some dependencies?
> 
> It's a Shelly 1 set to "Garage Door Opener". I don't know if the TYPE=0 is right when it's a garagedoor or not.
> I tried changing to 1 and 2 but unfortunately it makes no difference.

garage door is type 5 see here https://github.com/mongoose-os-apps/shelly-homekit/blob/16fc1fcf444a6e69437d7f18a48ff119536fbca5/src/shelly_component.hpp#L26

blaxxor, 2020-12-21:
Got "null" as response this time but nothing happens on the door unfortunately.

Debug log only says:
178666988 mg_rpc.c:305            Shelly.SetState via HTTP 192.168.100.20:55614

Tried reboot and still nothing. The log says the same thing.

Du you know if the state should be "true" even tough its a garagedoor? The GUI says "Toggle" to open the door.

andyblac, 2020-12-21:
> Got "null" as response this time but nothing happens on the door unfortunately.
>
> Debug log only says:
> 178666988 mg_rpc.c:305 Shelly.SetState via HTTP 192.168.100.20:55614
>
> Tried reboot and still [...]

not sure on state, you could try "open" or "close" maybe. @rojer will have to confim usage. but "null" mean command was recieved and actioned correctly.

rojer, 2020-12-21:
null means successful completion, yes.
@blaxxor give me the exact command you're sending.

andyblac, 2020-12-21:
@rojer maybe garage door 'toggle' he should be using "input toggle" as he is not using a shelly 2.5 like
`http://shelly1-test.local/rpc/Shelly.SetState?id=1&event=1`

blaxxor, 2020-12-21:
@rojer 
http://192.168.100.25/rpc/Shelly.SetState?id=1&type=5&state=%7b%22state%22%3atrue%7d

Shelly 1.
2.7.0-beta2 (build 20201219-023230/2.6.1-32-g841aab0-master)

rojer, 2020-12-21:
garage door opener's SetState method handler takes a boolean "toggle" parameter ([here](https://github.com/mongoose-os-apps/shelly-homekit/blob/e80de90b4f81a7a698b3092a1c4a16c4e8443f1e/src/shelly_hap_garage_door_opener.cpp#L179)), try `http://192.168.100.25/rpc/Shelly.SetState?id=1&type=5&state=%7b%22toggle%22%3atrue%7d`

andyblac, 2020-12-21:
> garage door opener's SetState method handler takes a boolean "toggle" parameter [...]

i'll add it to the wiki

blaxxor, 2020-12-22:
@rojer @andyblac 

It works like a charm! Thanks for the help. Both of you. Really appreciate it! 👍 

Awesome support

grizz479, 2021-05-26:
Hi there,
What about roller shutter http commands.
Giving it a several tries, without success... Any tips?

Thanks in advance!

Spy-1, 2022-11-16:
Can someone please give me assistance with an http request?
The following request works:
192.132.198.112/rpc/Shelly.SetState?id=1&type=0&state={%22state%22%3atrue}

However, if a password is set in the interface, I can't get the request to work.
How and where do I set the password in the string above?