mongoose-os-apps/shelly-homekit #218: Shelly 1 Garage Door status blocked

Draf76, 2020-10-23:
Hi, I have a quick question. When I setup a Shelly 1 as garage door opener, the status is after closing always showing up as “stopped” on the website and in HomeKit with exclamation mark instead of closed. There is nothing attached to the Shelly by now. What am I doing wrong?

rojer, 2020-10-23:
this is "obstruction detected" condition.
garage door opener requires a "fully closed" contact sensor. because nothing is attached to your shelly yet, we don't get the fully closed confirmation after 15 seconds (movement time) and assume that door could not be fully closed.

Draf76, 2020-10-23:
What sensor would you recommend for this? Would it have to be connected to the Shelly then?

rojer, 2020-10-23:
See wiring diagram here - https://github.com/mongoose-os-apps/shelly-homekit/wiki/Garage-Door-Opener-Settings

Anything that activates when the door is closed will do. "Activates" could be open or close the circuit, both are supported and can be selected with "close sensor type" option.

note that i do not own a garage door opener myself, but others have posted suggestions on https://github.com/mongoose-os-apps/shelly-homekit/issues/142.

closed 2020-10-23 by rojer