mongoose-os-apps/shelly-homekit #219: Shutter Calibration not possible anymore

housy-CH, 2020-10-24:
First a big thank you goes out to the great homekit integration done from Rojes crew. THX.
I want to show you an issue i have since few days. I can tell you what this issue was triggered from, because the Shelly with your firmware was working until yesterday evening. 
Yesterday the shelly in shutter mode wasnst responding at all. I have had to shutdown the power completly and reset it by pressing the button 5 times. Next i have reverted the shelly to stock FW. With stock FW i was able to controll the shutter without any problem. 
Afterwards i have updated it again to you FW 2.3.1. This went fine and the shelly is working fine in switch mode. But If i set the shelly in shutter mode and press on calibrate then the shutter starts to move up and few seconds later i always get an network error in my browser and the shelly is gone. This behaves like that everytime i try to calibrate the shutter. If you want i can send you the full debug log?
The issue appeared yester evening for the first time, even i didnt changed anything in my setup. Have tried with FM 2.3.0 but the issue occurs in the same way...

12690469 shelly_hap_window_c:300 WC 1: State: idle -> precal0 (0 -> 10)
12788537 shelly_hap_window_c:398 Begin calibration
12818650 mgos_sys_config.c:232   Loading conf2.json
12841082 mgos_sys_config.c:232   Loading conf3.json
12971749 mgos_sys_config.c:174   Saved to conf9.json
12986489 shelly_output.cpp:56    Output 1: off -> on (precal0)
12996571 shelly_hap_window_c:300 WC 1: State: precal0 -> cal0 (10 -> 11)
452116 mgos_rpc_channel_mq:203 0x3fff4054 shellyswitch25-FB1662/rpc
465517 mgos_rpc_channel_ua:313 0x3fff4ae4 UART0
487141 mgos_ade7953.c:63       ADE7953 reg read error (0x702)
496340 mgos_ade7953.c:126      Failed to communicate with ADE7953
504638 shelly_pm_ade7953.cp:97 Failed to init ADE7953
513128 shelly_main.cpp:304     === Creating accessories
525558 shelly_hap_window_c:170 WC 1: not calibrated
535247 shelly_main.cpp:319     === Accessory not provisioned
548069 mgos_init.c:33          Init done, RAM: 38712 total, 26684 free, 26696 min free
560371 mgos_mongoose.c:66      New heap free LWM: 26432
574263 mgos_ota_core.c:1335    UID: 036059ecdaeb4a0d, license: none
585046 mgos_mdns.c:82          Listening on udp://:5353
599952 mgos_mongoose.c:66      New heap free LWM: 25168
609528 mgos_mdns.c:55          mDNS socket closed

housy-CH, 2020-10-24:
Furter information: I have noticed, if i revert back to stock Firmware, the GUI doesnt show any power consumption, during the moving of the shutter. It does always show 0W. Does this mean that there is something broken with the shelly?

rojer, 2020-10-24:
lack of power consumption readings explains calibration failure, but how to explain lack of power readings - i don't know.
could be a hardware issue, especially since stock fw reports 0 as well.

rojer, 2020-10-25:
i am going to close this one on the assumption that the problem is not with the firmware.
feel free to reopen if you think it is.

closed 2020-10-25 by rojer