mongoose-os-apps/shelly-homekit #220: Request: Inverted (Reversed) SW input

jsiegenthaler, 2020-10-24:
The Shelly1 has the ability to set the SW input type to **Reverse inputs** (which is simply inverted input state)

The Shelly1 with mongoose has only Input Mode = Momentary, Toggle, Edge and Detached.
It does not have Reverse (Inverted)

I have a situation where one bathroom has induced voltage on the wiring due to a switchboard right behind the wiring.
I managed to reduce the induced voltage, but it resulted in the input being inverted (reversed).

Not a problem on a Shelly1 with normal Shelly firmware, but it is a problem for the Mongoose OS.

Any chance of adding a Reversed Input Mode (you could also call it Inverted)?
Logically, it would be very simple to implement...

rojer, 2020-10-24:
sounds reasonable. to make sure i understand correctly: your input is normally conductive and interrupts teh circuit when activated (button pressed)?

jsiegenthaler, 2020-10-24:
Hi @rojer 
I mean this:
Normal mode:   SW ON = Relay ON, SW OFF = Relay OFF
Reversed mode:   SW OFF = Relay ON, SW ON = Relay OFF

rojer, 2020-10-24:
yeah, ok. it's definitely doable.

lucaspinelli85, 2020-11-16:
I need this inverted function too!

lucaspinelli85, 2020-11-16:
> Hi @rojer
> I mean this:
> Normal mode: SW ON = Relay ON, SW OFF = Relay OFF
> Reversed mode: SW OFF = Relay ON, SW ON = Relay OFF

Can it be add? Thanks

lucaspinelli85, 2020-11-17:
Solution?
If you want a reverse mode you have to choose if you want swipe input or output or both (like shelly 2.5!)
Could you add this shelly 2.5 feature in shelly 1/1PM?
[image]

lucaspinelli85, 2020-11-17:
The result is this, i take like exemple a sensor nc but could be everything....
[image]

lucaspinelli85, 2020-11-18:
what do you say @rojer ? This needs for normally closed input working reverse like a sensor alarm and normally closed output working reverse like my irrication central that swithed off at close circuit and switched on at open circuit!

> The result is this, i take like exemple a sensor nc but could be everything....
> [image]

rojer, 2020-11-18:
yes, it makes sense, we should do it.

rojer, 2020-11-25:
@lucaspinelli85 ok, so actually what you are asking is different, you are asking for a feature to report switch as off when it's actually on (relay is on), as well as to have ability to invert the input (normally open / normally close).
this also has nothing to do with what "swap inputs" and "swap outputs" does on shelly 2 and 2.5: in that case, swapping inputs means use SW1 pin to control (logical) switch 2 and vice versa, and swapping outputs means logical switch 1 will control O2 instead of O1 and vice versa.
so, i am going to implement the inverting input part, for inverting output state reported to homekit please file a separate issue.

rojer, 2020-11-25:
please help test 2.6.0-beta - https://github.com/mongoose-os-apps/shelly-homekit/issues/291

closed 2020-11-25 by rojer