mongoose-os-apps/shelly-homekit #222: Bug: Web GUI link for System Settings (?) is broken, plus inconsistent Title for WiFi Settings

jsiegenthaler, 2020-10-25:
**Broken Link**
The System Settings help button (?) on the Web GUI for firmware v2.3.1 goes to 
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Firmware
This page is empty.

It should be pointing to
https://github.com/mongoose-os-apps/shelly-homekit/wiki/System-Settings

**Wiki Page Inconsistent Title**
All (?) help links for the Web GUI point to a page named the same Title as the setting they refer to, **except** for this one:
Shelly GUI: **WiFi Settings**
Wiki Title: **WiFi Setup**
https://github.com/mongoose-os-apps/shelly-homekit/wiki/WiFi-setup

A small consistency error, but the sort of thing I see immediately....

rojer, 2020-10-25:
thanks for reporting, this one is already fixed on master by https://github.com/mongoose-os-apps/shelly-homekit/commit/38052505c919010c19fcacc2b30edeee3d836178

closed 2020-10-25 by rojer

jsiegenthaler, 2020-10-25:
@rojer: the second link is not fixed yet. 
https://github.com/mongoose-os-apps/shelly-homekit/wiki/WiFi-setup
should be renamed to 
https://github.com/mongoose-os-apps/shelly-homekit/wiki/WiFi-Settings

The existing WiFi-setup Wiki page should be renamed WiFi-Settings 
A new Wiki page WiFi-setup should be made directing the user to Wifi-Settings for backwards compatibility to existing firmware in the field. Text could be something like "This page has moved to [[WiFi Settings]]"

andyblac, 2020-10-25:
ok, done, @rojer can you update webUI to point to "https://github.com/mongoose-os-apps/shelly-homekit/wiki/WiFi-Settings", one done ill remove temp page.

rojer, 2020-10-25:
done