mongoose-os-apps/shelly-homekit #223: Small Typos in Wiki: I can fix them if you're OK with that

jsiegenthaler, 2020-10-25:
Hello Great Mongoose Team

This is a fantastic project and I'd like to help in any small way I can.

I have started doing so by correcting very small English typos in the Wiki, hope you don't mind.
I am an electronics engineer with also many decades of programming experience and now roking as a database admin.
I have a brain that sees each and every error very quickly, and a desire to have everything correct. Some people call me a pedantic b*§tard.
But I'm really good at testing.

I already identified and fixed a couple of small Wiki Typos just now, hope I'm not stepping on your toes.
https://github.com/mongoose-os-apps/shelly-homekit/wiki
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Setup
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery

andyblac, 2020-10-25:
sure, more help the better 😄

andyblac, 2020-10-25:
close the tickets, when your done.

rojer, 2020-10-25:
yes, we'll take all the help we can get.
welcome!

closed 2020-10-25 by rojer

jsiegenthaler, 2020-10-25:
Done, fixed a whole heap of typos