mongoose-os-apps/shelly-homekit #224: Wiki Page System Settings does not reflect current firmware 2.3.1

jsiegenthaler, 2020-10-25:
Hi All
This Wiii page...
https://github.com/mongoose-os-apps/shelly-homekit/wiki/System-Settings
...has an old screenshot and old text

It shows Model: for the first line in the screenshot, but the current firmware shows Device: as the first line

The screenshot and text need updating to match the current firmware version 2.3.1

I can help with text, but the screenshot came from some browser that looks different to mine. For consistency, the replacement screenshot should come from the same browser as the existing screenshot.

andyblac, 2020-10-25:
Already fixed for next released, but thanks https://github.com/mongoose-os-apps/shelly-homekit/commit/38052505c919010c19fcacc2b30edeee3d836178

andyblac, 2020-10-25:
image updated.

closed 2020-10-25 by andyblac

jsiegenthaler, 2020-10-25:
Text done

andyblac, 2020-10-25:
if you find anymore images that get out of sync give me a shout, i'll be happy to fix. This project is moving fast and sometime hard to keep up with all the changes.