mongoose-os-apps/shelly-homekit #226: Only one tile for shelly 25 possible?

konagar, 2020-10-25:
Hello,
i use several shelly 25 for roller shutters. Unfortunately, I cannot use the roller shutter mode, as my roller shutters (Velux) are operated with 24 volts DC and no current measurement is possible here.

So I just use switches for open and closed. These were in older firmware versions e.g. 2.1.0 shown with a tile under which 2 switches were hidden. In newer versions from e.g. 2.1.5 or newer I unfortunately always get 2 tiles, one for each switch. Can I also get the display with just one tile in newer versions?

Thank you for your great software

Renje

rojer, 2020-10-25:
this was an intentional change to enable assigning switches to different rooms.
previously they were services under the same accessory, now they are separate accessories with 1 switch each.
we can be pretty flexible here, single/multiple accessories can be configurable.

sebastianheise, 2020-11-10:
yeah, I got caught up with that as well - but I actually don't mind. But I can see how that can get crowded if someone has a larger amount of devices. ;) Might be good to make it a configuration option.

lucaspinelli85, 2020-11-17:
> Hello,
> i use several shelly 25 for roller shutters. Unfortunately, I cannot use the roller shutter mode, as my roller shutters (Velux) are operated with 24 volts DC and no current measurement is [...]

Why don't you use shelly2.5 with 2 relè DPDT? You have the benefits that you have roller shutter mode (with calibration etc..) with 24v motor (or other tension different by shelly).
Pratically shelly2.5 in 220v, output1 in relè1 (220v), output2 in relè2 (220v). Motor cabled powered normal trought relè with - - poles (motor stop), relè1 swap pole 1 in + - (forward), relè2 swap pole 2 in - + (back).
If use DPDT relè you can cable the power 24v to cut off when not using by relè, if use SPDT relè, the power 24v is always on.
Hope you understand me

lucaspinelli85, 2020-11-17:
[image]

lucaspinelli85, 2020-11-17:
[image]

lucaspinelli85, 2020-11-17:
Why don't you use shelly2.5 with 2 relè DPDT? You have the benefits that you have roller shutter mode (with calibration etc..) with 24v motor (or other tension different by shelly).
Pratically shelly2.5 in 220v, output1 in relè1 (220v) and power supply, output2 in relè2 (220v) and power supply. Motor cabled powered normal trought relè with - - poles (motor stop), relè1 swap pole 1 in + - (forward) and powered supply, relè2 swap pole 2 in - + (back) and powered supply. It should work..
[image]

konagar, 2020-11-17:
Hello, thank you for your mail. I understand exactly what you mean. This is really a brilliant solution. Unfortunately I didn't know this one. I'm going to rebuild my control system, I installed it last weekend. I have to get six 230 volt relays for my 3 Velux shutters. Many greetings and thanks again. Can you answer me briefly, I'm not sure whether you will get this message when I press replay in my mail program.
I use google translator because my english is not great. I hope the translation is understandable.

konagar, 2020-11-17:
this is my current solution. Is exactly your picture power supply always on

lucaspinelli85, 2020-11-17:
I delete the previusly schemes because not exactly correct! Because if you want to calibrate it, the power supply have to take energy from output of shelly (not from general line!, the previus schemes work the same only if the stroke to open/close is set by timer, not by calibrate with power measure because at the output there was only the relay under charging, actually in firmware homekit there is only possibility to calibrate and not to set timer)!
Only in this last case shelly can measure power to calibrate stroke of open/close roller shutter because power supply is powered by shelly outputs!
In this way it should work better... I think...
Sorry for my english traslation, i'm from Italy, hope you understand!

konagar, 2020-11-17:
I just see when I reply to the mail it ends up in this thread.

konagar, 2020-11-17:
At the moment I don't use the shutter mode, but 2 switches for open and closed with a timer. But I'll rebuild it as you suggested. I understand your problem with the translation, I have the same problem, I'm from Germany.

lucaspinelli85, 2020-11-17:
If you want use single power supply and velux are near use this scheme, otherwise 3 of previusly!
[image]

konagar, 2020-11-17:
Thank you for your efforts, very nice. I planned to do it with just one power supply. I am currently looking for 2-pole 230V relays that are already assembled on a circuit board, as I now use for my 24 volts. But that doesn't seem to be the case, so I have to buy 6 individual relays.

lucaspinelli85, 2020-11-17:
glad you were helpful, let me know if it works!

konagar, 2020-12-05:
Hello, I have rebuilt everything as in the suggestion from lucaspinelli85. Calibration worked after a few attempts. But I had to install the original Shelly firmware again to set off_power to 5 watts (default is 2 watts). This was necessary because the roller shutter needs about 12-15 watts in operation and drops to 3 watts when it hits the stop and not to 0 watts. Unfortunately, the Velux roller shutter always needs a while to start running when it gets power. The Shelly only waits about a second for power consumption. Unfortunately, this has the consequence that the roller shutters can usually not be controlled. The relay only picks up briefly and then drops out again. Can you set the Shelly to wait longer for power consumption?

konagar, 2020-12-05:
if the Velux motor does not start running immediately when it receives power, the shelly 2.5 switches off after a second, as already mentioned. But in order to get an immediate power consumption, I connected an RC element with a diode for both directions. The solution isn't pretty, but it works.

konagar, 2020-12-06:
If you use one power supply for all three shutters at the same time, you cannot stop a single shutter. Only if you stop all the shutters will they all stop. The relays remain switched on if they still receive voltage from one of the other relays (gray wire).

andyblac, 2021-03-03:
can this be closed ?

konagar, 2021-03-21:
Hello, just saw this topic closed. It would be really great if it were possible to hide a shelly 25 under a tile. With all DC motors, the roller shutter mode cannot easily be used. On my projector screen I have now installed a version 1.1 again to combine the up and down movement under one tile.

rojer, 2021-03-21:
yes, this request makes sense.

konagar, 2021-03-21:
thank you in advance

lokino, 2022-08-23:
@lucaspinelli85 please on your wiring Diagram where GREY and BLUE lines comes from? On my 24V adapter I have output for + and - but not sure where I should wire BLUE and GREY line from