mongoose-os-apps/shelly-homekit #228: Shelly 2.5 Garage Door opening issue

guidau, 2020-10-25:
Hello everybody,

first of all a big compliment and thank you to everyone, especially to Rojer !
I have been following the development of the Shelly HomeKit for a while now. I have a Hoermann Supramatic E garage door operator with sectional door (serial number with B at the end / older than 12 years).
I have been using the garage door with the Shelly1 without magnetic sensors for a few weeks and was very happy about the extension in the new firmware.
What works in any case is the Shelly 1 with a magnetic sensor that signals the closed door.
I also find it very good to show both stats (open/closed), so I have now connected a Shelly 2.5.
Here I have the following phenomenon:
The garage door is closed, I start an opening process via the Home app, the Shelly relay clicks, the Home app shows Garage door is opened, but nothing happens.
I press a second or even third time and the door opens, but the first time I press it, the Home app shows Garage door is closed. If it is completely open, the status is OK and also shows open, now e.g. a single trigger is enough to close the door again.

Basically I connected the Shellys via 24V of the Hoermann Supramatic, I also connected an additional 24V power supply to the Shelly 2.5, without any changes.
Here I have taken the pin assignment from the Wiki. Power via contact 5 (24V+) and 20 (24V-).
The garage door switch via contact 20(-) and 21(+).

Do you have an idea where the problem could be ?

Additionally I have a question, the Hoermann Supramatic E operator has the possibility to open/close/stop the door with one button on the connection 20&21 (as already used with the Shelly).
But there is also the possibility to use a double push-button: door open (connection 15&14) and door closed (connection 17&14). 
I think this makes sense if you are not in the immediate near of the garage door and want to make sure that the door only opens and does not close.
With the Shelly 2.5 it would be possible to trigger two switching operations. Is there a future plan in development ?

Thanks in advance for your efforts.
And the most important thing: Stay healthy :-)

_Originally posted by @guidau in https://github.com/mongoose-os-apps/shelly-homekit/issues/142#issuecomment-716181519_

guidau, 2020-10-25:
Maybe the log helps:
11792156297 mgos_sys_config.c:232   Loading conf2.json
11792179839 mgos_sys_config.c:232   Loading conf3.json
11792312217 mgos_sys_config.c:174   Saved to conf9.json
11792461103 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.xx.xx:51839
11800045610 shelly_main.cpp:482     Up 11800.02, HAP 0/6/9 ns 6, RAM: 20108/38712; 5.1: cur:closed tgt:closed cl:1 op:0
11807500628 shelly_output.cpp:56    Output 2: off -> on (GDO:open)
11807512249 shelly_hap_garage_d:294 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
11807574530 shelly_hap_garage_d:209 GDO 1: Cur State: closed -> opening (1 -> 2)
11807810676 shelly_output.cpp:56    Output 2: on -> off (pulse_off)
11808035503 shelly_main.cpp:482     Up 11808.02, HAP 0/6/9 ns 6, RAM: 16864/38712; 5.1: cur:opening tgt:open cl:1 op:0
11808816235 shelly_output.cpp:56    Output 1: off -> on (GDO:close)
11808827969 shelly_hap_garage_d:294 GDO 1: Tgt State: open -> closed (0 -> 1) (HAPclose)
11808839186 shelly_hap_garage_d:209 GDO 1: Cur State: opening -> stopped (2 -> 4)
11808852011 shelly_hap_garage_d:209 GDO 1: Cur State: stopped -> closed (4 -> 1)
11808906434 mgos_mongoose.c:66      New heap free LWM: 11816
11809126382 shelly_output.cpp:56    Output 1: on -> off (pulse_off)
11809402139 shelly_input.cpp:72     Input 1: change (state 0)
11809443713 shelly_input.cpp:72     Input 1: change (state 1)
11809636481 shelly_input.cpp:72     Input 1: change (state 0)
11809650064 shelly_hap_garage_d:294 GDO 1: Tgt State: closed -> open (1 -> 0) (ext)
11809662250 shelly_hap_garage_d:209 GDO 1: Cur State: closed -> opening (1 -> 2)
11810003387 shelly_input.cpp:72     Input 1: single (state 0)
11816035591 shelly_main.cpp:482     Up 11816.02, HAP 0/6/9 ns 6, RAM: 19876/38712; 5.1: cur:opening tgt:open cl:0 op:0
11824035505 shelly_main.cpp:482     Up 11824.02, HAP 0/6/9 ns 6, RAM: 19852/38712; 5.1: cur:opening tgt:open cl:0 op:0
11829740629 shelly_hap_garage_d:209 GDO 1: Cur State: opening -> open (2 -> 0)
11829777002 shelly_input.cpp:72     Input 2: change (state 1)
11830787922 shelly_input.cpp:72     Input 2: long (state 1)
11832035355 shelly_main.cpp:482     Up 11832.02, HAP 0/6/9 ns 6, RAM: 19852/38712; 5.1: cur:open tgt:open cl:0 op:1
11833652603 shelly_output.cpp:56    Output 1: off -> on (GDO:close)
11833665360 shelly_hap_garage_d:294 GDO 1: Tgt State: open -> closed (0 -> 1) (HAPclose)
11833726137 shelly_hap_garage_d:209 GDO 1: Cur State: open -> closing (0 -> 3)
11833963763 shelly_output.cpp:56    Output 1: on -> off (pulse_off)
11834014768 shelly_input.cpp:72     Input 2: change (state 0)
11840036006 shelly_main.cpp:482     Up 11840.02, HAP 0/6/9 ns 6, RAM: 19844/38712; 5.1: cur:closing tgt:closed cl:0 op:0
11848035942 shelly_main.cpp:482     Up 11848.02, HAP 0/6/9 ns 6, RAM: 19844/38712; 5.1: cur:closing tgt:closed cl:0 op:0
11850183417 shelly_input.cpp:72     Input 1: change (state 1)
11850240637 shelly_hap_garage_d:209 GDO 1: Cur State: closing -> closed (3 -> 1)
11850430041 shelly_input.cpp:72     Input 1: change (state 0)
11850443876 shelly_hap_garage_d:294 GDO 1: Tgt State: closed -> open (1 -> 0) (ext)
11850455039 shelly_hap_garage_d:209 GDO 1: Cur State: closed -> opening (1 -> 2)
11850525900 shelly_input.cpp:72     Input 1: change (state 1)
11851586474 shelly_input.cpp:72     Input 1: long (state 1)
11855540892 shelly_hap_garage_d:294 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
11855552097 shelly_hap_garage_d:209 GDO 1: Cur State: opening -> closed (2 -> 1)
11856035747 shelly_main.cpp:482     Up 11856.02, HAP 0/6/9 ns 6, RAM: 19844/38712; 5.1: cur:closed tgt:closed cl:1 op:0
11860238990 shelly_output.cpp:56    Output 2: off -> on (GDO:open)
11860250664 shelly_hap_garage_d:294 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
11860261826 shelly_hap_garage_d:209 GDO 1: Cur State: closed -> opening (1 -> 2)
11860549148 shelly_output.cpp:56    Output 2: on -> off (pulse_off)
11864035515 shelly_main.cpp:482     Up 11864.02, HAP 0/6/9 ns 6, RAM: 19852/38712; 5.1: cur:opening tgt:open cl:1 op:0
11865340922 shelly_hap_garage_d:294 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
11865353173 shelly_hap_garage_d:209 GDO 1: Cur State: opening -> closed (2 -> 1)
11867237450 shelly_output.cpp:56    Output 2: off -> on (GDO:open)
11867249130 shelly_hap_garage_d:294 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
11867261338 shelly_hap_garage_d:209 GDO 1: Cur State: closed -> opening (1 -> 2)
11867547605 shelly_output.cpp:56    Output 2: on -> off (pulse_off)
11872035483 shelly_main.cpp:482     Up 11872.02, HAP 0/6/9 ns 6, RAM: 19852/38712; 5.1: cur:opening tgt:open cl:1 op:0
11872340888 shelly_hap_garage_d:294 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
11872352104 shelly_hap_garage_d:209 GDO 1: Cur State: opening -> closed (2 -> 1)
11873969225 shelly_output.cpp:56    Output 2: off -> on (GDO:open)
11873981955 shelly_hap_garage_d:294 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
11873993124 shelly_hap_garage_d:209 GDO 1: Cur State: closed -> opening (1 -> 2)
11874280474 shelly_output.cpp:56    Output 2: on -> off (pulse_off)
11879041262 shelly_hap_garage_d:294 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
11879102922 shelly_hap_garage_d:209 GDO 1: Cur State: opening -> closed (2 -> 1)
11880035760 shelly_main.cpp:482     Up 11880.02, HAP 0/6/9 ns 6, RAM: 19844/38712; 5.1: cur:closed tgt:closed cl:1 op:0
11882400270 shelly_output.cpp:56    Output 2: off -> on (GDO:open)
[... cut ...]
11888035702 shelly_main.cpp:482     Up 11888.02, HAP 0/6/9 ns 6, RAM: 19844/38712; 5.1: cur:closed tgt:closed cl:1 op:0
11896035616 shelly_main.cpp:482     Up 11896.02, HAP 0/6/9 ns 6, RAM: 19844/38712; 5.1: cur:closed tgt:closed cl:1 op:0
11904035615 shelly_main.cpp:482     Up 11904.02, HAP 0/6/9 ns 6, RAM: 19844/38712; 5.1: cur:closed tgt:closed cl:1 op:0
11912035935 shelly_main.cpp:482     Up 11912.02, HAP 0/6/9 ns 6, RAM: 19844/38712; 5.1: cur:closed tgt:closed cl:1 op:0
11918413283 mgos_http_server.c:180  0x3fff4784 HTTP connection from 192.168.xx.xx:51847

cskallo, 2020-10-28:
Már egy ideje követem a Shelly HomeKit fejlesztését. 
Nem Vagyok programozó de tudna valaki segíteni valaki nekem hogyan kell beírni Automatikus kikapcsolási késleltetés???
Nemigazán értelmezem.
nem tudok időt beállítani.

guidau, 2020-10-29:
Solved in original issue !

closed 2020-10-29 by guidau