mongoose-os-apps/shelly-homekit #236: Enhancements in logging

pivale, 2020-10-27:
Hey,

I have several issues that are occurring but I would like to be able to find them first in the logs and submit them properly.
The problem is, with two kids around the house, it's hard to run to the logs and find the situation once it occurs.

If we could improve the logging, I think it would make it easier on everyone to identify and report problems.

My suggestion for that would be:
- Change the log timestamp to something human readable, or clarify how to read it
- Allow to configure a local address where all accessories could dump their own log in a continuous away (without rotation).

Is this something we could implement?

marcus-aa, 2020-11-26:
Syslog would be nice....

timoschilling, 2021-05-26:
The timestamps are msec since boot, a remote logging can already be configured: https://mongoose-os.com/docs/mongoose-os/userguide/debug.md#debug-configuration-settings-quick-reference

closed 2021-05-26 by timoschilling