mongoose-os-apps/shelly-homekit #237: Adding Switch, can we use Switch name?

andyblac, 2020-10-27:
When adding a new accessory can we use teh Switch Names from webUI in HomeKit setup (see screenshots), at the moment when we add a new accessory the "Switch Name" is used from the Device ID.

[image]
[image]

emartife, 2021-02-18:
With version 2.7.3 after configuring it you have to reboot, so the correct name appears when it is added to HomeKit

andyblac, 2021-02-18:
in 2.7.3 the switch name is indeed used, but on adding the accessory it show the wrong switch name, it shows "Device Name", but once added it does show the "Switch Name"

rojer, 2021-02-20:
hm. i guess we could se accessory name = service name if service is the only one in within the accessory

andyblac, 2021-02-20:
if adding when in bridge mode, everything works as expected both the hub, accessories are added with all corect names.
[image]
[image]
[image]
[image]
[image]

it only when adding a single accessory it shows wrong name:
[image]

andyblac, 2021-02-20:
it could just be a iOS thing?

rojer, 2021-02-20:
no, it's our thing, will do

rojer, 2021-02-20:
done, will be in alpha2

rojer, 2021-03-15:
had to back this one out due to DNS name conflicts, see description in https://github.com/mongoose-os-apps/shelly-homekit/commit/acb57e02caef82438a2c82da5c67265a8c76fda0

rojer, 2021-03-15:
requires https://github.com/mongoose-os-libs/dns-sd/issues/5

andyblac, 2021-03-15:
> had to back this one out due to DNS name conflicts, see description in [acb57e0](https://github.com/mongoose-os-apps/shelly-homekit/commit/acb57e02caef82438a2c82da5c67265a8c76fda0)

how about using the original `shelly1-xxxxxx` for A RECORD, but adding an CNAME for the devcie name ?, i've asked krill to do same in stock firmware as they need to keep `shelly1-xxxxxxx` as A RECORD due the way there app detection works.

rojer, 2021-03-15:
they will still fight for the same name if there is a clash, resulting in confusion.
the spec outlines what the behavior should be, it just needs to be implemented.