mongoose-os-apps/shelly-homekit #238: Missing "Input Mode" in Shelly Plug

Leo-W536, 2020-10-28:
Hello, 
a big Thanks for the great job integrating shellys in homekit. 

In ShellyPlug I am missing the option to choose Input Mode like shown in wiki:
[https://github.com/mongoose-os-apps/shelly-homekit/wiki/Switch-Settings](url)

Thank you for fixing!
[image]

andyblac, 2020-10-28:
there is no input mode for devices like plugs that have no external inputs

andyblac, 2020-10-28:
I have updated the wiki, to reflect this.

Leo-W536, 2020-10-28:
But the Shelly Plug has an InputButton!

andyblac, 2020-10-28:
ok, then this one is for @rojer, i was going off git comits https://github.com/mongoose-os-apps/shelly-homekit/commit/4bdf776d1667662cbfeb44c19d53924267ad3e54

andyblac, 2020-10-28:
> But the Shelly Plug has an InputButton!

is it not just always the same type though? ie push button ?

rojer, 2020-10-28:
it has a button, but that's not the input the "input mode" section is about.
making the button control the plug is a good idea though.

rojer, 2020-10-28:
yes, it's a push button. single press of external button should toggle output.

andyblac, 2020-10-28:
> it has a button, but that's not the input the "input mode" section is about.
> making the button control the plug is a good idea though.

ye, maybe just force to Momentary ?

rojer, 2020-10-29:
fixed. short press of the physical button on shellies will now toggle the switch state.
on devices with more than one switch configure, it will cycle them through all possible combinations.

closed 2020-10-29 by rojer

Leo-W536, 2020-11-11:
Thank you for fixing! Works perfect for Shelly Plug. 

For Shelly Plug S it must be the same. I tried with FW 2.5.0 but it's not working yet. 

Please fix it for Plug S too! 
Thank you...

rojer, 2020-11-13:
@Leo-W536 done, will be in the next release