mongoose-os-apps/shelly-homekit #239: Ability to turn of temperature monitoring

fmms, 2020-10-28:
Hi,

I have updated to 2.4.0 yesterday and today all my lights are switching off. I think I have identified the issue which seems to be the temperature measurement.

- Can there be any bug in reading the temperature? It showing 87°C though previously they have been working for half a year.
- Is there any chance to disable the temperature shutoff so that I can monitor this more closely?

It is a shelly 2.5.

Thanks

ToblerCGN, 2020-10-28:
Same Problem here. Are the measurements really correct. My Shelly 2.5 has 87 degrees, when the switches are on. 68, when off. And I‘m running the device only for testing, whithout anything attached.

andyblac, 2020-10-28:
> Same Problem here. Are the measurements really correct. My Shelly 2.5 has 87 degrees, when the switches are on. 68, when off. And I‘m running the device only for testing, whithout anything attached.

mine seem ok, but my shellys are in the wall cavity, and not cramed into a backbox, i wanted more airflow for this very reason, i have read on forums in the passed shelly can overheat quite easily when in tight spaces.

[image]

pivale, 2020-10-28:
I have 2 shelly1pm in buttons working at 42-45º
My 5 shelly25 are behind wall switches (shutters) and are working at 62-64º

rojer, 2020-10-28:
default thresholds are 100 and 90 degrees: overheat protection kicks in at 100C and turns back off at 90C.
this is very high temperature, i would advice against changing it, but that said...

```
$ curl http://shelly25-test.local/rpc/Config.Set -d '{"config": {"shelly": {"overheat_on": 105, "overheat_off": 95}}, "save": true, "reboot": false}'
{"saved": true}
```

this will change it to 105/95. adjust hostname as needed (ip address or `device_name.local`).

fmms, 2020-10-28:
Thanks @rojer . I have increased the values right now and will report back.

Two questions:
- will these settings be persisted over reboots?
- is there any way to see all configuration options that are offered by the firmware?

rojer, 2020-10-29:
configuration options are defined in https://github.com/mongoose-os-apps/shelly-homekit/blob/master/mos.yml config_schema section (global and per-device). http://shelly1-test.local/rpc/Config.Get will get you current values for these and other options defined in other libraries.
however, i would advice against making random tweaks with Config.Set, as validation of paramters is not performed. you can brick your device if you are not careful.

rojer, 2020-10-29:
i'm going to close this issue. i'm not comfortable raising the cutoff, as current values are already very high. looks like your device does not get enough airflow to provide proper cooling. i gave you a way to manually override this if you know what you're doing, but ultimately i think everything is working as designed here.

closed 2020-10-29 by rojer