mongoose-os-apps/shelly-homekit #241: Shelly 1 Garage door movement time

peezy45, 2020-10-28:
Would it be possible to implement a toggle in the WebUI to completely disable the movement time for a Shelly 1 in Garage Door Opener mode so it would only show "open" or closed" but no "opening" or "closing" ? 

Thanks and kind regards 
Pascal

rojer, 2020-10-28:
i don't understand. can you explain what the problem is?

peezy45, 2020-10-28:
> i don't understand. can you explain what the problem is?

The problem is that if the gate is open, it shows as "open" in the home app but if I trigger the Shelly to close it it shows "opening" while in fact it's closing. I know it's a cosmetic problem so I'd like to completely disable the "opening" and "closing" status.

rojer, 2020-10-28:
it's not supposed to be doing that, somehow it gets out of sync with reality. can you enable debug log and post it here after such movement?

peezy45, 2020-10-28:
Hey , here's the log:

```

24140591369 mgos_sys_config.c:232   Loading conf2.json
24140614635 mgos_sys_config.c:232   Loading conf3.json
24140625769 mgos_config_util.c:108  Extra key: [relays]
24140634382 mgos_config_util.c:108  Extra key: [relays.r0]
24140644352 mgos_config_util.c:108  Extra key: [relays.r0.default_state]
24140653975 mgos_config_util.c:108  Extra key: [relays.r0]
24140662339 mgos_config_util.c:108  Extra key: [relays]
24140788821 mgos_sys_config.c:174   Saved to conf9.json
24140948523 mg_rpc.c:292            Shelly.GetInfo via HTTP 10.0.3.199:49938
24144090106 shelly_main.cpp:543     Up 24144.08, HAP 0/7/9 ns 7, RAM: 21948/41312; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
24152090101 shelly_main.cpp:543     Up 24152.08, HAP 0/7/9 ns 7, RAM: 22932/41312; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
24160090053 shelly_main.cpp:543     Up 24160.08, HAP 0/7/9 ns 7, RAM: 22932/41312; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
24168090035 shelly_main.cpp:543     Up 24168.08, HAP 0/7/9 ns 7, RAM: 22932/41312; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
24174749903 mgos_http_server.c:180  0x3fff6a9c HTTP connection from 10.0.1.252:52801
24174765842 mgos_http_server.c:180  0x3fff6b24 HTTP connection from 10.0.1.252:52802
24174777128 mgos_http_server.c:180  0x3fff6bac HTTP connection from 10.0.1.252:52803
24174788686 mgos_http_server.c:187  0x3fff6bac GET /
24174812854 mg_http.c:368           '/index.html.gz' '/index.html' 'text/html'
24174878827 mgos_http_server.c:187  0x3fff6a9c GET /axios.min.js.gz
24174930497 mg_http.c:368           '/axios.min.js.gz' '/axios.min.js' 'application/x-javascript'
24174954184 mgos_http_server.c:187  0x3fff6b24 GET /logo.png
24174973805 mgos_mongoose.c:66      New heap free LWM: 12368
24175039207 mgos_mongoose.c:66      New heap free LWM: 11264
24175578157 mgos_http_server.c:180  0x3fff6ba4 HTTP connection from 10.0.1.252:52805
24175589456 mgos_http_server.c:180  0x3fff6c2c HTTP connection from 10.0.1.252:52806
24175606777 mg_rpc.c:292            Shelly.GetInfo via HTTP 10.0.1.252:52805
24175628894 mgos_http_server.c:187  0x3fff6c2c GET /favicon.ico
24176094008 shelly_main.cpp:543     Up 24176.08, HAP 0/7/9 ns 7, RAM: 21440/41312; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1

```

Thank you very much.

rojer, 2020-10-28:
i don't see any door movement here

peezy45, 2020-10-28:
The close sensor is set to "normally open" and has contact when the gate is completely open. I don't know if that information helps somehow.

rojer, 2020-10-28:
> The close sensor is set to "normally open" and has contact when the gate is completely open

ah, that's the key. what you have is actually the open sensor. we expect at least close sensor, open sensor is optional. configuration with only the open sensor is not currently supported. "normally open" is a different thing, it's about whether sensor closes or opens the electric circuit when activated (see help). you need to install a "door fully closed" sensor for firmware to operate correctly.

peezy45, 2020-10-28:
> The close sensor is set to "normally open" and has contact when the gate is completely open. I don't know if that information helps somehow. 

That while opening through home app 

```

24500962329 shelly_output.cpp:56    Output 1: off -> on (GDO:open)
24500973844 shelly_hap_garage_d:294 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
24500984838 shelly_hap_garage_d:209 GDO 1: Cur State: closed -> opening (1 -> 2)
24501272348 shelly_output.cpp:56    Output 1: on -> off (pulse_off)
24504089976 shelly_main.cpp:543     Up 24504.08, HAP 0/7/9 ns 7, RAM: 22904/41312; st 0; 5.1: cur:opening tgt:open cl:1 op:-1
24504722281 mgos_http_server.c:180  0x3fff5c84 HTTP connection from 10.0.1.252:52813
24506055440 shelly_hap_garage_d:294 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
24506066472 shelly_hap_garage_d:209 GDO 1: Cur State: opening -> closed (2 -> 1)
24512089926 shelly_main.cpp:543     Up 24512.08, HAP 0/7/9 ns 7, RAM: 22724/41312; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
24517178674 shelly_input.cpp:72     Input 1: change (state 1)
24517246033 shelly_hap_garage_d:294 GDO 1: Tgt State: closed -> open (1 -> 0) (ext)
24517257073 shelly_hap_garage_d:209 GDO 1: Cur State: closed -> opening (1 -> 2)
24517390868 shelly_input.cpp:72     Input 1: change (state 0)
24517423507 shelly_input.cpp:72     Input 1: change (state 1)
24518027288 shelly_input.cpp:72     Input 1: change (state 0)
24518036260 shelly_input.cpp:72     Input 1: single (state 0)
24518070773 shelly_input.cpp:72     Input 1: change (state 1)
24518334270 shelly_input.cpp:72     Input 1: change (state 0)
24518480039 shelly_input.cpp:72     Input 1: change (state 1)
24518777011 mgos_http_server.c:180  0x3fff5c84 HTTP connection from 10.0.1.252:52814

```

That's while closing through home app

```

4519499477 shelly_input.cpp:72     Input 1: long (state 1)
24520089880 shelly_main.cpp:543     Up 24520.08, HAP 0/7/9 ns 7, RAM: 22724/41312; st 0; 5.1: cur:opening tgt:open cl:0 op:-1
24527345519 shelly_hap_garage_d:209 GDO 1: Cur State: opening -> open (2 -> 0)
24528089707 shelly_main.cpp:543     Up 24528.08, HAP 0/7/9 ns 7, RAM: 22732/41312; st 0; 5.1: cur:open tgt:open cl:0 op:-1
24536089550 shelly_main.cpp:543     Up 24536.08, HAP 0/7/9 ns 7, RAM: 22924/41312; st 0; 5.1: cur:open tgt:open cl:0 op:-1
24544089575 shelly_main.cpp:543     Up 24544.08, HAP 0/7/9 ns 7, RAM: 22924/41312; st 0; 5.1: cur:open tgt:open cl:0 op:-1
24552089595 shelly_main.cpp:543     Up 24552.08, HAP 0/7/9 ns 7, RAM: 22924/41312; st 0; 5.1: cur:open tgt:open cl:0 op:-1
24560089583 shelly_main.cpp:543     Up 24560.08, HAP 0/7/9 ns 7, RAM: 22924/41312; st 0; 5.1: cur:open tgt:open cl:0 op:-1
24568089676 shelly_main.cpp:543     Up 24568.08, HAP 0/7/9 ns 7, RAM: 22948/41312; st 0; 5.1: cur:open tgt:open cl:0 op:-1
24576089638 shelly_main.cpp:543     Up 24576.08, HAP 0/7/9 ns 7, RAM: 22948/41312; st 0; 5.1: cur:open tgt:open cl:0 op:-1
24584089594 shelly_main.cpp:543     Up 24584.08, HAP 0/7/9 ns 7, RAM: 22948/41312; st 0; 5.1: cur:open tgt:open cl:0 op:-1
24592089703 shelly_main.cpp:543     Up 24592.08, HAP 0/7/9 ns 7, RAM: 22948/41312; st 0; 5.1: cur:open tgt:open cl:0 op:-1
24600089583 shelly_main.cpp:543     Up 24600.08, HAP 0/7/9 ns 7, RAM: 22948/41312; st 0; 5.1: cur:open tgt:open cl:0 op:-1
24603446526 shelly_output.cpp:56    Output 1: off -> on (GDO:close)
24603458156 shelly_hap_garage_d:294 GDO 1: Tgt State: open -> closed (0 -> 1) (HAPclose)
24603469006 shelly_hap_garage_d:209 GDO 1: Cur State: open -> closing (0 -> 3)
24603756668 shelly_output.cpp:56    Output 1: on -> off (pulse_off)
24604121464 shelly_input.cpp:72     Input 1: change (state 0)
24604145500 shelly_hap_garage_d:209 GDO 1: Cur State: closing -> closed (3 -> 1)
24608090016 shelly_main.cpp:543     Up 24608.08, HAP 0/7/9 ns 7, RAM: 22916/41312; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
24611649538 mgos_http_server.c:180  0x3fff5c84 HTTP connection from 10.0.1.252:52819

```

peezy45, 2020-10-28:
> ah, that's the key. what you have is actually the open sensor. we expect at least close sensor, open sensor is optional. configuration with only the open sensor is not currently supported. "normally [...]

Okay I see. 
Thank you very much so far. 
Do you have a Amazon link or something with the correct sensor for me ?

rojer, 2020-10-28:
no, not really. some sort of magnetic contact that reliably opens and closes a circuit will do.

peezy45, 2020-10-28:
> no, not really. some sort of magnetic contact that reliably opens and closes a circuit will do.

I've used these. 

https://www.amazon.de/dp/B07Z4NCWDD/ref=cm_sw_r_cp_api_i_-9CMFbHK0KA0B?_encoding=UTF8&psc=1

rojer, 2020-10-28:
seem like they should do, but i haven't used them.

i'll close this bug, for now, if you are still having issues after installing a close sensor, please open a new one.

closed 2020-10-28 by rojer