mongoose-os-apps/shelly-homekit #242: automata kikapcsolás Shelly1

cskallo, 2020-10-28:
Már egy ideje követem a Shelly HomeKit garcía t.
Nem vagyok programozo s nem igazán értem az automata kikapcsolás  idö beállitását
ha 10másodpercet szeretnék azt miép lehet?.

Marcusyo, 2020-10-29:
If google translater worked right you want to know how to set the auto off delay to 10 seconds.
I attached a file where i made the settings for that. you can copy that for yourself

the format for the time is **D:HH:MM:SS.sss**
Day Hour Hour Minute Minute Second Secons millisecond millisecond millisecond

[image]

rojer, 2020-10-29:
@cskallo  we are all from different countries here and official language of the communication is english.
please be respectful of other members and do your part and take the time to express your issue so others can understand.

closed 2020-10-29 by rojer