mongoose-os-apps/shelly-homekit #246: Question: What is defining if a 2.5 appears as a hub?

ursusHU, 2020-11-02:
I’d like to understand what setting makes a 2.5 appear in HomeKit as a hub? I have 5 of them with the same fw version 2.4, however 3 of them appears as a hub 2 as simple accessory.

pivale, 2020-11-02:
I think it might depend on how you defined the input.
If detached, it needs to appear as a bridge/hub to support it.

You can check here for more information:
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Switch-Settings#detached-input

ursusHU, 2020-11-02:
@pivale Thanks, the thing is that I don't see any difference between the settings of them. As an example I attach a screenshot from 2 Shellies which behaving differently. (living room appears as hub, bathroom as a normal accessory)
[image]

andyblac, 2020-11-02:
iirc, if you have changed to a setting that requires a hub, but then put it back to one that doesn't it will stay in hub variant. ie enable detached or roller shutter, for example, then put back to normal, it will stay as a hub.

rojer, 2020-11-02:
a hub is created if we have more than one independent services to export. such as 2 switches (so they can be moved between rooms independently) or if we have a stateless switch (as a result of putting one of the inputs into a detached mode).
so, shelly 2.5 in switch mode will be a hub with 2 accessories but in roller shutter mode will be a single accessory with window covering service. if, however, you set input mode to single, an input is freed up and will be exported as a stateless switch, which will convert it into a hub with wc and stateless input accessories.
this should all be fairly transparent and inconsequential.

ursusHU, 2020-11-02:
> iirc, if you have changed to a setting that requires a hub, but then put it back to one that doesn't it will stay in hub variant. ie enable detached or roller shutter, for example, then put back to normal, it will stay as a hub.

I have different experience now, Although I never used my Shelly 2.5s as roller shutters or detached switch removed one from HomeKit  I reverted it back to the original firmware and then reinstalled this firmware. And still it is recognized as a hub.

rojer, 2020-11-02:
if you actually paid attention to what i said, you'd realise that this is the expected behavior for shelly 2.5.

what is the problem with it? does it cause any issues for you?

ursusHU, 2020-11-02:
@rojer No it doesn’t cause me problems, it was just a question. If you are still interested I kindly ask you to check my previous comment with the 2 pictures from 2 shellies. One appears as a hub, which is expected according to your description, however the other appears as a normal accessory. I don’t see any difference on their settings. 
I was thinking about the possible difference and I could only think of that the ones which not appearing now as a hub weren’t included when I installed the previous firmware which first (?) included the hub feature.

rojer, 2020-11-02:
there is one other case: back when the hub mode was introduced, there was problem of preserving backward compatibility during upgrade.
if we were to change everyone to hub organization, this would reset all the setting associated with the accessories.
this wouldn't be a good user experience, so a special quirk was added to preserve "legacy layout" for previously set up accessories (explained [here](https://github.com/mongoose-os-apps/shelly-homekit/issues/9#issuecomment-694418580)).

so, if you reset your previously set up device (remove pairing, press "reset" in the hap section of the UI) or perform a "structural change", such as switch to roller shutter mode and back or detach an input, legacy layout will be turned off and the device will become a hub, same as newly set up one.

closed 2020-11-02 by rojer