mongoose-os-apps/shelly-homekit #247: Can't add Shelly to HomeKit: "Unable to Add Accessory; Accessory is not reachable"

ricardopereira, 2020-11-02:
Hi, awesome project!

I want to add HomeKit support to my single Shelly v2.5 so I applied the 2.4.0 firmware. I read the Wiki and I defined a setup code successfully but when I try to add the Shelly it fails with "Unable to Add Accessory; Accessory is not reachable". The Shelly appears in the initial screen so it's weird that the accessory is not reachable.

I pinged the Shelly and I have no package lost:

```
➜  ~ ping 192.168.1.100
PING 192.168.1.100 (192.168.1.100): 56 data bytes
64 bytes from 192.168.1.100: icmp_seq=0 ttl=255 time=333.390 ms
64 bytes from 192.168.1.100: icmp_seq=1 ttl=255 time=36.592 ms
64 bytes from 192.168.1.100: icmp_seq=2 ttl=255 time=13.500 ms
64 bytes from 192.168.1.100: icmp_seq=3 ttl=255 time=8.430 ms
64 bytes from 192.168.1.100: icmp_seq=4 ttl=255 time=5.597 ms
64 bytes from 192.168.1.100: icmp_seq=5 ttl=255 time=9.252 ms
64 bytes from 192.168.1.100: icmp_seq=6 ttl=255 time=6.272 ms
64 bytes from 192.168.1.100: icmp_seq=7 ttl=255 time=10.822 ms
64 bytes from 192.168.1.100: icmp_seq=8 ttl=255 time=9.655 ms
64 bytes from 192.168.1.100: icmp_seq=9 ttl=255 time=17.009 ms
64 bytes from 192.168.1.100: icmp_seq=10 ttl=255 time=6.631 ms
64 bytes from 192.168.1.100: icmp_seq=11 ttl=255 time=8.615 ms
64 bytes from 192.168.1.100: icmp_seq=12 ttl=255 time=10.669 ms
64 bytes from 192.168.1.100: icmp_seq=13 ttl=255 time=5.644 ms
^C
--- 192.168.1.100 ping statistics ---
14 packets transmitted, 14 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 5.597/34.434/333.390/83.264 ms
```

I activated logging and retried the HomeKit setup but no success.
Anything suspicious?

**Logs:**

```
7059671274 mgos_http_server.c:180  0x3fff459c HTTP connection from 192.168.1.73:50924
7059682763 mgos_http_server.c:187  0x3fff459c GET /favicon.ico
7065120360 shelly_main.cpp:543     Up 7065.11, HAP 0/0/9 ns 0, RAM: 25032/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7073120239 shelly_main.cpp:543     Up 7073.11, HAP 0/0/9 ns 0, RAM: 25624/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7081120294 shelly_main.cpp:543     Up 7081.11, HAP 0/0/9 ns 0, RAM: 25624/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7089120286 shelly_main.cpp:543     Up 7089.11, HAP 0/0/9 ns 0, RAM: 25624/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7097120251 shelly_main.cpp:543     Up 7097.11, HAP 0/0/9 ns 0, RAM: 25624/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7105120293 shelly_main.cpp:543     Up 7105.11, HAP 0/0/9 ns 0, RAM: 25624/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7108584073 mgos_http_server.c:180  0x3fff5494 HTTP connection from 192.168.1.73:50926
7113129622 shelly_main.cpp:543     Up 7113.11, HAP 0/0/9 ns 0, RAM: 26464/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7121120263 shelly_main.cpp:543     Up 7121.11, HAP 0/0/9 ns 0, RAM: 25628/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7129120137 shelly_main.cpp:543     Up 7129.11, HAP 0/0/9 ns 0, RAM: 25628/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7137120467 shelly_main.cpp:543     Up 7137.11, HAP 0/0/9 ns 0, RAM: 25628/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7137951492 HAPPlatformTCPStrea:120 0x3fff5ba4 192.168.1.96:49692 Incoming HAP connection, ns 0/0/9
7137970071 HAPPlatformTCPStrea:274 0x3fff5ba4 192.168.1.96:49692 Accepted HAP connection, ns 0/1/9 ts 0x3fff5b5c
7139630462 HAPPlatformTCPStrea:104 0x3fff5ba4 192.168.1.96:49692 HAP connection closed, ns 0/1/9 i 2
7139643000 HAPPlatformTCPStrea:301 HAPPlatformTCPStreamClose ts 0x3fff5b5c nc 0x0 0/0/9
7139701076 HAPPlatformTCPStrea:120 0x3fff5ba4 192.168.1.96:49693 Incoming HAP connection, ns 0/0/9
7141059380 HAPPlatformTCPStrea:274 0x3fff5ba4 192.168.1.96:49693 Accepted HAP connection, ns 0/1/9 ts 0x3fff5b5c
7142769908 HAPPlatformTCPStrea:104 0x3fff5ba4 192.168.1.96:49693 HAP connection closed, ns 0/1/9 i 2
7142781684 HAPPlatformTCPStrea:301 HAPPlatformTCPStreamClose ts 0x3fff5b5c nc 0x0 0/0/9
7147695344 shelly_main.cpp:543     Up 7147.68, HAP 0/0/9 ns 0, RAM: 25628/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7155695095 shelly_main.cpp:543     Up 7155.68, HAP 0/0/9 ns 0, RAM: 25628/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7163695197 shelly_main.cpp:543     Up 7163.68, HAP 0/0/9 ns 0, RAM: 25628/41104; st 65; 4.1: c:0 mp:0.00 mt_ms:0 cp:0.00 tp:0.00 lemd:0
7165734311 mgos_http_server.c:180  0x3fff5ba4 HTTP connection from 192.168.1.73:50928
```

**Some screens:**

[image]

|   |   |   |
|---|---|---|
| [image] | [image] | [image] |
| [image]| [image] |   |
|   |   |   | 

Any clue?

Thank you for your precious time 😊

andyblac, 2020-11-02:
are you on a mesh network ?

ricardopereira, 2020-11-02:
@andyblac No but I have some powerline Wifi extension.

andyblac, 2020-11-02:
> @andyblac No but I have a powerline Wifi extension.

is the shelly device connectd to that AP ?, It could be that is not handling MultCast verry well if at all. try reading this thread https://github.com/mongoose-os-apps/shelly-homekit/issues/215

ricardopereira, 2020-11-02:
@andyblac I see. The Shelly is connected to the AP, at least it seems so from the router dashboard. I'm not very fluent in network terms and specifications 😅 Maybe I can try turning the AP off and force the Shelly connect to the main router, see if I can setup with HomeKit.

stefanzilske, 2020-11-02:
I have the same issue. I am on a mesh network. I have read through the mentioned issue, but what is the recommended solution?

My APs are working in auto mode as far as channels are concerned. What can I change to get the Shelly working?

andyblac, 2020-11-02:
check you hardward setup and make sure it fully supports MultiCast, 2nd nake sure you are NOT sharing a single SSID for 2.4Ghz and 5Ghz, if both of these are not possible then unforuntaly the only thik you can do, is replace your network hardware, for ones that fully support MultiCast forwarding and split SSID's for each wifi band.

ricardopereira, 2020-11-03:
Thanks @andyblac. That's it, I added the Shelly successfully to my HomeKit 🙌
I just split the SSID, each band has it's own SSID (2.4Ghz and 5Ghz).

ricardopereira, 2020-11-03:
@andyblac Well, the Shelly button in Home.app does nothing so I thought it was because I didn't do the calibration. So, now I issues in the calibration. It starts moving the roller shutter but after 2-3 seconds the calibration stops and Shelly starts to randomly switch every 1-2 seconds.

Should I open a new issue?

**Log:**

```
36940369765 shelly_hap_window_c:306 WC 1: State: moving -> stop (23 -> 24)
36940414998 mgos_sys_config.c:232   Loading conf2.json
36940438591 mgos_sys_config.c:232   Loading conf3.json
36940602664 mgos_sys_config.c:174   Saved to conf9.json
36940619655 shelly_hap_window_c:306 WC 1: State: stop -> stopping (24 -> 25)
36940637040 shelly_hap_window_c:306 WC 1: State: stopping -> idle (25 -> 0)
36940735206 shelly_hap_window_c:306 WC 1: State: idle -> move (0 -> 20)
36940834774 shelly_output.cpp:56    Output 1: off -> on (move)
36940845737 shelly_hap_window_c:306 WC 1: State: move -> rampup (20 -> 22)
36940934770 shelly_hap_window_c:511 P = 0.00 -> 0.00
36940944798 shelly_hap_window_c:306 WC 1: State: rampup -> moving (22 -> 23)
36941035983 shelly_output.cpp:56    Output 1: on -> off (moving)
36941046959 shelly_hap_window_c:306 WC 1: State: moving -> stop (23 -> 24)
36941157100 mgos_sys_config.c:232   Loading conf2.json
36941180707 mgos_sys_config.c:232   Loading conf3.json
36941310806 mgos_sys_config.c:174   Saved to conf9.json
36941328788 shelly_hap_window_c:306 WC 1: State: stop -> stopping (24 -> 25)
36941347938 shelly_hap_window_c:306 WC 1: State: stopping -> idle (25 -> 0)
36941446091 shelly_hap_window_c:306 WC 1: State: idle -> move (0 -> 20)
36941545523 shelly_output.cpp:56    Output 2: off -> on (move)
36941592522 shelly_hap_window_c:306 WC 1: State: move -> rampup (20 -> 22)
36941645829 shelly_hap_window_c:511 P = 0.00 -> 0.00
36941655854 shelly_hap_window_c:306 WC 1: State: rampup -> moving (22 -> 23)
36941746802 shelly_output.cpp:56    Output 2: on -> off (moving)
36941756796 shelly_hap_window_c:306 WC 1: State: moving -> stop (23 -> 24)
36941868243 mgos_sys_config.c:232   Loading conf2.json
36941891851 mgos_sys_config.c:232   Loading conf3.json
36942025912 mgos_sys_config.c:174   Saved to conf9.json
36942076838 shelly_hap_window_c:306 WC 1: State: stop -> stopping (24 -> 25)
36942093727 shelly_hap_window_c:306 WC 1: State: stopping -> idle (25 -> 0)
36942191897 shelly_hap_window_c:306 WC 1: State: idle -> move (0 -> 20)
36942291223 shelly_output.cpp:56    Output 1: off -> on (move)
36942301255 shelly_hap_window_c:306 WC 1: State: move -> rampup (20 -> 22)
36942391417 shelly_hap_window_c:511 P = 0.00 -> 0.00
36942401445 shelly_hap_window_c:306 WC 1: State: rampup -> moving (22 -> 23)
36942492819 shelly_output.cpp:56    Output 1: on -> off (moving)
36942502855 shelly_hap_window_c:306 WC 1: State: moving -> stop (23 -> 24)
36942614091 mgos_sys_config.c:232   Loading conf2.json
36942638676 mgos_sys_config.c:232   Loading conf3.json
36942807123 mgos_sys_config.c:174   Saved to conf9.json
36942824135 shelly_hap_window_c:306 WC 1: State: stop -> stopping (24 -> 25)
36942841303 shelly_hap_window_c:306 WC 1: State: stopping -> idle (25 -> 0)
36942939347 shelly_hap_window_c:306 WC 1: State: idle -> move (0 -> 20)
36943038817 shelly_output.cpp:56    Output 2: off -> on (move)
36943048816 shelly_hap_window_c:306 WC 1: State: move -> rampup (20 -> 22)
36943138920 shelly_hap_window_c:511 P = 0.00 -> 0.00
36943149917 shelly_hap_window_c:306 WC 1: State: rampup -> moving (22 -> 23)
36943241345 shelly_hap_window_c:315 WC 1: Cur pos 53.98 -> 46.67, P = 0.00
36943286762 shelly_output.cpp:56    Output 2: on -> off (moving)
36943297725 shelly_hap_window_c:306 WC 1: State: moving -> stop (23 -> 24)
36943363288 mgos_sys_config.c:232   Loading conf2.json
36943386916 mgos_sys_config.c:232   Loading conf3.json
36943515412 mgos_sys_config.c:174   Saved to conf9.json
36943532436 shelly_hap_window_c:306 WC 1: State: stop -> stopping (24 -> 25)
36943549268 shelly_hap_window_c:306 WC 1: State: stopping -> idle (25 -> 0)
36943647574 shelly_hap_window_c:306 WC 1: State: idle -> move (0 -> 20)
36943746659 shelly_output.cpp:56    Output 1: off -> on (move)
36943757661 shelly_hap_window_c:306 WC 1: State: move -> rampup (20 -> 22)
36943846928 shelly_hap_window_c:511 P = 0.00 -> 0.00
36943856966 shelly_hap_window_c:306 WC 1: State: rampup -> moving (22 -> 23)
36943948109 shelly_output.cpp:56    Output 1: on -> off (moving)
36943959115 shelly_hap_window_c:306 WC 1: State: moving -> stop (23 -> 24)
36944071454 mgos_sys_config.c:232   Loading conf2.json
[... 17 lines cut ...]
36945062693 shelly_hap_window_c:326 WC 1: Tgt pos 50.00 -> 0.00 (rpc)
36945096269 mgos_sys_config.c:232   Loading conf2.json
36945120926 mgos_sys_config.c:232   Loading conf3.json
36945286737 mgos_sys_config.c:174   Saved to conf9.json
36945310347 shelly_hap_window_c:306 WC 1: State: stopping -> idle (25 -> 0)
36945408621 shelly_hap_window_c:306 WC 1: State: idle -> move (0 -> 20)
36945507856 shelly_output.cpp:56    Output 2: off -> on (move)
36945517889 shelly_hap_window_c:306 WC 1: State: move -> rampup (20 -> 22)
36945608136 shelly_hap_window_c:511 P = 0.00 -> 0.00
36945618197 shelly_hap_window_c:306 WC 1: State: rampup -> moving (22 -> 23)
36945808241 shelly_hap_window_c:561 Moving to 0, p 0.00
36946209955 shelly_hap_window_c:315 WC 1: Cur pos 2.31 -> 0.00, P = 0.00
36946449483 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.73:57073
36946515717 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.73:57073
[... cut ...]
36946832113 mgos_sys_config.c:232   Loading conf2.json
36946855772 mgos_sys_config.c:232   Loading conf3.json
36946984049 mgos_sys_config.c:174   Saved to conf9.json
36947035055 shelly_hap_window_c:306 WC 1: State: stop -> stopping (24 -> 25)
36947052164 shelly_hap_window_c:306 WC 1: State: stopping -> idle (25 -> 0)
36947517836 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.1.73:57073
36952212467 shelly_main.cpp:543     Up 36952.20, HAP 0/3/9 ns 3, RAM: 23684/41104; st 63; 4.1: c:1 mp:0.00 mt_ms:1080 cp:0.00 tp:0.00 lemd:2
```

andyblac, 2020-11-03:
i dont have a roller shutter, so sorry i can't help, they are a few roller shutter tickets maybe take a look at those, if that does not help, yeah feel free to open a new ticket.

closed 2020-11-03 by andyblac

rojer, 2020-11-03:
judging by the log, motor is not using any current. this is very suspicious.

ricardopereira, 2020-11-03:
@rojer a malfunction Shelly?

stefanzilske, 2020-11-10:
> check you hardware setup and make sure it fully supports MultiCast, 2nd nake sure you are NOT sharing a single SSID for 2.4Ghz and 5Ghz, if both of these are not possible then unforuntaly the only thik you can do, is replace your network hardware, for ones that fully support MultiCast forwarding and split SSID's for each wifi band.

Understood. Is this specific to the HomeKit firmware, or does this also apply to the stock firmware?

andyblac, 2020-11-10:
> Understood. Is this specific to the HomeKit firmware, or does this also apply to the stock firmware?

the latest version of stock (fw 1.8 or above) also has reported issue do to CoAP.