mongoose-os-apps/shelly-homekit #248: Motor doesnt start most of the time, but works with stock firmware

C-Git-C, 2020-11-02:
Hi Rojer,

many, many thanks for the great project. i am using a shelly plug s and a shelly 2.5 so far.

My latest purchase are two shelly 2.5 for roller shutters. Both Shellys work with the delivered stock firmware and my motors 
(Oximo WT 10/17-2419)
Open and close works immediately via the buttons on the window as well as via the web interface.

But of course I want to use homekit. :-)

My Details:
Device: Model: Shelly25, ID: shellyswitch25-76xxxx
Version: 2.4.0 (build 20201026-115648/2.4.0-g94f5804-master)

In case of version: 2.4.0 (and earlier) I sometimes have to press the button on the window or button on the web interface up to 5 times until the shutter closes. Only in 1 of 10 times it seems to work immediately.

It seems that the motor starts with 1 second delay and then stops immediately.
I suspect that the current power is measured by the Shelly and the motor takes a little too long until the current power consumption is high enough so that the Shelly believes that no power is used and immediately stops.

If the motor is running and I trigger a directional change then it always works.

Could you help me? That would be very nice.
Regards
C

rojer, 2020-11-02:
you are probably right about startup taking longer.
nevreteless, please enable debug log, reproduce the issue and attach the log here.

C-Git-C, 2020-11-02:
hi,

i reproduced the issue.

517963591132 mgos_sys_config.c:232   Loading conf2.json
517963732062 mgos_sys_config.c:174   Saved to conf9.json
517963772778 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.x.46:54835
517966252511 shelly_main.cpp:543     Up 517966.24, HAP 0/2/9 ns 2, RAM: 23528/41104; st 71; 4.1: c:1 mp:130.15 mt_ms:16289 cp:100.00 tp:100.00 lemd:0
517974252603 shelly_main.cpp:543     Up 517974.24, HAP 0/2/9 ns 2, RAM: 23528/41104; st 71; 4.1: c:1 mp:130.15 mt_ms:16289 cp:100.00 tp:100.00 lemd:0
517978234646 shelly_input.cpp:72     Input 2: change (state 1)
517978245636 shelly_hap_window_c:326 WC 1: Tgt pos 100.00 -> 0.00 (ext)
517978255365 shelly_hap_window_c:306 WC 1: State: idle -> move (0 -> 20)
517978267908 shelly_output.cpp:56    Output 2: off -> on (move)
517978278859 shelly_hap_window_c:306 WC 1: State: move -> rampup (20 -> 22)
517978359747 shelly_hap_window_c:511 P = 0.00 -> 130.15
517978459457 shelly_hap_window_c:511 P = 0.00 -> 130.15
517978560601 shelly_hap_window_c:511 P = 0.00 -> 130.15
517978571714 shelly_input.cpp:72     Input 2: change (state 0)
517978615568 shelly_hap_window_c:511 P = 0.00 -> 130.15
517978659546 shelly_hap_window_c:511 P = 0.00 -> 130.15
517978759571 shelly_hap_window_c:511 P = 0.00 -> 130.15
517978769629 shelly_input.cpp:72     Input 2: single (state 0)
517978859485 shelly_hap_window_c:511 P = 0.00 -> 130.15
517978959594 shelly_hap_window_c:511 P = 0.00 -> 130.15
517979059593 shelly_hap_window_c:511 P = 0.00 -> 130.15
517979159644 shelly_hap_window_c:511 P = 0.00 -> 130.15
517979259533 shelly_hap_window_c:511 P = 1.04 -> 130.15
517979359601 shelly_hap_window_c:511 P = 0.00 -> 130.15
517979367814 shelly_hap_window_c:518 Failed to start moving
517979377700 shelly_hap_window_c:306 WC 1: State: rampup -> stop (22 -> 24)
517979459032 shelly_output.cpp:56    Output 2: on -> off (stop)
517979522642 mgos_sys_config.c:232   Loading conf2.json
517979668766 mgos_sys_config.c:174   Saved to conf9.json
517979685759 shelly_hap_window_c:306 WC 1: State: stop -> stopping (24 -> 25)
517979702197 shelly_hap_window_c:306 WC 1: State: stopping -> idle (25 -> 0)
517979800539 shelly_hap_window_c:306 WC 1: State: idle -> error (0 -> 100)
517979901320 shelly_hap_window_c:326 WC 1: Tgt pos 0.00 -> 100.00 (error)
517979911187 shelly_hap_window_c:306 WC 1: State: error -> idle (100 -> 0)
517981723144 shelly_input.cpp:72     Input 2: change (state 1)
517981765775 shelly_hap_window_c:326 WC 1: Tgt pos 100.00 -> 0.00 (ext)
517981775519 shelly_hap_window_c:306 WC 1: State: idle -> move (0 -> 20)
517981799486 shelly_output.cpp:56    Output 2: off -> on (move)
517981809485 shelly_hap_window_c:306 WC 1: State: move -> rampup (20 -> 22)
517981900196 shelly_hap_window_c:511 P = 0.00 -> 130.15
517982000404 shelly_hap_window_c:511 P = 0.00 -> 130.15
517982038201 shelly_input.cpp:72     Input 2: change (state 0)
517982047806 shelly_hap_window_c:511 P = 0.00 -> 130.15
517982100175 shelly_hap_window_c:511 P = 0.00 -> 130.15
517982200257 shelly_hap_window_c:511 P = 0.00 -> 130.15
517982252515 shelly_main.cpp:543     Up 517982.24, HAP 0/2/9 ns 2, RAM: 23488/41104; st 71; 4.1: c:1 mp:130.15 mt_ms:16289 cp:100.00 tp:0.00 lemd:0
517982294608 shelly_input.cpp:72     Input 2: single (state 0)
517982306259 shelly_hap_window_c:511 P = 0.00 -> 130.15
517982400361 shelly_hap_window_c:511 P = 1.05 -> 130.15
517982500402 shelly_hap_window_c:511 P = 69.12 -> 130.15
517982600686 shelly_hap_window_c:511 P = 140.32 -> 130.15
517982610736 shelly_hap_window_c:306 WC 1: State: rampup -> moving (22 -> 23)
517982702378 shelly_hap_window_c:315 WC 1: Cur pos 100.00 -> 99.52, P = 143.08
517983400818 shelly_hap_window_c:561 Moving to 0, p 143.11
517983502341 shelly_hap_window_c:315 WC 1: Cur pos 95.22 -> 94.61, P = 143.02
517984200842 shelly_hap_window_c:561 Moving to 0, p 143.07
517984302346 shelly_hap_window_c:315 WC 1: Cur pos 90.31 -> 89.69, P = 142.91
517985000927 shelly_hap_window_c:561 Moving to 0, p 142.87
517985102337 shelly_hap_window_c:315 WC 1: Cur pos 85.40 -> 84.79, P = 142.66
517985800757 shelly_hap_window_c:561 Moving to 0, p 142.43
517985902358 shelly_hap_window_c:315 WC 1: Cur pos 80.48 -> 79.87, P = 142.48
517986600938 shelly_hap_window_c:561 Moving to 0, p 142.55
517986702407 shelly_hap_window_c:315 WC 1: Cur pos 75.57 -> 74.96, P = 142.45
517987400771 shelly_hap_window_c:561 Moving to 0, p 142.25
517987502392 shelly_hap_window_c:315 WC 1: Cur pos 70.66 -> 70.05, P = 142.20
517988200702 shelly_hap_window_c:561 Moving to 0, p 142.49
[... 136 lines cut ...]
518022074845 shelly_hap_window_c:315 WC 1: Cur pos 66.79 -> 67.40, P = 139.39
518022085121 shelly_hap_window_c:561 Moving to 100, p 139.39
518022252451 shelly_main.cpp:543     Up 518022.24, HAP 0/2/9 ns 2, RAM: 23532/41104; st 72; 4.1: c:1 mp:130.15 mt_ms:16289 cp:68.02 tp:100.00 lemd:0
518022874867 shelly_hap_window_c:315 WC 1: Cur pos 71.70 -> 72.31, P = 139.20
518022885159 shelly_hap_window_c:561 Moving to 100, p 139.20
518023674909 shelly_hap_window_c:315 WC 1: Cur pos 76.61 -> 77.22, P = 138.88
518023684232 shelly_hap_window_c:561 Moving to 100, p 138.88
518024474959 shelly_hap_window_c:315 WC 1: Cur pos 81.52 -> 82.14, P = 138.86
518024484295 shelly_hap_window_c:561 Moving to 100, p 138.86
518025274988 shelly_hap_window_c:315 WC 1: Cur pos 86.43 -> 87.05, P = 139.46
518025284313 shelly_hap_window_c:561 Moving to 100, p 139.46
518026074943 shelly_hap_window_c:315 WC 1: Cur pos 91.34 -> 91.96, P = 139.21
[... cut ...]
518038252651 shelly_main.cpp:543     Up 518038.24, HAP 0/2/9 ns 2, RAM: 23528/41104; st 72; 4.1: c:1 mp:130.15 mt_ms:16289 cp:100.00 tp:100.00 lemd:0
518046252351 shelly_main.cpp:543     Up 518046.24, HAP 0/2/9 ns 2, RAM: 23528/41104; st 72; 4.1: c:1 mp:130.15 mt_ms:16289 cp:100.00 tp:100.00 lemd:0
518053475958 mgos_http_server.c:180  0x3fff63ac HTTP connection from 192.168.x.46:54935
518054252713 shelly_main.cpp:543     Up 518054.24, HAP 0/2/9 ns 2, RAM: 23524/41104; st 72; 4.1: c:1 mp:130.15 mt_ms:16289 cp:100.00 tp:100.00 lemd:0

C-Git-C, 2020-11-11:
I reproduced the issue again with both of my shelly shutters 

Device: Model: Shelly25, ID: shellyswitch25-76xxxx
Version: **2.5.0** (build 20201109-203929/2.5.0-g42dbdba-master)

121870478585 shelly_hap_window_c:318 WC 1: Cur pos 50.40 -> 49.78, P = 144.84
121870487607 shelly_hap_window_c:581 Moving to 0, p 144.84
121871211764 shelly_hap_window_c:318 WC 1: Cur pos 45.49 -> 44.87, P = 144.85
121871220772 shelly_hap_window_c:581 Moving to 0, p 144.85
121871263422 shelly_input.cpp:70     Input 2: change (state 1)
121871315768 shelly_hap_window_c:329 WC 1: Tgt pos 0.00 -> 44.23 (ext)
121871327976 shelly_hap_window_c:329 WC 1: Tgt pos 44.23 -> 44.16 (fixup)
121871337095 shelly_output.cpp:56    Output 2: on -> off (moving)
121871348663 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
121871392277 mgos_sys_config.c:232   Loading conf2.json
121871534514 mgos_sys_config.c:174   Saved to conf9.json
121871551525 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
121871569431 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
121871639866 shelly_input.cpp:70     Input 2: change (state 0)
121871858223 shelly_input.cpp:70     Input 2: single (state 0)
121873684694 shelly_main.cpp:530     Up 121873.67, HAP 0/2/9 ns 2, RAM: 23488/41024; st 69; 4.1: c:1 mp:130.15 mt_ms:16289 cp:44.16 tp:44.16 lemd:2 lhmd:0
121881684700 shelly_main.cpp:530     Up 121881.67, HAP 0/2/9 ns 2, RAM: 23488/41024; st 69; 4.1: c:1 mp:130.15 mt_ms:16289 cp:44.16 tp:44.16 lemd:2 lhmd:0
121888856092 shelly_debug.cpp:203    Tailing
121889684695 shelly_main.cpp:530     Up 121889.67, HAP 0/2/9 ns 2, RAM: 23092/41024; st 69; 4.1: c:1 mp:130.15 mt_ms:16289 cp:44.16 tp:44.16 lemd:2 lhmd:0
121897684197 shelly_main.cpp:530     Up 121897.67, HAP 0/2/9 ns 2, RAM: 23092/41024; st 69; 4.1: c:1 mp:130.15 mt_ms:16289 cp:44.16 tp:44.16 lemd:2 lhmd:0
121905684337 shelly_main.cpp:530     Up 121905.67, HAP 0/2/9 ns 2, RAM: 23092/41024; st 69; 4.1: c:1 mp:130.15 mt_ms:16289 cp:44.16 tp:44.16 lemd:2 lhmd:0
121910276325 shelly_input.cpp:70     Input 1: change (state 1)
121910328629 shelly_hap_window_c:329 WC 1: Tgt pos 44.16 -> 100.00 (ext)
121910340496 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
121910366159 shelly_output.cpp:56    Output 1: off -> on (move)
121910377129 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
121910467122 shelly_hap_window_c:531 P = 0.00 -> 130.15
121910526990 shelly_input.cpp:70     Input 1: change (state 0)
121910537030 shelly_hap_window_c:531 P = 0.00 -> 130.15
121910566771 shelly_hap_window_c:531 P = 0.00 -> 130.15
121910667293 shelly_hap_window_c:531 P = 0.00 -> 130.15
121910769011 shelly_hap_window_c:531 P = 0.00 -> 130.15
121910850357 shelly_input.cpp:70     Input 1: single (state 0)
121910867164 shelly_hap_window_c:531 P = 0.00 -> 130.15
121910966987 shelly_hap_window_c:531 P = 0.00 -> 130.15
121911067124 shelly_hap_window_c:531 P = 26.51 -> 130.15
121911167313 shelly_hap_window_c:531 P = 137.06 -> 130.15
121911177874 shelly_hap_window_c:309 WC 1: State: rampup -> moving (22 -> 23)
121911768966 shelly_hap_window_c:318 WC 1: Cur pos 47.09 -> 47.71, P = 144.25
121911867934 shelly_hap_window_c:581 Moving to 100, p 144.20
121912569164 shelly_hap_window_c:318 WC 1: Cur pos 52.00 -> 52.62, P = 144.12
121912668118 shelly_hap_window_c:581 Moving to 100, p 144.10
121913369685 shelly_hap_window_c:318 WC 1: Cur pos 56.92 -> 57.54, P = 144.16
121913467611 shelly_hap_window_c:581 Moving to 100, p 144.07
121913684469 shelly_main.cpp:530     Up 121913.67, HAP 0/2/9 ns 2, RAM: 23044/41024; st 69; 4.1: c:1 mp:130.15 mt_ms:16289 cp:59.37 tp:100.00 lemd:1 lhmd:0
121914168963 shelly_hap_window_c:318 WC 1: Cur pos 61.83 -> 62.44, P = 143.86
121914267659 shelly_hap_window_c:581 Moving to 100, p 143.56
121914968824 shelly_hap_window_c:318 WC 1: Cur pos 66.74 -> 67.35, P = 143.65
121915067641 shelly_hap_window_c:581 Moving to 100, p 143.87
121915768842 shelly_hap_window_c:318 WC 1: Cur pos 71.65 -> 72.26, P = 143.82
121915867736 shelly_hap_window_c:581 Moving to 100, p 143.50
121916569146 shelly_hap_window_c:318 WC 1: Cur pos 76.56 -> 77.18, P = 143.72
121916667800 shelly_hap_window_c:581 Moving to 100, p 143.76
121917369345 shelly_hap_window_c:318 WC 1: Cur pos 81.47 -> 82.09, P = 143.86
121917467728 shelly_hap_window_c:581 Moving to 100, p 143.37
121918169060 shelly_hap_window_c:318 WC 1: Cur pos 86.38 -> 87.00, P = 143.84
121918267742 shelly_hap_window_c:581 Moving to 100, p 143.62
121918968783 shelly_hap_window_c:318 WC 1: Cur pos 91.29 -> 91.91, P = 143.64
121919067746 shelly_hap_window_c:581 Moving to 100, p 143.74
121919768681 shelly_hap_window_c:318 WC 1: Cur pos 96.21 -> 96.82, P = 143.48
121919867685 shelly_hap_window_c:581 Moving to 100, p 143.47
121920667227 shelly_hap_window_c:581 Moving to 100, p 90.71
121920767476 shelly_output.cpp:56    Output 1: on -> off (moving)
121920779619 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
121920888531 mgos_sys_config.c:232   Loading conf2.json
[... 317 lines cut ...]
121983771252 shelly_hap_window_c:531 P = 0.00 -> 130.15
121983871602 shelly_hap_window_c:531 P = 0.00 -> 130.15
121983971722 shelly_hap_window_c:531 P = 1.03 -> 130.15
121984073287 shelly_hap_window_c:531 P = 1.12 -> 130.15
121984171323 shelly_hap_window_c:531 P = 0.00 -> 130.15
121984271327 shelly_hap_window_c:531 P = 46.73 -> 130.15
121984371499 shelly_hap_window_c:531 P = 140.57 -> 130.15
[... cut ...]
121985684330 shelly_main.cpp:530     Up 121985.67, HAP 0/2/9 ns 2, RAM: 23472/41024; st 70; 4.1: c:1 mp:130.15 mt_ms:16289 cp:18.50 tp:18.50 lemd:2 lhmd:0
121993684236 shelly_main.cpp:530     Up 121993.67, HAP 0/2/9 ns 2, RAM: 23472/41024; st 70; 4.1: c:1 mp:130.15 mt_ms:16289 cp:18.50 tp:18.50 lemd:2 lhmd:0
122001684208 shelly_main.cpp:530     Up 122001.67, HAP 0/2/9 ns 2, RAM: 23472/41024; st 69; 4.1: c:1 mp:130.15 mt_ms:16289 cp:18.50 tp:18.50 lemd:2 lhmd:0
122009684232 shelly_main.cpp:530     Up 122009.67, HAP 0/2/9 ns 2, RAM: 23472/41024; st 69; 4.1: c:1 mp:130.15 mt_ms:16289 cp:18.50 tp:18.50 lemd:2 lhmd:0

C-Git-C, 2020-11-23:
Hi Rojer,

I reproduced the issue again.

At the beginning Shutter position was near to the midle.
I pushed the button the first time to open the shutter - shutter didnt start to move there was only a short jerk 
(**Failed to start moving**)
I pushed the button the second time to open the shutter - shutter didnt start to move there was only a short jerk
(**Failed to start moving**)
I pushed the button the third time to open the shutter - shutter started to move and opend the complety

Sometimes it works instantly but most of the it needs serveral tries to move shutter.

Input Mode: Seperate - momentary
Device: Model: Shelly25, ID: shellyswitch25-76xxxx
Version: 2.5.0 (build 20201109-203929/2.5.0-g42dbdba-master)

1128049306072:1606143055177241 mgos_sys_config.c:232   Loading conf2.json
1128049487688 mgos_sys_config.c:174   Saved to conf9.json
1128049529329 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.2.46:63704
1128054902858 shelly_main.cpp:530     Up 1128054.89, HAP 0/2/9 ns 2, RAM: 23848/41024; st 70; 4.1: c:1 mp:130.15 mt_ms:16289 cp:40.16 tp:40.16 lemd:1 lhmd:0
1128062902811 shelly_main.cpp:530     Up 1128062.89, HAP 0/2/9 ns 2, RAM: 23848/41024; st 69; 4.1: c:1 mp:130.15 mt_ms:16289 cp:40.16 tp:40.16 lemd:1 lhmd:0
1128063326019 shelly_input.cpp:70     Input 1: change (state 1)
1128063336163 shelly_hap_window_c:329 WC 1: Tgt pos 40.16 -> 100.00 (ext)
1128063345921 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
1128063414555 shelly_output.cpp:56    Output 1: off -> on (move)
1128063425574 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
1128063515195 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128063527048 shelly_input.cpp:70     Input 1: change (state 0)
1128063536665 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128063615201 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128063715172 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128063815317 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128063855118 shelly_input.cpp:70     Input 1: single (state 0)
1128063915054 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128064015128 shelly_hap_window_c:531 P = 1.01 -> 130.15
1128064115195 shelly_hap_window_c:531 P = 1.15 -> 130.15
1128064215123 shelly_hap_window_c:531 P = 1.37 -> 130.15
1128064315188 shelly_hap_window_c:531 P = 1.03 -> 130.15
1128064415417 shelly_hap_window_c:531 P = 1.21 -> 130.15
1128064515163 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128064523430 shelly_hap_window_c:538 **Failed to start moving**
1128064533348 shelly_hap_window_c:309 WC 1: State: rampup -> stop (22 -> 24)
1128064614296 shelly_output.cpp:56    Output 1: on -> off (stop)
1128064647596 mgos_sys_config.c:232   Loading conf2.json
1128064788041 mgos_sys_config.c:174   Saved to conf9.json
1128064806022 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
1128064865972 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
1128064964419 shelly_hap_window_c:309 WC 1: State: idle -> error (0 -> 100)
1128065065152 shelly_hap_window_c:329 WC 1: Tgt pos 100.00 -> 40.16 (error)
1128065076025 shelly_hap_window_c:309 WC 1: State: error -> idle (100 -> 0)
1128066695554 shelly_input.cpp:70     Input 1: change (state 1)
1128066705731 shelly_hap_window_c:329 WC 1: Tgt pos 40.16 -> 100.00 (ext)
1128066716461 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
**1128066763248 shelly_output.cpp:56    Output 1: off -> on (move)**
1128066773334 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
1128066863979 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128066875245 shelly_input.cpp:70     Input 1: change (state 0)
1128066884876 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128066963915 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128067063950 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128067164225 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128067225712 shelly_input.cpp:70     Input 1: single (state 0)
1128067264065 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128067363906 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128067464040 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128067563967 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128067664015 shelly_hap_window_c:531 P = 1.05 -> 130.15
1128067763943 shelly_hap_window_c:531 P = 0.00 -> 130.15
1128067864080 shelly_hap_window_c:531 P = 0.00 -> 130.15
[... 35 lines cut ...]
1128073868649 shelly_hap_window_c:581 Moving to 100, p 143.78
1128074270225 shelly_hap_window_c:318 WC 1: Cur pos 60.90 -> 61.51, P = 143.91
1128074668669 shelly_hap_window_c:581 Moving to 100, p 144.02
1128075070258 shelly_hap_window_c:318 WC 1: Cur pos 65.81 -> 66.43, P = 143.73
1128075468700 shelly_hap_window_c:581 Moving to 100, p 144.13
1128075870040 shelly_hap_window_c:318 WC 1: Cur pos 70.72 -> 71.33, P = 143.57
1128076268660 shelly_hap_window_c:581 Moving to 100, p 143.57
1128076670010 shelly_hap_window_c:318 WC 1: Cur pos 75.63 -> 76.24, P = 143.94
1128077068660 shelly_hap_window_c:581 Moving to 100, p 143.27
1128077469971 shelly_hap_window_c:318 WC 1: Cur pos 80.54 -> 81.15, P = 143.52
1128077868620 shelly_hap_window_c:581 Moving to 100, p 143.58
1128078270088 shelly_hap_window_c:318 WC 1: Cur pos 85.46 -> 86.07, P = 143.65
1128078668548 shelly_hap_window_c:581 Moving to 100, p 143.15
1128078903209 shelly_main.cpp:530     Up 1128078.89, HAP 0/2/9 ns 2, RAM: 23848/41024; st 70; 4.1: c:1 mp:130.15 mt_ms:16289 cp:89.75 tp:100.00 lemd:1 lhmd:0
1128079070262 shelly_hap_window_c:318 WC 1: Cur pos 90.37 -> 90.98, P = 143.55
[... cut ...]
1128080178584 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
1128080291309 mgos_sys_config.c:232   Loading conf2.json
1128080474720 mgos_sys_config.c:174   Saved to conf9.json
1128080492763 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
1128080509888 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
1128086408666 shelly_debug.cpp:203    Tailing
1128086902771 shelly_main.cpp:530     Up 1128086.89, HAP 0/2/9 ns 2, RAM: 23460/41024; st 69; 4.1: c:1 mp:130.15 mt_ms:16289 cp:100.00 tp:100.00 lemd:1 lhmd:0

C-Git-C, 2020-11-23:
I am not sure but do this to lines show the power consumtion? Seems to be higher and then the shutter moves

1128070668258 shelly_hap_window_c:531 P = 74.91 -> 130.15
1128070768481 shelly_hap_window_c:531 P = 141.15 -> 130.15

C-Git-C, 2020-11-29:
Hi Rojer,

I reproduced the issue again.

At the beginning Shutter position was near to the midle.
I pushed the button **six** times to open the shutter - shutter didnt start to move there was only a short jerk
(Failed to start moving)

I pushed the button a **seventh** time to open the shutter - shutter started to move and opend the completly

Sometimes it works instantly but most of the it needs serveral tries to move shutter.

Input Mode: Seperate - momentary
Device: Model: Shelly25, ID: shellyswitch25-76xxxx
Version: 2.6.1 (build 20201128-161048/2.6.1-g3ba3275-master)

4134630634:1606669120692110 mgos_sys_config.c:232   Loading conf2.json
4134771532 mgos_sys_config.c:174   Saved to conf9.json
4134812615 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.2.46:63321
4137170322 shelly_main.cpp:554     Up 4137.16, HAP 0/2/9 ns 2, RAM: 23844/41008; st 63; 4.1: c:1 mp:130.15 mt_ms:16289 cp:47.98 tp:47.98 lemd:0 lhmd:0
4145170362 shelly_main.cpp:554     Up 4145.16, HAP 0/2/9 ns 2, RAM: 23844/41008; st 63; 4.1: c:1 mp:130.15 mt_ms:16289 cp:47.98 tp:47.98 lemd:0 lhmd:0
4146000624 shelly_input.cpp:70     Input 1: change (state 1)
4146010785 shelly_hap_window_c:329 WC 1: Tgt pos 47.98 -> 100.00 (ext)
4146020541 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
4146096795 shelly_output.cpp:56    Output 1: off -> on (move)
4146107838 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
4146197831 shelly_hap_window_c:531 P = 0.00 -> 130.15
4146228543 shelly_input.cpp:70     Input 1: change (state 0)
4146238181 shelly_hap_window_c:531 P = 0.00 -> 130.15
4146297627 shelly_hap_window_c:531 P = 0.00 -> 130.15
4146397690 shelly_hap_window_c:531 P = 0.00 -> 130.15
4146497547 shelly_hap_window_c:531 P = 1.09 -> 130.15
4146529868 shelly_input.cpp:70     Input 1: single (state 0)
4146598070 shelly_hap_window_c:531 P = 0.00 -> 130.15
4146697591 shelly_hap_window_c:531 P = 1.07 -> 130.15
4146799159 shelly_hap_window_c:531 P = 1.11 -> 130.15
4146897561 shelly_hap_window_c:531 P = 1.02 -> 130.15
4146997567 shelly_hap_window_c:531 P = 0.00 -> 130.15
4147097503 shelly_hap_window_c:531 P = 0.00 -> 130.15
4147197490 shelly_hap_window_c:531 P = 0.00 -> 130.15
4147205746 shelly_hap_window_c:538 **Failed to start moving**
4147215648 shelly_hap_window_c:309 WC 1: State: rampup -> stop (22 -> 24)
4147296756 shelly_output.cpp:56    Output 1: on -> off (stop)
4147330944 mgos_sys_config.c:232   Loading conf2.json
4147471893 mgos_sys_config.c:174   Saved to conf9.json
4147488920 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
4147504695 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
4147602986 shelly_hap_window_c:309 WC 1: State: idle -> error (0 -> 100)
4147703573 shelly_hap_window_c:329 WC 1: Tgt pos 100.00 -> 47.98 (error)
4147713488 shelly_hap_window_c:309 WC 1: State: error -> idle (100 -> 0)
4149875875 shelly_input.cpp:70     Input 1: change (state 1)
4149886065 shelly_hap_window_c:329 WC 1: Tgt pos 47.98 -> 100.00 (ext)
4149895845 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
4149908448 shelly_output.cpp:56    Output 1: off -> on (move)
4149919504 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
4150002760 shelly_hap_window_c:531 P = 0.00 -> 130.15
4150085556 shelly_input.cpp:70     Input 1: change (state 0)
4150095227 shelly_hap_window_c:531 P = 0.00 -> 130.15
4150108391 shelly_hap_window_c:531 P = 1.09 -> 130.15
4150202953 shelly_hap_window_c:531 P = 0.00 -> 130.15
4150302785 shelly_hap_window_c:531 P = 1.08 -> 130.15
4150402800 shelly_hap_window_c:531 P = 0.00 -> 130.15
4150413004 shelly_input.cpp:70     Input 1: single (state 0)
4150502806 shelly_hap_window_c:531 P = 1.02 -> 130.15
4150602973 shelly_hap_window_c:531 P = 1.14 -> 130.15
4150702942 shelly_hap_window_c:531 P = 1.43 -> 130.15
4150803018 shelly_hap_window_c:531 P = 1.01 -> 130.15
4150902737 shelly_hap_window_c:531 P = 0.00 -> 130.15
4151002511 shelly_hap_window_c:531 P = 0.00 -> 130.15
4151010792 shelly_hap_window_c:538 **Failed to start moving**
[... 149 lines cut ...]
4170642189 shelly_hap_window_c:581 Moving to 100, p 146.26
4171243706 shelly_hap_window_c:318 WC 1: Cur pos 61.97 -> 62.58, P = 145.90
4171442408 shelly_hap_window_c:581 Moving to 100, p 146.31
4172043891 shelly_hap_window_c:318 WC 1: Cur pos 66.88 -> 67.49, P = 145.79
4172242251 shelly_hap_window_c:581 Moving to 100, p 145.87
4172843794 shelly_hap_window_c:318 WC 1: Cur pos 71.80 -> 72.40, P = 146.08
4173042413 shelly_hap_window_c:581 Moving to 100, p 145.46
4173644005 shelly_hap_window_c:318 WC 1: Cur pos 76.70 -> 77.32, P = 145.90
4173842155 shelly_hap_window_c:581 Moving to 100, p 145.67
4174443537 shelly_hap_window_c:318 WC 1: Cur pos 81.61 -> 82.23, P = 145.87
4174642411 shelly_hap_window_c:581 Moving to 100, p 145.86
4175243630 shelly_hap_window_c:318 WC 1: Cur pos 86.52 -> 87.14, P = 145.42
4175442428 shelly_hap_window_c:581 Moving to 100, p 145.66
4176043907 shelly_hap_window_c:318 WC 1: Cur pos 91.44 -> 92.05, P = 145.54
4176242458 shelly_hap_window_c:581 Moving to 100, p 145.00
4176843940 shelly_hap_window_c:318 WC 1: Cur pos 96.35 -> 96.97, P = 145.50
4177042468 shelly_hap_window_c:581 Moving to 100, p 145.10
4177170608 shelly_main.cpp:554     Up 4177.16, HAP 0/2/9 ns 2, RAM: 23844/41008; st 63; 4.1: c:1 mp:130.15 mt_ms:16289 cp:98.81 tp:100.00 lemd:1 lhmd:0
4177742194 shelly_output.cpp:56    Output 1: on -> off (moving)
4177752390 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
4177863689 mgos_sys_config.c:232   Loading conf2.json
4178006297 mgos_sys_config.c:174   Saved to conf9.json
4178023430 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
4178041597 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
4185583080 shelly_debug.cpp:203    Tailing

C-Git-C, 2020-12-07:
Hi Rojer,

do you need me to explain that issue clearer?

andyblac, 2021-01-27:
@rojer can you offer any assistance here ?

C-Git-C, 2021-01-27:
hello Andy, 
hello Rojer

yesterday I upgraded both of my shutters to the latest version version: 2.7.3 (build 20210124-130648/2.7.3-g883914a-master). 

The problem persists unchanged.

Regards
C

C-Git-C, 2021-02-02:
Hi Rojer,

I reproduced the issue again. Version: 2.7.3 (build 20210124-130648/2.7.3-g883914a-master)

At the beginning Shutter position was open
I pushed the button three times to open the shutter - shutter didnt start to move there was only a short jerk
(Failed to start moving)

I pushed the button a fourth time to open the shutter - shutter started to move and opend

Sometimes it works instantly but most of the it needs serveral tries to move shutter.

584583649433:1612269478533635 shelly_rpc_service.:223 SetConfig ok, 0
584583677276 mgos_sys_config.c:232   Loading conf2.json
584583821733 mgos_sys_config.c:174   Saved to conf9.json
584585120069 shelly_main.cpp:543     Up 584585.11, HAP 0/2/12 ns 2, RAM: 20144/37616; st 62; 4.1: c:1 mp:130.15 mt_ms:16289 cp:100.00 tp:100.00 lemd:1 lhmd:0
584585151913 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584585975696 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584586977354 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584587976458 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584588977127 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584589985493 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584590364684 mgos_http_server.c:180  0x3fff7324 HTTP connection from 192.168.2.46:53507
584590376086 mgos_http_server.c:180  0x3fff73ac HTTP connection from 192.168.2.46:53508
584590562622 shelly_debug.cpp:203    Tailing
584590976281 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584591976206 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584592984366 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584593119466 shelly_main.cpp:543     Up 584593.11, HAP 0/2/12 ns 2, RAM: 19636/37616; st 62; 4.1: c:1 mp:130.15 mt_ms:16289 cp:100.00 tp:100.00 lemd:1 lhmd:0
584593976461 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584594987096 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584595976645 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584596974883 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584597046563 mgos_mongoose.c:66      New heap free LWM: 15400
584597978826 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584598409862 mg_rpc.c:305            Shelly.SetState via WS_in 192.168.2.46:53491
584598421355 shelly_hap_window_c:342 WC 1: Tgt pos 100.00 -> 0.00 (RPC)
584598440575 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
584598458624 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584598478304 mgos_mongoose.c:66      New heap free LWM: 15360
584598539773 shelly_output.cpp:56    Output 2: off -> on (move)
584598550340 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
584598640546 shelly_hap_window_c:544 P = 0.00 -> 130.15
584598740431 shelly_hap_window_c:544 P = 0.00 -> 130.15
584598840589 shelly_hap_window_c:544 P = 0.00 -> 130.15
584598940529 shelly_hap_window_c:544 P = 0.00 -> 130.15
584598975377 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584599040648 shelly_hap_window_c:544 P = 0.00 -> 130.15
584599140519 shelly_hap_window_c:544 P = 0.00 -> 130.15
584599240587 shelly_hap_window_c:544 P = 0.00 -> 130.15
584599340551 shelly_hap_window_c:544 P = 0.00 -> 130.15
584599440605 shelly_hap_window_c:544 P = 1.05 -> 130.15
584599540548 shelly_hap_window_c:544 P = 0.00 -> 130.15
584599640588 shelly_hap_window_c:544 P = 0.00 -> 130.15
**584599649391 shelly_hap_window_c:551 Failed to start moving**
584599660018 shelly_hap_window_c:322 WC 1: State: rampup -> stop (22 -> 24)
584599739805 shelly_output.cpp:56    Output 2: on -> off (stop)
584599772616 mgos_sys_config.c:232   Loading conf2.json
584599917861 mgos_sys_config.c:174   Saved to conf9.json
584599929853 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
584599948258 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
584599976649 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584599996292 mgos_mongoose.c:66      New heap free LWM: 15352
584600046528 shelly_hap_window_c:322 WC 1: State: idle -> error (0 -> 100)
584600147108 shelly_hap_window_c:342 WC 1: Tgt pos 0.00 -> 100.00 (error)
584600157562 shelly_hap_window_c:322 WC 1: State: error -> idle (100 -> 0)
584600988780 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584601119719 shelly_main.cpp:543     Up 584601.11, HAP 0/2/12 ns 2, RAM: 19636/37616; st 62; 4.1: c:1 mp:130.15 mt_ms:16289 cp:100.00 tp:100.00 lemd:1 lhmd:0
584601975721 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.46:53491
584602745386 mg_rpc.c:305            Shelly.SetState via WS_in 192.168.2.46:53491
[... 107 lines cut ...]
584619507721 shelly_hap_window_c:331 WC 1: Cur pos 47.96 -> 47.34, P = 146.19
584620205462 shelly_hap_window_c:594 Moving to 0, p 146.38
584620307712 shelly_hap_window_c:331 WC 1: Cur pos 43.05 -> 42.43, P = 146.31
584621005601 shelly_hap_window_c:594 Moving to 0, p 145.58
584621107212 shelly_hap_window_c:331 WC 1: Cur pos 38.14 -> 37.52, P = 145.51
584621805416 shelly_hap_window_c:594 Moving to 0, p 145.33
584621907253 shelly_hap_window_c:331 WC 1: Cur pos 33.22 -> 32.61, P = 145.30
584622605449 shelly_hap_window_c:594 Moving to 0, p 145.65
584622707702 shelly_hap_window_c:331 WC 1: Cur pos 28.31 -> 27.70, P = 145.73
[... cut ...]
584626305857 shelly_output.cpp:56    Output 2: on -> off (moving)
584626316517 shelly_hap_window_c:322 WC 1: State: moving -> stop (23 -> 24)
584626426169 mgos_sys_config.c:232   Loading conf2.json
584626573362 mgos_sys_config.c:174   Saved to conf9.json
584626585482 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
584626603836 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
584633118808 shelly_main.cpp:543     Up 584633.11, HAP 0/2/12 ns 2, RAM: 19636/37616; st 62; 4.1: c:1 mp:130.15 mt_ms:16289 cp:0.00 tp:0.00 lemd:1 lhmd:0

andyblac, 2021-03-03:
@rojer can you help ?

C-Git-C, 2021-03-03:
Hi Rojer,

I reproduced the issue again. Version: 2.8.0-alpha3 Build: 20210223-015212/2.7.3-103-g20412b5-master

At the beginning Shutter position was closed
I pushed the button two times to open the shutter - shutter didnt start to move there was only a short jerk
(Failed to start moving)

I pushed the button a third time to open the shutter - shutter started to move and opened

Sometimes it works instantly but most of the it needs serveral tries to move shutter.

680220822451 shelly_main.cpp:554     Up 680220.81, HAP 0/2/12 ns 2, RAM: 20524/37648; st 63; 4.1: c:1 mp:139.36 mt_ms:16189 cp:0.00 tp:0.00 lemd:2 lhmd:0
680224785507 shelly_input.cpp:76     Input 1: change (state 1)
680224796015 shelly_hap_window_c:342 WC 1: Tgt pos 0.00 -> 100.00 (ext)
680224806583 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
680224840163 shelly_output.cpp:58    Output 1: off -> on (move)
680224851700 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
680224941138 shelly_hap_window_c:544 P = 0.00 -> 139.36
680224986019 shelly_input.cpp:76     Input 1: change (state 0)
680224996114 shelly_hap_window_c:544 P = 0.00 -> 139.36
680225040869 shelly_hap_window_c:544 P = 0.00 -> 139.36
680225140916 shelly_hap_window_c:544 P = 0.00 -> 139.36
680225240872 shelly_hap_window_c:544 P = 0.00 -> 139.36
680225316933 shelly_input.cpp:76     Input 1: single (state 0)
680225340792 shelly_hap_window_c:544 P = 1.26 -> 139.36
680225440964 shelly_hap_window_c:544 P = 0.00 -> 139.36
680225540959 shelly_hap_window_c:544 P = 0.00 -> 139.36
680225641006 shelly_hap_window_c:544 P = 1.27 -> 139.36
680225740806 shelly_hap_window_c:544 P = 1.32 -> 139.36
680225840911 shelly_hap_window_c:544 P = 1.08 -> 139.36
680225940776 shelly_hap_window_c:544 P = 0.00 -> 139.36
**680225949458 shelly_hap_window_c:551 Failed to start moving**
680225959889 shelly_hap_window_c:322 WC 1: State: rampup -> stop (22 -> 24)
680226040066 shelly_output.cpp:58    Output 1: on -> off (stop)
680226069963 mgos_sys_config.c:232   Loading conf2.json
680226215943 mgos_sys_config.c:174   Saved to conf9.json
680227665669 shelly_input.cpp:76     Input 1: change (state 1)
680227676204 shelly_hap_window_c:342 WC 1: Tgt pos 0.00 -> 100.00 (ext)
680227687743 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
680227749701 shelly_output.cpp:58    Output 1: off -> on (move)
680227760269 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
680227850630 shelly_hap_window_c:544 P = 0.00 -> 139.36
680227881819 shelly_input.cpp:76     Input 1: change (state 0)
680227891948 shelly_hap_window_c:544 P = 0.00 -> 139.36
680227950363 shelly_hap_window_c:544 P = 0.00 -> 139.36
680228051112 shelly_hap_window_c:544 P = 1.12 -> 139.36
680228149956 shelly_hap_window_c:544 P = 0.00 -> 139.36
680228197838 shelly_input.cpp:76     Input 1: single (state 0)
680228250411 shelly_hap_window_c:544 P = 0.00 -> 139.36
680228350045 shelly_hap_window_c:544 P = 1.08 -> 139.36
680228450483 shelly_hap_window_c:544 P = 1.14 -> 139.36
680228550596 shelly_hap_window_c:544 P = 1.18 -> 139.36
680228650587 shelly_hap_window_c:544 P = 1.11 -> 139.36
680228750665 shelly_hap_window_c:544 P = 1.04 -> 139.36
680228822799 shelly_main.cpp:554     Up 680228.81, HAP 0/2/12 ns 2, RAM: 18928/37648; st 63; 4.1: c:1 mp:139.36 mt_ms:16189 cp:0.00 tp:100.00 lemd:1 lhmd:0
680228851261 shelly_hap_window_c:544 P = 1.29 -> 139.36
**680228860023 shelly_hap_window_c:551 Failed to start moving**
680228871682 shelly_hap_window_c:322 WC 1: State: rampup -> stop (22 -> 24)
680228949070 shelly_output.cpp:58    Output 1: on -> off (stop)
680228978981 mgos_sys_config.c:232   Loading conf2.json
680229125222 mgos_sys_config.c:174   Saved to conf9.json
680229138266 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
680229156853 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
680229255101 shelly_hap_window_c:322 WC 1: State: idle -> error (0 -> 100)
680229356000 shelly_hap_window_c:342 WC 1: Tgt pos 100.00 -> 0.00 (error)
680229367427 shelly_hap_window_c:322 WC 1: State: error -> idle (100 -> 0)
680230817356 shelly_input.cpp:76     Input 1: change (state 1)
680230827921 shelly_hap_window_c:342 WC 1: Tgt pos 0.00 -> 100.00 (ext)
680230839472 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
[... 37 lines cut ...]
680240956822 shelly_hap_window_c:331 WC 1: Cur pos 56.69 -> 57.31, P = 145.42
680241055956 shelly_hap_window_c:594 Moving to 100, p 145.53
680241756836 shelly_hap_window_c:331 WC 1: Cur pos 61.63 -> 62.25, P = 145.55
680241855989 shelly_hap_window_c:594 Moving to 100, p 145.48
680242556754 shelly_hap_window_c:331 WC 1: Cur pos 66.58 -> 67.19, P = 145.45
680242655504 shelly_hap_window_c:594 Moving to 100, p 145.77
680243356850 shelly_hap_window_c:331 WC 1: Cur pos 71.52 -> 72.14, P = 145.90
680243455995 shelly_hap_window_c:594 Moving to 100, p 146.08
680244156862 shelly_hap_window_c:331 WC 1: Cur pos 76.46 -> 77.08, P = 145.53
680244256015 shelly_hap_window_c:594 Moving to 100, p 145.20
680244822943 shelly_main.cpp:554     Up 680244.81, HAP 0/2/12 ns 2, RAM: 20524/37648; st 63; 4.1: c:1 mp:139.36 mt_ms:16189 cp:80.78 tp:100.00 lemd:1 lhmd:0
680244956736 shelly_hap_window_c:331 WC 1: Cur pos 81.40 -> 82.02, P = 145.82
680245055843 shelly_hap_window_c:594 Moving to 100, p 145.35
680245756978 shelly_hap_window_c:331 WC 1: Cur pos 86.34 -> 86.96, P = 145.17
680245855517 shelly_hap_window_c:594 Moving to 100, p 145.74
[... cut ...]
680247278193 shelly_hap_window_c:322 WC 1: State: moving -> stop (23 -> 24)
680247374249 mgos_sys_config.c:232   Loading conf2.json
680247521655 mgos_sys_config.c:174   Saved to conf9.json
680247533763 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
680247552022 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
680252823174 shelly_main.cpp:554     Up 680252.81, HAP 0/2/12 ns 2, RAM: 20524/37648; st 63; 4.1: c:1 mp:139.36 mt_ms:16189 cp:100.00 tp:100.00 lemd:1 lhmd:0

closed 2021-04-11 by github-actions[bot]