mongoose-os-apps/shelly-homekit #251: Shelly 2.5 in garage door modus , restarting and failed to communicate with ade7953

Pixel-Chris, 2020-11-03:
I still have a problem with my garagedoor setup

Sometimes I open the garage door (via HomeKit) and the door opens normally but the Shelly is not working, I now captured that...

5631900 mgos_mongoose.c:66      New heap free LWM: 23712
450960 mgos_rpc_channel_mq:203 0x3fff390c shellyswitch25-03EEF0/rpc
461205 mgos_rpc_channel_ua:313 0x3fff42fc UART0
482287 mgos_ade7953.c:63       ADE7953 reg read error (0x702)
492415 mgos_ade7953.c:126      Failed to communicate with ADE7953
500616 shelly_pm_ade7953.cp:97 Failed to init ADE7953
509219 shelly_main.cpp:295     === Creating accessories
519862 shelly_hap_garage_do:96 GDO 1: cur_state 1
531110 shelly_main.cpp:319     === Starting HAP server (CN 7)
598689 HAPPlatformTCPStrea:185 Listening on 9000
609236 mgos_mdns.c:82          Listening on udp://:5353
624631 mgos_init.c:33          Init done, RAM: 41104 total, 27440 free, 27336 min free

Pixel-Chris, 2020-11-03:
And then after a few minutes:

cur:closed tgt:closed cl:1 op:0
369689777 shelly_main.cpp:543     Up 369.68, HAP 0/3/9 ns 3, RAM: 25148/41104; st 63; 5.1: cur:closed tgt:closed cl:1 op:0
377689850 shelly_main.cpp:543     Up 377.68, HAP 0/3/9 ns 3, RAM: 25148/41104; st 63; 5.1: cur:closed tgt:closed cl:1 op:0
451705 mgos_rpc_channel_mq:203 0x3fff390c shellyswitch25-03EEF0/rpc
461042 mgos_rpc_channel_ua:313 0x3fff42fc UART0
579902 mgos_ade7953.c:63       ADE7953 reg read error (0x702)
693291 mgos

Shelly restarts again

Pixel-Chris, 2020-11-03:
Can that be again a power problem?

Pixel-Chris, 2020-11-06:
@rojer 
Some ideas?

Pixel-Chris, 2020-11-08:
I captured now 3 more restarts with the beta2, but i think here is not something usable:

1441415399 mgos_sntp.c:95          SNTP query to time.google.com
1448738702 shelly_main.cpp:530     Up 1448.73, HAP 0/2/9 ns 2, RAM: 23548/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1456738904 shelly_main.cpp:530     Up 1456.73, HAP 0/2/9 ns 2, RAM: 23548/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1464738704 shelly_main.cpp:530     Up 1464.73, HAP 0/2/9 ns 2, RAM: 23548/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1472738674 shelly_main.cpp:530     Up 1472.73, HAP 0/2/9 ns 2, RAM: 23784/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1480738787 shelly_main.cpp:530     Up 1480.73, HAP 0/2/9 ns 2, RAM: 23808/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1488739049 shelly_main.cpp:530     Up 1488.73, HAP 0/2/9 ns 2, RAM: 23808/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1496738776 shelly_main.cpp:530     Up 1496.73, HAP 0/2/9 ns 2, RAM: 23808/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1497441696 mgos_sntp.c:95          SNTP query to time.google.com
1504738863 shelly_main.cpp:530     Up 1504.73, HAP 0/2/9 ns 2, RAM: 23572/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1512738752 shelly_main.cpp:530     Up 1512.73, HAP 0/2/9 ns 2, RAM: 23576/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1520738817 shelly_main.cpp:530     Up 1520.73, HAP 0/2/9 ns 2, RAM: 23576/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1528738855 shelly_main.cpp:530     Up 1528.73, HAP 0/2/9 ns 2, RAM: 23808/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1536738695 shelly_main.cpp:530     Up 1536.73, HAP 0/2/9 ns 2, RAM: 23808/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1544738846 shelly_main.cpp:530     Up 1544.73, HAP 0/2/9 ns 2, RAM: 23808/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1552616940 mgos_sntp.c:95          SNTP query to time.google.com
1552738931 shelly_main.cpp:530     Up 1552.73, HAP 0/2/9 ns 2, RAM: 22900/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1560738818 shelly_main.cpp:530     Up 1560.73, HAP 0/2/9 ns 2, RAM: 23576/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
1568738803 shelly_main.cpp:530     Up 1568.73, HAP 0/2/9 ns 2, RAM: 23576/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1576738822 shelly_main.cpp:530     Up 1576.73, HAP 0/2/9 ns 2, RAM: 23576/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0

Restart

...

1376749614 shelly_main.cpp:530     Up 1376.74, HAP 0/3/9 ns 3, RAM: 23040/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1381309190 mgos_sntp.c:95          SNTP query to time.google.com
1384749513 shelly_main.cpp:530     Up 1384.74, HAP 0/3/9 ns 3, RAM: 22812/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1392749503 shelly_main.cpp:530     Up 1392.74, HAP 0/3/9 ns 3, RAM: 22784/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1400749503 shelly_main.cpp:530     Up 1400.74, HAP 0/3/9 ns 3, RAM: 22784/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1408749467 shelly_main.cpp:530     Up 1408.74, HAP 0/3/9 ns 3, RAM: 22784/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1416749494 shelly_main.cpp:530     Up 1416.74, HAP 0/3/9 ns 3, RAM: 23012/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1424749470 shelly_main.cpp:530     Up 1424.74, HAP 0/3/9 ns 3, RAM: 23012/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1432749436 shelly_main.cpp:530     Up 1432.74, HAP 0/3/9 ns 3, RAM: 23012/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1440749461 shelly_main.cpp:530     Up 1440.74, HAP 0/3/9 ns 3, RAM: 23012/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1443251383 mgos_sntp.c:95          SNTP query to time.google.com
1443270658 mgos_mongoose.c:66      New heap free LWM: 17088
1443287517 mgos_mongoose.c:66      New heap free LWM: 16360
1448749492 shelly_main.cpp:530     Up 1448.74, HAP 0/3/9 ns 3, RAM: 22808/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1456749566 shelly_main.cpp:530     Up 1456.74, HAP 0/3/9 ns 3, RAM: 22832/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1464749409 shelly_main.cpp:530     Up 1464.74, HAP 0/3/9 ns 3, RAM: 22832/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1472749505 shelly_main.cpp:530     Up 1472.74, HAP 0/3/9 ns 3, RAM: 23064/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
1480749484 shelly_main.cpp:530     Up 1480.74, HAP 0/3/9 ns 3, RAM: 23064/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0

Restart

cur:closed tgt:closed cl:1 op:0
328690014 shelly_main.cpp:530     Up 328.68, HAP 0/1/9 ns 1, RAM: 24412/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
332207603 mgos_sntp.c:95          SNTP query to time.google.com
336689955 shelly_main.cpp:530     Up 336.68, HAP 0/1/9 ns 1, RAM: 24184/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
344689929 shelly_main.cpp:530     Up 344.68, HAP 0/1/9 ns 1, RAM: 24184/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
352689997 shelly_main.cpp:530     Up 352.68, HAP 0/1/9 ns 1, RAM: 24184/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
360689913 shelly_main.cpp:530     Up 360.68, HAP 0/1/9 ns 1, RAM: 24184/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
368690024 shelly_main.cpp:530     Up 368.68, HAP 0/1/9 ns 1, RAM: 24412/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
376689990 shelly_main.cpp:530     Up 376.68, HAP 0/1/9 ns 1, RAM: 24412/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
384689852 shelly_main.cpp:530     Up 384.68, HAP 0/1/9 ns 1, RAM: 24412/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
390779306 mgos_sntp.c:95          SNTP query to time.google.com
392689788 shelly_main.cpp:530     Up 392.68, HAP 0/1/9 ns 1, RAM: 24184/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0
400689880 shelly_main.cpp:530     Up 400.68, HAP 0/1/9 ns 1, RAM: 24184/41024; st 62; 5.1: cur:closed tgt:closed cl:1 op:0

Restart

Pixel-Chris, 2020-11-08:
next one

72122909 mgos_sntp.c:95          SNTP query to time.google.com
272680928 shelly_main.cpp:530     Up 272.67, HAP 0/0/9 ns 0, RAM: 25292/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
280680845 shelly_main.cpp:530     Up 280.67, HAP 0/0/9 ns 0, RAM: 25292/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
288681001 shelly_main.cpp:530     Up 288.67, HAP 0/0/9 ns 0, RAM: 25292/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
296680796 shelly_main.cpp:530     Up 296.67, HAP 0/0/9 ns 0, RAM: 25292/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
304680673 shelly_main.cpp:530     Up 304.67, HAP 0/0/9 ns 0, RAM: 25528/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
312680662 shelly_main.cpp:530     Up 312.67, HAP 0/0/9 ns 0, RAM: 25528/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
317027857 HAPPlatformTCPStrea:120 0x3fff5e4c 192.168.111.224:50193 Incoming HAP connection, ns 0/0/9
317042435 HAPPlatformTCPStrea:274 0x3fff5e4c 192.168.111.224:50193 Accepted HAP connection, ns 0/1/9 ts 0x3fff5524
320680676 shelly_main.cpp:530     Up 320.67, HAP 0/1/9 ns 1, RAM: 25036/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0

restart

Pixel-Chris, 2020-11-08:
can there be a problem, when the Shelly tries to query time.google.com with SNTP, but internet connections are blocked?

Pixel-Chris, 2020-11-08:
and is that:
482287 mgos_ade7953.c:63 ADE7953 reg read error (0x702)
492415 mgos_ade7953.c:126 Failed to communicate with ADE7953
500616 shelly_pm_ade7953.cp:97 Failed to init ADE7953

normal for the garage door mode ??

ADE7953 is for the power metering right?

Pixel-Chris, 2020-11-08:
and again a restart...

216690896 shelly_main.cpp:530     Up 216.68, HAP 0/0/9 ns 0, RAM: 24544/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
224690926 shelly_main.cpp:530     Up 224.68, HAP 0/0/9 ns 0, RAM: 24544/41024; st 60; 5.1: cur:closed tgt:closed cl:1 op:0
232690771 shelly_main.cpp:530     Up 232.68, HAP 0/0/9 ns 0, RAM: 24544/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
240690838 shelly_main.cpp:530     Up 240.68, HAP 0/0/9 ns 0, RAM: 24544/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
248690774 shelly_main.cpp:530     Up 248.68, HAP 0/0/9 ns 0, RAM: 24776/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
256690684 shelly_main.cpp:530     Up 256.68, HAP 0/0/9 ns 0, RAM: 24776/41024; st 60; 5.1: cur:closed tgt:closed cl:1 op:0
264690732 shelly_main.cpp:530     Up 264.68, HAP 0/0/9 ns 0, RAM: 24776/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
272690740 shelly_main.cpp:530     Up 272.68, HAP 0/0/9 ns 0, RAM: 24776/41024; st 61; 5.1: cur:closed tgt:closed cl:1 op:0
273154972 mgos_sntp.c:95          SNTP query to time.google.com

right after the log entry "SNTP query to time.google.com"

Pixel-Chris, 2020-11-11:
No one any idea?

Andre000Dias, 2020-11-12:
shelly 2.5 garage door? can you put a diagram with the connections? have sensor?ths

Pixel-Chris, 2020-11-12:
[image]

marcus-aa, 2020-11-17:
Hi,

I have also reboots on 2.5 in Garage door opener mode, currently just in test but unfortunately not useable, as it will open the gate on every reboot...

```
8776997857 shelly_main.cpp:530     Up 8776.99, HAP 0/4/9 ns 4, RAM: 23088/41024; st 64; 5.1: cur:closed tgt:closed cl:1 op:0
8784997742 shelly_main.cpp:530     Up 8784.99, HAP 0/4/9 ns 4, RAM: 23088/41024; st 64; 5.1: cur:closed tgt:closed cl:1 op:0
8792998193 shelly_main.cpp:530     Up 8792.99, HAP 0/4/9 ns 4, RAM: 23088/41024; st 64; 5.1: cur:closed tgt:closed cl:1 op:0
8800997850 shelly_main.cpp:530     Up 8800.99, HAP 0/4/9 ns 4, RAM: 23088/41024; st 64; 5.1: cur:closed tgt:closed cl:1 op:0
8808997809 shelly_main.cpp:530     Up 8808.99, HAP 0/4/9 ns 4, RAM: 23088/41024; st 64; 5.1: cur:closed tgt:closed cl:1 op:0
8816997927 shelly_main.cpp:530     Up 8816.99, HAP 0/4/9 ns 4, RAM: 23088/41024; st 64; 5.1: cur:closed tgt:closed cl:1 op:0
453306 mgos_rpc_channel_mq:203 0x3fff395c shellyswitch25-FA8D06/rpc
465459 mgos_rpc_channel_ua:313 0x3fff434c UART0
488957 mgos_ade7953.c:96       ADE7953 silicon version: 0x02 (2)
522470 shelly_main.cpp:301     === Creating accessories
534134 shelly_hap_garage_do:96 GDO 1: cur_state 1
544461 shelly_main.cpp:325     === Starting HAP server (CN 5)
612491 HAPPlatformTCPStrea:185 Listening on 9000
623158 mgos_mdns.c:82          Listening on udp://:5353
664166 mgos_init.c:33          Init done, RAM: 41024 total, 26792 free, 26808 min free
678849 shelly_main.cpp:359     HAP server state: 1
687590 mgos_mongoose.c:66      New heap free LWM: 26544
696816 mgos_mdns.c:55          mDNS socket closed
706433 mgos_mdns.c:82          Listening on udp://:5353
721617 mgos_ota_core.c:1329    UID: 036d060f7c2f7235, license: none
736243 mgos_mongoose.c:66      New heap free LWM: 25480
744945 mgos_mdns.c:55          mDNS socket closed
754563 mgos_mdns.c:82          Listening on udp://:5353
765020 mgos_net.c:86           WiFi STA: connecting
3413840 esp_main.c:158          SDK: scandone
4306245 esp_main.c:158          SDK: state: 0 -> 2 (b0)
4322606 esp_main.c:158          SDK: state: 2 -> 3 (0)
4337077 esp_main.c:158          SDK: state: 3 -> 5 (10)
4345635 esp_main.c:158          SDK: add 0
4353081 esp_main.c:158          SDK: aid 6
4360720 esp_main.c:158          SDK: cnt 
4383479 esp_main.c:158          SDK: 
4393348 esp_main.c:158          SDK: connected with Akesson, channel 1
4403311 esp_main.c:158          SDK: dhcp client start...
4418676 mgos_wifi.c:133         WiFi STA: Connected, BSSID 02:27:22:fb:39:a9 ch 1 RSSI -65
4429989 mgos_net.c:90           WiFi STA: connected
4461453 esp_main.c:158          SDK: ip:192.168.40.180,mask:255.255.255.0,gw:192.168.40.1
4479281 mgos_net.c:101          WiFi STA: ready, IP 192.168.40.180, GW 192.168.40.1, DNS 192.168.84.103
4493392 mgos_mongoose.c:66      New heap free LWM: 24592
5430914 mgos_sntp.c:95          SNTP query to time.google.com
5444417 mgos_mongoose.c:66      New heap free LWM: 23080
5497479 mgos_mongoose.c:66      New heap free LWM: 22672
5561350 mgos_sntp.c:58          SNTP reply from 216.239.35.12: time 1605620526.128519, local 5.548108, delta 1605620520.580411
8642214:1605620529205008 shelly_main.cpp:530     Up 8.63, HAP 0/0/9 ns 0, RAM: 26244/41024; st 64; 5.1: cur:closed tgt:closed cl:1 op:0
11132731 mgos_http_server.c:180  0x3fff5bfc HTTP connection from 192.168.40.134:53911
11492665 shelly_debug.cpp:203    Tailing
14361500 esp_main.c:158          SDK: pm open,type:0 0
16641923 shelly_main.cpp:530     Up 16.63, HAP 0/0/9 ns 0, RAM: 25160/41024; st 64; 5.1: cur:closed tgt:closed cl:1 op:0
24643265 shelly_main.cpp:530     Up 24.63, HAP 0/0/9 ns 0, RAM: 25168/41024; st 64; 5.1: cur:closed tgt:closed cl:1 op:0
32642248 shelly_main.cpp:530     Up 32.63, HAP 0/0/9 ns 0, RAM: 25168/41024; st 65; 5.1: cur:closed tgt:closed cl:1 op:0
40642114 shelly_main.cpp:530     Up 40.63, HAP 0/0/9 ns 0, RAM: 25168/41024; st 65; 5.1: cur:closed tgt:closed cl:1 op:0
```

konagar, 2020-11-17:
Hello, I'm not sure if I really understand the problem. My English is not that good. My Shelly 2.5 reboots more often when switching. They are not stable at 24 volts. I increased the voltage with an additional power supply unit. After that there were no more reboots.

marcus-aa, 2020-11-17:
> Hello, I'm not sure if I really understand the problem. My English is not that good. My Shelly 2.5 reboots more often when switching. They are not stable at 24 volts. I increased the voltage with an additional power supply unit. After that there were no more reboots.

I think you are right. I had no problem when using 230VAC, now after a while on 24VDC it got stuck in a boot loop. Increased voltage to 30VDC and it started, will see if it is stable.

Problem is that automation and gate control is running at 24VDC...

Pixel-Chris, 2020-12-05:
@marcus-aa 
Is it stable with 30VDC ?

marcus-aa, 2020-12-15:
Yes it is stable all the way down to 25VDC. But around 24V things start to happen. At 23.5 volts this fw is in a boot loop, the stoch fw is OK, but reboots as soon as a relay activates. As the stock fw is also affected this is most likely a hw issue and I cant see any way that a 2.5 can ever work with 20V as stated in the specs.

So the critical point is somewhere exactly around 24V, which is way too poor margin. Systems that run 24V backup batteries will for example vary from 27 to 23V.

[image]

Pixel-Chris, 2020-12-15:
Specs say 24-60V:

[image]

Since 2.61 less reboots in my wiring with 24V
So i will upgrade to 25 or 26V
Or I will wait until the Shelly 1 supports the AddOn #186

marcus-aa, 2020-12-31:
> specs say 24-60V:

I have seen different limits, one says 30-50V.

But anyway, it is not stable on 24V either and with reboots on 23.5 it is not strange, there is too little margin.

andyblac, 2021-03-03:
can this be closed ?

Pixel-Chris, 2021-03-03:
Hi,
when it is ok, to power it with more than 24 V.
Then it can be closed.

andyblac, 2021-03-03:
@rojer can you advise?

Pixel-Chris, 2021-03-03:
Ist should be okay
Specs das 24-60V

closed 2021-04-11 by github-actions[bot]

rojer, 2021-10-16:
no, this change is a no-op as we don't actually use ADE's interrupt function.
this pin was unconfigured before and, as it happens, it defaults to output mode (the only GPIO to do so). so we had output connected to outuput, which is a no-no - it sinks current and wastes energy. it may even fry one of the chips, but looks like in this case this does not happen.

did you start getting this error after the update? it wasn't happening on 2.10.0?

Pixel-Chris, 2021-10-16:
@rojer 

> did you start getting this error after the update? it wasn't happening on 2.10.0?

No, not jet updated.

rojer, 2021-10-16:
ok. so no, the update is unlikely to fix it.

safepay, 2023-01-16:
I had a Shelly 2.5 reboot with the 2.11.2 firmware. Whenever the web page was reloaded or anything ws done via the web screen. I couldn't use it. I was using the 24V output from the garage door motor.

The access point was in the next room.

Then, by chance, I installed an access point in the garage with line of sight to the Shelly 2.5. And guess what? No more reboots.

Probably related to power consumption, but this fixed it for me and I can still use the Shelly from the Garage Motor 24v power supply.