mongoose-os-apps/shelly-homekit #254: Issue: Roller Shutter automations opening instead of closing.

pivale, 2020-11-06:
Hello there,

I have 5 roller shutters configured using shelly's 2.5.
They are all using momentary input configuration and I have push buttons on all.

[image]

I also have an automation to close my shutters after sunset on the bedrooms and home office.

[image]

It is "usually" working really well, but from time to time, some random shutters open instead of closing.
I haven't managed to snag any logs, because when it happens, I only notice later and then the logs are gone... But I will try to get them and post here.

Maybe some internal state gets mixed up?
Or a homekit bug?

MilanBacaCz, 2020-11-06:
I have same problem.

S pozdravem,
Milan Bača

rojer, 2020-11-06:
logs should have all the info needed to fix this, the only question is how to capture them.
we need a script that saves them at a regular interval, say every 5 minutes.
@andyblac perhaps you can put one together?

andyblac, 2020-11-06:
is there a command line we use to grab logs from devcies, i was thinking maybe a cron job on mac / linux to pull log from device every 10 mins ?

nliviu, 2020-11-06:
```
curl http://ip_or_name/rpc/FS.List
["index.html.gz","style.css.gz","axios.min.js.gz","conf0.json","favicon.ico","logo.png","conf2.json","kvs.json","conf9.json","log_080-20201106-123249.log"]
curl http://ip_or_name/rpc/FS.Get -d '{"filename":"log_080-20201106-123249.log"}'
{"data": "MjM0MzMzNzYzNTY0IHNoZW...mlhIEhUVFAgMTkyLjE2OC4xLjExNzo1NTQ1NAo=", "left": 0}

```
The `data` part is base64 encoded content of the log file.

rojer, 2020-11-06:
curl http://device.local/debug/log

yuvalbeck, 2020-11-06:
Same problem here

MilanBacaCz, 2020-11-06:
It happened to me just now.

288480435978 shelly_hap_window_c:315 WC 1: Cur pos 83.88 -> 84.13, P = 105.71
288480444854 shelly_hap_window_c:561 Moving to 100, p 105.71
288481163607 shelly_hap_window_c:315 WC 1: Cur pos 85.84 -> 86.09, P = 105.71
288481172921 shelly_hap_window_c:561 Moving to 100, p 105.71
288481963646 shelly_hap_window_c:315 WC 1: Cur pos 87.80 -> 88.05, P = 106.05
288482019689 shelly_hap_window_c:561 Moving to 100, p 106.05
288482763514 shelly_hap_window_c:315 WC 1: Cur pos 89.76 -> 90.01, P = 105.60
288482772826 shelly_hap_window_c:561 Moving to 100, p 105.60
288483563778 shelly_hap_window_c:315 WC 1: Cur pos 91.73 -> 91.97, P = 105.77
288483573096 shelly_hap_window_c:561 Moving to 100, p 105.77
288484363532 shelly_hap_window_c:315 WC 1: Cur pos 93.69 -> 93.93, P = 105.29
288484373843 shelly_hap_window_c:561 Moving to 100, p 105.29
288485163590 shelly_hap_window_c:315 WC 1: Cur pos 95.65 -> 95.90, P = 104.96
288485172915 shelly_hap_window_c:561 Moving to 100, p 104.96
288485963526 shelly_hap_window_c:315 WC 1: Cur pos 97.61 -> 97.86, P = 105.20
288485973837 shelly_hap_window_c:561 Moving to 100, p 105.20
288486663860 shelly_hap_window_c:315 WC 1: Cur pos 99.57 -> 100.00, P = 4.46
288486719712 shelly_output.cpp:56    Output 1: on -> off (moving)
288486730485 shelly_hap_window_c:306 WC 1: State: moving -> stop (23 -> 24)
288486784117 mgos_sys_config.c:232   Loading conf2.json
288486808067 mgos_sys_config.c:232   Loading conf3.json
288486940703 mgos_sys_config.c:174   Saved to conf9.json
288486957786 shelly_hap_window_c:306 WC 1: State: stop -> stopping (24 -> 25)
288486981086 shelly_hap_window_c:306 WC 1: State: stopping -> idle (25 -> 0)
288487869250 shelly_main.cpp:543     Up 288487.86, HAP 0/4/9 ns 4, RAM: 23460/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288495869322 shelly_main.cpp:543     Up 288495.86, HAP 0/4/9 ns 4, RAM: 23584/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288503869332 shelly_main.cpp:543     Up 288503.86, HAP 0/4/9 ns 4, RAM: 23584/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288511869138 shelly_main.cpp:543     Up 288511.86, HAP 0/4/9 ns 4, RAM: 23584/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288519869064 shelly_main.cpp:543     Up 288519.86, HAP 0/4/9 ns 4, RAM: 23460/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288527160732 HAPPlatformTCPStrea:104 0x3fff676c 192.168.0.139:49243 HAP connection closed, ns 0/4/9 i 2
288527172535 HAPPlatformTCPStrea:301 HAPPlatformTCPStreamClose ts 0x3fff2b1c nc 0x0 0/3/9
288527869581 shelly_main.cpp:543     Up 288527.86, HAP 0/3/9 ns 3, RAM: 23872/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288534931748 HAPPlatformTCPStrea:120 0x3fff699c 192.168.0.12:49406 Incoming HAP connection, ns 0/3/9
288534949802 HAPPlatformTCPStrea:274 0x3fff699c 192.168.0.12:49406 Accepted HAP connection, ns 0/4/9 ts 0x3fff2b1c
288536090292 shelly_main.cpp:543     Up 288536.08, HAP 0/4/9 ns 4, RAM: 22820/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288543868948 shelly_main.cpp:543     Up 288543.86, HAP 0/4/9 ns 4, RAM: 23476/41104; st 71; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288551868944 shelly_main.cpp:543     Up 288551.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288559868928 shelly_main.cpp:543     Up 288559.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288567868905 shelly_main.cpp:543     Up 288567.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288575868906 shelly_main.cpp:543     Up 288575.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288583869034 shelly_main.cpp:543     Up 288583.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288591868920 shelly_main.cpp:543     Up 288591.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288599868999 shelly_main.cpp:543     Up 288599.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288607868870 shelly_main.cpp:543     Up 288607.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288615868885 shelly_main.cpp:543     Up 288615.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288623869022 shelly_main.cpp:543     Up 288623.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288631868835 shelly_main.cpp:543     Up 288631.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288639868983 shelly_main.cpp:543     Up 288639.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288647868891 shelly_main.cpp:543     Up 288647.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288655868818 shelly_main.cpp:543     Up 288655.86, HAP 0/4/9 ns 4, RAM: 23580/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
[... cut ...]
288671849542 mgos_mongoose.c:66      New heap free LWM: 13440
288671877557 shelly_main.cpp:543     Up 288671.86, HAP 0/4/9 ns 4, RAM: 13840/41104; st 70; 4.1: c:1 mp:100.98 mt_ms:40781 cp:100.00 tp:100.00 lemd:1
288672092157 mgos_http_server.c:180  0x3fff66f4 HTTP connection from 192.168.0.140:59577
288672106678 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.0.140:59577
288672146664 mgos_http_server.c:180  0x3fff58ec HTTP connection from 192.168.0.140:59578
288672208257 mgos_http_server.c:187  0x3fff58ec GET /favicon.ico

rojer, 2020-11-06:
@MilanBacaCz in this log i see tail end of curtain movement to the fully open position.
unfortunately the log doesn't capture the start of the movement so if it was supposed to close instead, we don't see the state it was in when the command was sent.

pivale, 2020-11-06:
>  we need a script that saves them at a regular interval, say every 5 minutes.
> @andyblac perhaps you can put one together?

based on the log from MilanBacaCz, is every 5min enough?
it might be hard not to miss data...

Is there any way to write the log on a remote location instead of reading it from time to time?
Or even having something similar to 'tail -f' but for an http request like http://device.local/debug/log

andyblac, 2020-11-06:
> based on the log from MilanBacaCz, is every 5min enough?
> it might be hard not to miss data...
>
> Is there any way to write the log on a remote location instead of reading it from time to time?
> Or even [...]

ca you not do it via this way https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#console-logs

rojer, 2020-11-06:
> Or even having something similar to 'tail -f' but for an http request like http://device.local/debug/log

not a bad idea, i will add that.

rojer, 2020-11-06:
http://rojer.me/files/shelly/2.5.0-beta1/ has this enabled
`http://DEVICE/debug/log?follow=1` will serve last log file (if enabled) and tail for new records.

pivale, 2020-11-07:
Hey @rojer,

Just installed beta and tested it and noticed two things:

1. Close button on the physical switch stopped working for all my roller shutters (all 4 of them). I open on the physical switch but after I need to use the app to close it (or webui).

2. I've managed to replicate the situation where it opens instead of closing. Here is how I managed:
Step 1: Shutter is fully closed
Step 2: Open in physical switch and stop with same button
Step 3: In home app, use the slider to close (it starts to open) <- This is where the problem lies
Step 4: Press in home app the tile to stop
Step 5: In home app, use the slider to close (it starts to close now)

Let me know if you need further clarification, I've managed to replicate both whenever I want.

Please check logs attached with both situations, I hope you have there the necessary information.
[shellyswitch25-escritorio-persiana.log](https://github.com/mongoose-os-apps/shelly-homekit/files/5505631/shellyswitch25-escritorio-persiana.log)

rojer, 2020-11-08:
@pivale thank you for the excellent report! along with the log it made both problems clear.
i submitted fixes ([1](https://github.com/mongoose-os-apps/shelly-homekit/commit/085f3295057dee7d194664c5501d335d3da8d4c3) and [2](https://github.com/mongoose-os-apps/shelly-homekit/commit/b24fa37a5fed3a319e13c61090279358a107f6e5)) and pushed 2.5.0-beta2. please upgrade (for devices running beta1 button in ui should be enough, for other devices files are [here](http://rojer.me/files/shelly/2.5.0-beta2/)).

pivale, 2020-11-08:
That was fast!!
Thank you so much. 

I've already installed and let you know

pivale, 2020-11-09:
Hey @rojer,

Regarding 1, the buttons are working as intended.
Regarding 2, yesterday was the first time the automation closed all correctly. I think I will need more time to make sure, but until now all working great 😄

raphtabe, 2020-11-11:
Hello,

I had the same issue, and it was solved thanks to 2.5 update, thanks.
I still have a little minor issue : 
During a closing automation some store are showing opening in place of closing. They all are closing actually, only the label is wrong. 
When done, all is ok. 
[image]
[image]

pivale, 2020-11-12:
All working great for me, at least from the end result experience: all closed when expected. 
But I'll try to confirm if I experience the same issue @raphtabe is seeing.

rojer, 2020-11-12:
@raphtabe can you confirm that this is indeed from activation by automation and not the result of tapping the tile by hand?
in the latter case, display label can go out of sync due to the cycling mode implementation, and it is normal.

raphtabe, 2020-11-12:
Yes, just tested 2 cicle of manual opening and closing : all is normal.
The error occurs only with an automation.

raphtabe, 2020-11-12:
I've juste tried to reproduce the issue with automation, without success...

SilvioMigu, 2020-12-04:
Hello good afternoon
Is there a connection scheme for the shelly 2.5 on the Roller Shutter, of the same kind that exists for the garage with the shelly 1?

ivofernandespt, 2020-12-23:
Similar issue here:
I have an automation in HomeKit when window open (magnetic sensor), open blind and when window close, close the blind. It works as expected except if when it is opening or closing I close or open the window:
If bling if opening but I close the window I want that blind starts closing but what happens is that blind stops and if open window again it will close... 

Yet with stock firmware + home bridge it works good: if blind is opening and I close the window it immediately starts closing...

ivofernandespt, 2020-12-23:
Today I'm facing an other issue: if I ask Siri to close shade and when the shade is closing I ask to open it stops and I have to ask to open again... something that doesn't happen with stock firmware + Homebriedge.

I've tried all setting in Shelly. Can anyone help? I don't know if there is a setting problem or a firmware bug.

andyblac, 2021-01-25:
still an issue?

pivale, 2021-01-25:
The original problem was solved.
Not sure if anyone else still has problems.

andyblac, 2021-01-25:
ok, I'll close as it has not been updated in a month, and as the OP is happy issue is sorted.

closed 2021-01-25 by andyblac