mongoose-os-apps/shelly-homekit #255: New Feature: Use GPIO-Pins for the garage door opener contact sensor

NotClear, 2020-11-06:
Hey there,

To use a switch and the contact sensor at the same time, connect the switch to the Shelly as usual and connect the contact sensor via the GPIO pins.
like this: https://www.youtube.com/watch?v=dLNxpc3ausA&t=53s

With my garage door motor there is only one connection for one switch and so it would be possible to connect the contact sensor and a switch at the same time.

Pixel-Chris, 2020-11-06:
See the Wiki...
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Garage-Door-Opener-Settings

For one contact sensor you will not need the gpio pins.

NotClear, 2020-11-06:
And where i connect the switch ?

NotClear, 2020-11-06:
> See the Wiki...
> https://github.com/mongoose-os-apps/shelly-homekit/wiki/Garage-Door-Opener-Settings
> 
> For one contact sensor you will not need the gpio pins.

And where i connect the switch?

Pixel-Chris, 2020-11-06:
[image]

Pixel-Chris, 2020-11-06:
Shelly pins: 0+1

NotClear, 2020-11-06:
> Shelly pins: 0+1

if I understand correctly I connect pins 0+1 to the switch input
 from the garage door controler but what about my physical switch

NotClear, 2020-11-06:
#

Pixel-Chris, 2020-11-06:
parallel wiring

Pixel-Chris, 2020-11-06:
Your physical switch is connected to the garage door opener board 

Now you connect the Shelly to the same pins on your garage door board

NotClear, 2020-11-06:
> Your physical switch is connected to the garage door opener board
> 
> Now you connect the Shelly to the same pins on your garage door board

ah ok i understand thanks for your help i will try it out right away.

closed 2020-11-06 by NotClear