mongoose-os-apps/shelly-homekit #257: How to debug problems with Shelly and Homekit.

Tomer51m, 2020-11-07:
Hi guys, I was running into my first problem this week with Shelly 1 that controls my boiler. I have an automation in my Apple Home app to turn on 
the boiler every day at 5 am for one hour. 

The automation worked for a while (over two months) in the past few days, it doesn’t seem to work. 

Of course I can babysit my boiler like an animal 😂, bit I figured this would be an excellent time to learn from the pro’s how to debug problems with Shelly and HomeKit. 

I believe the log in the Shelly interface can be useful, but would love to hear from @rojer and others best practices. 

Thanks! 🙏

rojer, 2020-11-07:
absolutely, capturing log around the even is the first thing to do.

Tomer51m, 2020-11-07:
Does the log records only when changes happen?
can I turn it on before going to bed and watch what happened during the night?

rojer, 2020-11-08:
at info level (the default) usually we log "important" things when they happen, such as inputs changing state or e.g. roller shutter state transitions. it's subjective, but i try to put things that would help me understand what's going on at info level. there's also a "heartbeat" - every 8 seconds we log an overall status of the system.

yes, if you open the page and leave it open, it should continue to receive events as they come. since storage space on the device is extremely limited, we only store 10000 bytes of log, so that's what you get when you first open the page.

rojer, 2020-11-08:
oh, i should note that this continuous logging behavior was added recently and is only available in beta for now. please install 2.5.0-beta2 from here - http://rojer.me/files/shelly/2.5.0-beta2/

Tomer51m, 2020-11-09:
Hi, thanks for the explanation!

so I collected the logs from tonight. I should mention that the shelly was **on** when I started the log and it turned off after about an hour (so I was expecting to see a change of state to "off" somewhere in the log, but I don't see it).

Since you mentioned there is not a lot of storage on the device, the reason could be the length of time in which  the log was taking place? - around 5 hours.

**just to clarify - I believe that if I reset the device and reconfigure, it will work again, I just want to take advantage of the "opportunity" to learn and give other users insights from my problem.**

Two insights I have from the past few nights:
1. Perhaps a time stamp would be useful? 
2. Maybe have two types of logs? 
   a. User oriented log in which he can see time stamps with state changes, wifi signal strengh, ip connections and other user related data. This way can record data for longer periods of time (?)
   b. Technical log with system information ram, and other "important" 😆 data for development and bug fixes. 

Here's the log:
237039018577 shelly_main.cpp:543     Up 237038.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237046952445 shelly_main.cpp:543     Up 237046.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237054952431 shelly_main.cpp:543     Up 237054.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237062952439 shelly_main.cpp:543     Up 237062.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237070952502 shelly_main.cpp:543     Up 237070.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237078952911 shelly_main.cpp:543     Up 237078.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237086952493 shelly_main.cpp:543     Up 237086.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237094952499 shelly_main.cpp:543     Up 237094.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237102952389 shelly_main.cpp:543     Up 237102.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237110952592 shelly_main.cpp:543     Up 237110.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237118952671 shelly_main.cpp:543     Up 237118.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237126952525 shelly_main.cpp:543     Up 237126.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237134952449 shelly_main.cpp:543     Up 237134.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237142952471 shelly_main.cpp:543     Up 237142.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237150952429 shelly_main.cpp:543     Up 237150.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237158952634 shelly_main.cpp:543     Up 237158.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237166952474 shelly_main.cpp:543     Up 237166.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237174952450 shelly_main.cpp:543     Up 237174.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237182952800 shelly_main.cpp:543     Up 237182.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237190952560 shelly_main.cpp:543     Up 237190.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237198952491 shelly_main.cpp:543     Up 237198.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237206952477 shelly_main.cpp:543     Up 237206.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237214952659 shelly_main.cpp:543     Up 237214.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237222952461 shelly_main.cpp:543     Up 237222.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237230952404 shelly_main.cpp:543     Up 237230.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237238952623 shelly_main.cpp:543     Up 237238.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237246952514 shelly_main.cpp:543     Up 237246.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237254952736 shelly_main.cpp:543     Up 237254.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237262952417 shelly_main.cpp:543     Up 237262.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237270952366 shelly_main.cpp:543     Up 237270.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237278952440 shelly_main.cpp:543     Up 237278.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237286952473 shelly_main.cpp:543     Up 237286.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237294952541 shelly_main.cpp:543     Up 237294.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237302952526 shelly_main.cpp:543     Up 237302.94, HAP 0/2/9 ns 2, RAM: 25516/41312; st 0; 0.1: st:0 in_st:0 inm:2
237310400812 HAPPlatformTCPStrea:120 0x3fff4afc 192.168.22.58:64349 Incoming HAP connection, ns 0/2/9
237310414953 HAPPlatformTCPStrea:274 0x3fff4afc 192.168.22.58:64349 Accepted HAP connection, ns 0/3/9 ts 0x3fff4564
237311505992 shelly_main.cpp:543     Up 237311.50, HAP 0/3/9 ns 3, RAM: 24220/41312; st 0; 0.1: st:0 in_st:0 inm:2
237318952421 shelly_main.cpp:543     Up 237318.94, HAP 0/3/9 ns 3, RAM: 24832/41312; st 0; 0.1: st:0 in_st:0 inm:2
[... cut ...]
237337698657 mgos_http_server.c:187  0x3fff652c GET /axios.min.js.gz
237339155562 mg_rpc.c:292            Shelly.GetInfo via HTTP 192.168.22.58:64499
237340923223 mgos_http_server.c:180  0x3fff6674 HTTP connection from 192.168.22.58:64584
237340975401 mgos_http_server.c:187  0x3fff6674 GET /favicon.ico
237342952806 shelly_main.cpp:543     Up 237342.94, HAP 0/3/9 ns 3, RAM: 22980/41312; st 0; 0.1: st:0 in_st:0 inm:2
237350952589 shelly_main.cpp:543     Up 237350.94, HAP 0/3/9 ns 3, RAM: 22980/41312; st 0; 0.1: st:0 in_st:0 inm:2

Tomer51m, 2020-11-10:
Hi @rojer, any remarks?

andyblac, 2021-01-25:
@rojer any updates ?

andyblac, 2021-01-25:
@Tomer51m can you please report if still an issue in 2.7.3

andyblac, 2021-01-25:
my thoughts are that automation issues are normally down to HomeKit hub issues.

Tomer51m, 2021-02-03:
> @Tomer51m can you please report if still an issue in 2.7.3

Hi, I had disconnection that was solved by cutting the power to the device, and turning it back on - this is before I updated to 2.7.3.

Other than that, automations work flawlessly in the past couple of months.

My only issues are random disconnections from wifi here and there (one device out of nine, every 2-3 months) - I have static IP's, 5Ghz turned off, and auto channels switching off.
Power reboot solves is, but it mean turning of many devices in my house (again - before updating to 2.7.3).

Overall very happy!

closed 2021-03-03 by andyblac