mongoose-os-apps/shelly-homekit #259: Make SNTP Server changeable

Pixel-Chris, 2020-11-08:
Hello, I would like to have an option, where I can change the SNTP server to my local one (router).
Is that possible?

Best regards PixelChris

andyblac, 2020-11-08:
ye, this would be nice, don't really like using `time.google.com`

buckmelanoma, 2020-11-15:
Agreed.  As a workaround I have time.google.com set to a local NTP server on my local DNS server.

rojer, 2020-11-16:
the change should made upstream, mongoose os should be getting SNTP server form DHCP lease, if present

buckmelanoma, 2020-11-16:
Thanks!  I added my local server as a DHCP option and can confirm it gets picked up and used (chronyc clients).

Pixel-Chris, 2020-11-17:
Not all routers support to set an additional dhcp option... you can only enable a time server but you need to configure the clients to use this.

rojer, 2021-09-14:
this is now done upstream, will be in the next release.
upstream changes:

https://github.com/mongoose-os/esp-idf/commit/b708237311b8be74ca1b34bfd1e5915e1192a64f
https://github.com/cesanta/mongoose-os/commit/815aaffe2b19a4e5db9147f20b90dabb6b25b4df
https://github.com/mongoose-os-libs/sntp/commit/77765613c8720d5a22be23bd410d1465c4606581
https://github.com/mongoose-os-libs/lwip/commit/2cff4ac32b5c9c4d4481123d99b29c59a84afc6c

closed 2021-09-14 by rojer