mongoose-os-apps/shelly-homekit #260: Automations are not triggered

jobl1978, 2020-11-08:
I bought two Shelly1 and flashed them to 2.4.0. One in edged mode and one in momentary mode. Everything is working fine. I can switch on and off from Home App and with physical switches. In my last test I created a automation where light two should be turned on automatically if light one is turned on. I turned light one on but light two doesn’t turn on. It looks that light two doesn’t receive an event from HomeKit. I tried another automation (turn light on at 5pm) and it also did not recognize to turn on the light. If I program this scenarios only with philips hue lights it works fine.

andyblac, 2020-11-08:
no issues here with shelly homekit and automations. what are you using as a hub?, homepod or appletv ?, i had nothing but issue when using AppleTV's as Hubs.

anyway, it sounds like HomeKit issue not firmware.

jobl1978, 2020-11-08:
> no issues here with shelly homekit and automations. what are you using as a hub?, homepod or appletv ?, i had nothing but issue when using AppleTV's as Hubs.
> 
> anyway, it sounds like HomeKit issue not firmware.

I use an Apple TV HD as hub

andyblac, 2020-11-08:
> > no issues here with shelly homekit and automations. what are you using as a hub?, homepod or appletv ?, i had nothing but issue when using AppleTV's as Hubs.
> > anyway, it sounds like HomeKit issue not firmware.
> 
> I use an Apple TV HD as hub

try rebooting it, but i would say thats your issue. they are notorious for being bad hubs. do yourself a favour, try a homepod mini.

jobl1978, 2020-11-08:
> try rebooting it, but i would say thats your issue. they are notorious for being bad hubs. do yourself a favour, try a homepod mini.

I already rebooted it. Didn’t solve my problem. It’s my plan to buy a HomePod mini when they will be available here starting 16th of November. Hopefully this will solve my issue.

andyblac, 2020-11-08:
pre-orders are open, got 2 on order, but del time has slipped to 1st dec.

jobl1978, 2020-11-08:
I can’t preorder it, because HomePods are not sold in Austria from Apple. I have to buy one in Germany (Munic). But anyway, I’ve tried it now with an iPad as hub and I can’t believe it, but it works now perfectly!!! Thanks for your input!!!

rojer, 2020-11-08:
thank Andy! i'm only curious what's the difference with Hue, what is it about it that makes ot work fine with tv while shelly doesn't? @jobl1978 can you double-check again? i'll report this to ADK developers if this is the case.

andyblac, 2020-11-08:
> thank Andy! i'm only curious what's the difference with Hue, what is it about it that makes ot work fine with tv while shelly doesn't? @jobl1978 can you double-check again? i'll report this to ADK developers if this is the case.

IMO prob due the fact that Hue has a hardwired bridge that has been setup for some time, and the AppleTV has had time register it, AppleTV's are well known to be flakey to act as Hubs, they need time to settle down, after registering new devcies , scenes, automations, if he left it for a day or two, it would probably have start working.

jobl1978, 2020-11-08:
Im not sure if it was the iPad that solves this issue. I wasn‘t able to remove the Apple TV as an Hub. Apple TV as Hub was deactivated but in Home App is was always shown as connected and the iPad shown in Standby. The only way was to delete my Home and create a new Home and reconnect the two shellys to it. I will now try to delete my Home again, activate the Apple TV and reconnect my shellys.

andyblac, 2020-11-08:
> Im not sure if it was the iPad that solves this issue. I wasn‘t able to remove the Apple TV as an Hub. Apple TV as Hub was deactivated but in Home App is was always shown as connected and the iPad [...]

i have 3 AppleTV's, i have homekit disabled on all of them, as if i enable it, it screws up my automations. I have 2 HomePod's as Hub, with 2 HomePod mini's on order. HomeKit also needs a minimum of 5 minuets to detect Hub changes, ie if you power one off it needs 5 mins at minimum to detect it is off, and switch to Standby. Just keep that in mind.

andyblac, 2020-11-08:
@rojer i have added a few tips to the WiKi https://github.com/mongoose-os-apps/shelly-homekit/wiki/Trouble-Shooting#issues-with-automations

pivale, 2020-11-09:
I'm using an Apple TV as a hub and it seems to be working fine except for presence automations (when someone enters or leaves home...)

andyblac, 2020-11-22:
@jobl1978 did you sort your issue ? still require help, or i close this ticket ?

jobl1978, 2020-11-23:
> @jobl1978 did you sort your issue ? still require help, or i close this ticket ?

Everything is working fine now. You can close the issue.

closed 2020-11-23 by andyblac