mongoose-os-apps/shelly-homekit #261: A big PROBLEM!!!

Nikola1236, 2020-11-08:
Good morning I am writing to you from Europe and I had some problems with the program that you created all 7 as a team. 
First of all as soon as I configured my shelly 2.5 with Apple HomeKit the same devices that I had in the app on purpose shelly said offline device BUT on the HomeKit app were working and now we would like to go back as before only with the app Shelly because we noticed that we are not comfortable with Apple HomeKit so here is the question
How can I go back to the beginning without Apple HomeKit but only with Shelly?
I look forward to an answer from you with affection _Nikola1236_

andyblac, 2020-11-08:
open the device in a web browser, scroll to the bottom of the page, click on 'Revert' button.

https://github.com/mongoose-os-apps/shelly-homekit/wiki/Device-webUI
https://github.com/mongoose-os-apps/shelly-homekit/wiki/System-Settings

Nikola1236, 2020-11-08:
it's okay?

it's okay?

Nikola1236, 2020-11-08:
http://www.mediafire.com/file/59zl1vbcckvvwya/2020-11-08_20-57-48.mkv/file
In this way?

Nikola1236, 2020-11-08:
> open the device IP in a web browser, scroll to the bottom of the page, click on 'Revert' button.
> 
> https://github.com/mongoose-os-apps/shelly-homekit/wiki/Device-webUI
> https://github.com/mongoose-os-apps/shelly-homekit/wiki/System-Settings

http://www.mediafire.com/file/59zl1vbcckvvwya/2020-11-08_20-57-48.mkv/file
In this way?
andyblac

andyblac, 2020-11-08:
yes, now just refresh page. it should be back on stock firmware.

Nikola1236, 2020-11-08:
> yes, now just refresh page. it should be back on stock firmware.

Not working _SAD_

andyblac, 2020-11-08:
what you mean not working ?

Nikola1236, 2020-11-08:
> what you mean not working ?

the revert button doesn't work and I can't see anything on shelly

andyblac, 2020-11-08:
so when you reload the webpage it is still on homekit firmware ?

Nikola1236, 2020-11-09:
> so when you reload the webpage it is still on homekit firmware ?

exact and now the only way I have is to change all shelly 2.5 / 1

andyblac, 2020-11-09:
are you Mac OS or windows ?

andyblac, 2020-11-09:
does the device have internet access ?

Nikola1236, 2020-11-09:
> are you Mac OS or windows ?

Windows

Nikola1236, 2020-11-09:
> does the device have internet access ?

Yes have internet access

andyblac, 2020-11-09:
i'm not sure why it did not work, just tested here and reverted fine. @rojer any suguesstions ?

rojer, 2020-11-09:
nope. @Nikola1236 when you power the device off and on, do you see and LED activity?

Nikola1236, 2020-11-09:
> nope. @Nikola1236 when you power the device off and on, do you see and LED activity?

Yes i see the led that turns on and off a certain number of times now I don't remember many times but it seems to me twice that it turns on and off

rojer, 2020-11-09:
ok, so it's at least somewhat alive. if the LED stays off after that, it means it connected to wifi successfully. do you know its ip address? you could check list of active clients on your wireless router

andyblac, 2020-11-12:
if i understood him correctly, he says it stuck on homekit fw, and can still access the webUI.

andyblac, 2020-11-16:
@Nikola1236 still a problem ?

andyblac, 2020-11-22:
@Nikola1236 do you still need help? else i'll close this ticket.

andyblac, 2020-11-24:
closing as no reponce from OP.

closed 2020-11-24 by andyblac