mongoose-os-apps/shelly-homekit #262: Flashing from 2.5.0 beta back to 2.4.0 failed

jobl1978, 2020-11-08:
My Shelly isn’t reachable after flashing back now from LAN. Is there any possibility to bring it to AP mode? Turn Power off/on and switching 5 times to hard reset the Shelly doesn’t work anymore.

andyblac, 2020-11-08:
try this https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery

andyblac, 2020-11-08:
i presume this is the shelly 2.5 ?, if so it has a button back of device press and hold the button for 10 seconds while the device is on

jobl1978, 2020-11-08:
No it’s a shelly1

andyblac, 2020-11-08:
Then try the on / off 10 times, it sounds like config in 2.5 changed and maybe incompatible with 2.4 @rojer ?

jobl1978, 2020-11-08:
I got it back it changed the IP Adress and was not reachable from the Home App anymore. After resetting the HomeKit settings and using a new Code it was able to reconnect.

andyblac, 2020-11-09:
can i close this now ?

jobl1978, 2020-11-09:
> can i close this now ?

Yes

closed 2020-11-09 by andyblac