mongoose-os-apps/shelly-homekit #265: Temp Sensor for HomeKit?

Rappelsack, 2020-11-10:
I am not sure if I should place it here:

There is a temp indication an all my shelly. Also on my shelly plug s.
Is it possible to integrate this as a temp sensor for homekit with a future update?

That would be awesome and with a shelly plug s perfect.

rojer, 2020-11-10:
yes, most shellies have a built-in temperature sensor, and it's used for overheating protection. it's not useful as ambient temperature sensor though, because shelly itself heats up quite a lot.

Rappelsack, 2020-11-10:
But the readout of the Plug S ist similar to my measured room temp. The under wall shellys are very hot.

rojer, 2020-11-10:
meh. i don't know, doesn't seem very useful. maybe, some day.

Ludelag, 2020-11-11:
Would be better to code a firmware for the Shelly H&T and use as Thermostat triggering the Shelly 1 for example connected to a heating system.

Rappelsack, 2020-11-11:
Thats right. But I have a Plug S with a decoration lamp which is just used at the weekend for approx 2 hrs. The rest of the week the shelly does not activate but it could be used as a temp sensor.

RedAnon, 2020-12-07:
> Thats right. But I have a Plug S with a decoration lamp which is just used at the weekend for approx 2 hrs. The rest of the week the shelly does not activate but it could be used as a temp sensor.

Dunno, I've just tested the 2.5 and after a minute or so it read 51°C with no load on it. Seems pretty useless to me.

> Any news?

I hope the devs got some free time over Christmas so even we could get some presents. 😅 Dimming and a proper heat sensor to keep us warm and comfy, you know.... 😅

rojer, 2021-01-05:
same as #384

closed 2021-01-05 by rojer