mongoose-os-apps/shelly-homekit #266: Revert after flashing wrong firmware

SebReuter, 2020-11-10:
Hi,

I accidentally flashed a shelly25 (original firmware) with the shelly1 (shelly-homekit) firmware. 

Now only one switch is available (but this one works pretty well :))
The device can be accessed and configured via its IP-addredd

I would like to flash the shelly25 but the following things do not work:
1) "Revert to stock"-functionality flashes the original shelly1 firmware 
2) "Update from file"-functionality crashes with "-13 Aborted by event handler"
3) http://IP/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip results in "Not found"

It would be great if flashing of the wrong firmware would be blocked. 

Do you have any suggestions what I should try to flash the shelly25 firmware over the air?

andyblac, 2020-11-10:
download the shelly2.5 firmware and upload it via the webUI.

here https://github.com/mongoose-os-apps/shelly-homekit/releases
and here
https://github.com/mongoose-os-apps/shelly-homekit/wiki/System-Settings

SebReuter, 2020-11-10:
It fails with
shelly_main.cpp:692     Wrong app name 'switch25'

andyblac, 2020-11-10:
looks like only option then is reflash via serial. unless @rojer has any ideas ?

rojer, 2020-11-10:
i'm surprised it was even possible to flash from stock with the wrong app name.
but anyhow now that we're here... it's possible to recover, i'll build you special firmware to flash.

SebReuter, 2020-11-10:
wow thank you. 

I was just thinking to disable the check in OTABeginCB(int ev, void *ev_data, void *userdata). Build Shelly1, flash it and then flash shelly25 over it.

What is your approach?

rojer, 2020-11-10:
yep, exactly that. if you are comfortable with building firmware yourself, please go ahead.

SebReuter, 2020-11-10:
will give it a try.

SebReuter, 2020-11-10:
Worked like a charm. Thank you guys for your help.

This project is awesome - and so is your work. Thank you!

rojer, 2020-11-10:
cheers!

closed 2020-11-10 by rojer