mongoose-os-apps/shelly-homekit #269: Better log system for HomeKit debugging.

Tomer51m, 2020-11-10:
Two insights I have from the past few nights pf debugging problems with HomeKit.  

Perhaps a time stamp would be useful?
Maybe have two types of logs?
User oriented log in which he can see time stamps with state changes, wifi signal strengh, ip connections and other user related data. This way can record data for longer periods of time (?)
Technical log with system information ram, and other "important" 😆 data for development and bug fixes.
Another approach can be a continues recording of the user related data (something like "recent history") and have a "debugging log" button for the technical stuff. Because many times you want to check if things are working as you expect, but turning a log every time you want to check is a bit weird. It was much nicer to just check back on the history of the device as it was running.

rojer, 2020-11-10:
when logging over UDP or http, first column is the timestamp - uptime in microseconds.
when logging over serial, `mos console` adds timestamp to each line, you can use that.

Tomer51m, 2020-11-11:
Unfortunately I’m not sure ho to that. I can work with milliseconds, I just didn’t know that was it. 
What do you think about my suggestion? is it possible?

jsiegenthaler, 2020-11-11:
I do agree that formatting the milliseconds into human-readable ISO date would be much easier for everyone to interpret.
YYYY-MM-DD_hh:mm:ss.millisecond

pivale, 2020-11-12:
I opened this issue #236 on the same topic.
I think we could benefit from a enhanced log experience to help reporting bugs more easily and more accurately

andyblac, 2020-11-22:
I'll close this as it is a duplicate.

closed 2020-11-22 by andyblac