mongoose-os-apps/shelly-homekit #271: flash_shelly.py fails to install zeroconf

patricks, 2020-11-11:
Hi,

I have used the `flash_shelly.py` script to update my shelly's the last few versions. Today I updated my local git repository to get the latest version of the script an now I get the following error, if I try to run the script:

```bash
 # ./flash_shelly.py  
Installing zeroconf...
Traceback (most recent call last):
  File "./flash_shelly.py", line 87, in <module>
    import zeroconf
ModuleNotFoundError: No module named 'zeroconf'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./flash_shelly.py", line 91, in <module>
    import zeroconf
ModuleNotFoundError: No module named 'zeroconf'
```

I am on macOS 11.0.1-RC2 (Big Sur). I already used the script with Big Sur beta versions.

Any ideas how I could fix this problem?

andyblac, 2020-11-11:
looks like you need to install zerocofn again, the script should do it for you, but looks like something chnaged in BigSur, try

`pip3.install zeroconf`

patricks, 2020-11-11:
That fixed my problems, thanks for you help.

closed 2020-11-11 by patricks