mongoose-os-apps/shelly-homekit #272: HomeKit over Thread

andyblac, 2020-11-12:
As the HomePod mini will support this, is this something that can be added to this firmware, as it might help people that are using mesh networks.

https://www.threadgroup.org/What-is-Thread/Thread-Benefits

rojer, 2020-11-12:
it's a long way off as there's currently no support for Thread in mos.

andyblac, 2020-11-12:
> it's a long way off as there's currently no support for Thread in mos.

ok, i'll leave ticket open, for future reference.

NotClear, 2020-11-22:
Theoretically thread runs on the ESP8266, there is no hardware limitation. However, the thread requires an IEEE 802.15.4 radio which the ESP does not have, so it would need to be upgraded.

The Openthread software is already available for the ESP32 but theoretically it would run on the ESP8266 so it only needs a migration of the software.

https://github.com/openthread/ot-esp32

MBarti, 2021-03-10:
Thread is incredible. It solves all the problems of Zigbee and works extremely well, according to tests and my own experience (With Nanoleaf A19 bulbs). Players like Nanoleaf and Eve Systems will move to 100% Thread and everyone else will follow. The open source libraries are there for the taking.
 
It's becoming clear that EVERY major producer will move to Thread within a few years, because it solves all the problems with Zigbee and all the Big Boys support it, as a spiritual successor to Zigbee, which lacked interoperability and required bridges. 
Since Homepod Mini seems to be a big hit, the demand is there. 

Thread itself supporters: (+Apple)
[image]

CHIP project supporters (Thread will be one of the main networking layers, NOT Zigbee):
[image]

timoschilling, 2021-04-14:
@rojer and @andyblac what do you think? I think this thread can be closed for multiple reasons:
1. Thread can't run on the current Shelly hardware, so the request is out of scope for this project
2. If a future hardware would be possible to run Thread aka supports IEEE 802.15.4, I would expect Shelly would already support Thread out of the box

For me this request is a feature request to run this firmware on devices which aren't exist right now, like I would request for camera support and Shelly didn't have a camera product. (I know they had one in the past, it's just an example)

andyblac, 2021-04-14:
fine by me.

rojer, 2021-04-21:
yes, closing

closed 2021-04-21 by rojer

gmuth, 2026-03-14:
I think with Gen4 the hardware (ESP-Shelly-C68F) now supports OpenThread:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32c6/api-guides/openthread.html
Could you reevaluate possible solutions?