mongoose-os-apps/shelly-homekit #274: Problem. Please help

moonloopsun, 2020-11-13:
Hi guys. Great project. 
I installed the Shelly 2.5 for my Rolls. 
They worked fine before. 
When i try to calibrate the motor is getting into a loop of toggling up down every second. And the wall switch doesn’t work anymore. Than when i press open or close the rolls open only a few centimeters. 
Please help. We can’t open the rolls now or close it...   
the open / close time should be 35 seconds
Each direction..
thanks

pivale, 2020-11-13:
I had a similar problem (only descended a few centimeters each time) after an upgrade. I suspect something went wrong.

I suggest:
1. activate logs in webui, test again calibrate and try to function. Collect logs and send here please.
2. reboot and redo step 1 and check if it solved.
3. revert firmware back to stock and reinstall again latest version of shelly-homekit (this solved for me)

SilvioMigu, 2020-12-04:
Hi
I wanted to try to put a 2.5 shelly on the Rolls, but I still haven't seen a connection and configuration scheme.
Do any of you have a scheme?

Draf76, 2020-12-24:
[image]

SilvioMigu, 2020-12-24:
Thank you for the answer. 
I've already ventured to install and it's that way it works, thank you[image]

torto-85, 2021-01-08:
> Hi guys. Great project.
> I installed the Shelly 2.5 for my Rolls.
> They worked fine before.
> When i try to calibrate the motor is getting into a loop of toggling up down every second. And the wall [...]

Hey,
did you solved the problem? I have the same too!
Cheers, Torto

andyblac, 2021-01-25:
@pivale can you offer any assistance ?

pivale, 2021-01-25:
> I had a similar problem (only descended a few centimeters each time) after an upgrade. I suspect something went wrong with the version upgrade.
> 
> I suggest:
> 
>     1. activate logs in webui, test again calibrate and try to function. Collect logs and send here please.
> 
>     2. reboot and redo step 1 and check if it solved.
> 
>     3. revert firmware back to stock and reinstall again latest version of shelly-homekit (this solved for me)

@torto-85 , have you tried these steps?
I had a similar problem and after revert to stock and reinstall homekit I got it to back to work

andyblac, 2021-03-03:
still need help ?

torto-85, 2021-03-03:
Hi, thanks for your suggestion (sorry @pivale: I didn't see your comment yet). I follow #404. Manual calibration seems a solution for me. Because I use one shelly to control more than one roller shutter the auto-calibration won't work for me.

andyblac, 2021-03-03:
so i can close this ?, can always reopen later is thing change.

torto-85, 2021-03-09:
> so i can close this ?, can always reopen later is thing change.

Yes, that‘s fine for me.

closed 2021-03-10 by andyblac