mongoose-os-apps/shelly-homekit #275: Shelly 2.5 as a screen protector instead of roller shutter

leilok, 2020-11-15:
Hi there!

I’ve been around this community some months ago, reading and learning.. so first of all I will like to say thanks to Al the people are working on making Shelly devices usable with HomeKit, especially to @rojer.

I arrived here trying because of the Shelly 2.5 and his capability of act like a roller shutter... and honestly the improvement has been enormous!!!

I just installed a motorised projector screen, and installed a Shelly 2.5 as a roller shutter... and the  “problem” is that the operation of both of them are opposite... I mean...

A roller shutter, up position is “Open” position... and a screen projector, up position is “Close” position... so right know it’s a little bit confused the situation, because when the screen is closed (up position) In the Home app the screen seems to be open.

I think, from the ignorance side, that this is not difficult to implement...

Thanks!!!

timoschilling, 2020-11-16:
I think you simply need to switch the wiring of your Shelly, replace SW1 with SW2 and SW2 with SW1, then your status matches your needs.

leilok, 2020-11-16:
> I think you simply need to switch the wiring of your Shelly, replace SW1 with SW2 and SW2 with SW1, then your status matches your needs.

I am not sure.... because I can close the screen saying to Siri (in spanish, sorry if is not the correct form in english) "go up", and open the screen saying (go down)... If I invert the wiring, I will invert this too... 

I was looking at the wiki, because what you are suggesting, is the same as the "Swap output" option...

Maybe I am wrong ¿?

Draf76, 2020-12-22:
How did you manage to calibrate the screen?

leilok, 2020-12-29:
> How did you manage to calibrate the screen?

leilok, 2020-12-29:
Sorry for the late reply...

I think it depends on the motor you are using... 

I mean... the motor use to have two manual limit switch. One to adjust the "up" position, and the other one to adjust the "Down" position... I think all the motors for a roller shutter has those 2 limit switch.

When I installed the screen, the first think I made, before install it was adjust the "Down" position (Screen open, ready to use). My screen came with manual remote with 2 keys (up, down and stop), and with remote control. Y "Cut" the manual remote an wired the motor directly to my integrated-wall switch, and put the shelly between that switch and the motor.

And everything went ok. Installed HomeKit firmware, and follow the wiki instructions... and calibrate the shelly, to learn when the screen is open and closed.

Read your screen manual, because I thing there must be the solution to your problem... because all the motors should be calibrated before installing the shelly.

timoschilling, 2022-03-17:
I reread this request and I think it's not possible, let me explain:

100% means open / up
0% means closed / down

What you want if I understand you right is

100% means closed / up
0% means open / down

That requirement make sense from your point of view. But we can't do that, because HomeKit didn't support open/close or up/down commands. It only knows a percentage value between 0-100%. If you say "Siri close" it makes it to 0% and send that to the shelly. So we are not able to change that. Thats why I close this request.

closed 2022-03-17 by timoschilling

andyblac, 2022-03-17:
@timoschilling just thinking gout loud, but would it not be possible to add an "Invert" option to UI so that 0% = 100% and vice versa in code ?

timoschilling, 2022-03-17:
That's what swap output already does.

timoschilling, 2022-03-17:
From my point of view his request will only work when HK knows commands like open/close and not only x%