mongoose-os-apps/shelly-homekit #276: Reset: How to recover to an working state with HomeKit and to make it assessable with IP adress?

mattze66, 2020-11-16:
Hello Programmers,
you're great. 5 pieces are running as HomeKit participants

Now I had to reset an paired shelly 1 (10x on/off). 
As result the shelly 1 device became visible as WiFi device only but not any longer as HomeKit device.
The App Store Shelly app does not recognize this device as I assume it is still flashed as HomeKit device.
HomeKit does not recognize it since it is not known to my wlan network.

Please share with me the procedure to get it back into HomeKit or as IP adress participant?

Thanks a lot
Mathias

andyblac, 2020-11-16:
you will have to resetup the device from the start, as you did originaly https://github.com/mongoose-os-apps/shelly-homekit/wiki/Device-webUI

andyblac, 2020-11-16:
oh forgot to say, remove it from homekit 1st before resetup a nd trying to re-add it to homekit.

(long press on tile, scroll tp bottom tap on remove)

rojer, 2020-11-16:
actually, you don't need to remove from homekit, if it rejoins wifi network it should be back.

connect to the wifi AP that shelly set up, go to http://192.168.33.1/ and configure wifi

andyblac, 2020-11-16:
> actually, you don't need to remove from homekit, if it rejoins wifi network it should be back.
> 
> connect to the wifi AP that shelly set up, go to http://192.168.33.1/ and configure wifi

so a hardreset does not reset the homekit code ? only the WiFi settings ?

rojer, 2020-11-16:
correct

andyblac, 2020-11-16:
> correct

ok, good to know. is there a way to fully reset the device ? is this something we should add ?

rojer, 2020-11-16:
there isn't a way now, i have the following in mind:
first external reset via on-off toggles -> wifi reset, second -> HAP reset as well.
long press of physical button - same thing, wifi then HAP.
boot with physical button pressed -> recovery mode, full config reset + start AP, to allow reflashing.

andyblac, 2020-11-16:
> there isn't a way now, i have the following in mind:
> first external reset via on-off toggles -> wifi reset, second -> HAP reset as well.
> long press of physical button - same thing, wifi then HAP.
> boot with physical button pressed -> recovery mode, full config reset + start AP, to allow reflashing.

> long press of physical button - same thing, wifi then HAP.
only within the 1st 60 seconds, as we use long press to bypass timer.

rojer, 2020-11-16:
by physical button i mean the tiny button on the device itself (all shellies have them, except shelly1). it already works as (wifi only) reset today.

andyblac, 2020-11-16:
> by physical button i mean the tiny button on the device itself (all shellies have them, except shelly1). it already works as (wifi only) reset today.

oh, ye forgot about those. noted thanks.

mattze66, 2020-11-16:
Hey guys,
thanks. It worked. I did not know the standard IP adress http://192.168.33.1/ before you told me. That was the missing piece
You might want to add this piece of information to the reset procedure description. 

Everythings fine now. I crashed two other devices as I was getting into: Somewhere between firmware upgrade and before homekit. Mostlikely I can bring them both back to life with the reset procedure and standard IP adress.

Thanks a lot
Mathias

closed 2020-11-16 by mattze66

andyblac, 2020-11-16:
i'll link the setup wiki to reset procedure thanks.