mongoose-os-apps/shelly-homekit #277: Shelly 1 garage notifications

SilvioMigu, 2020-11-16:
Good afternoon
From now on, I offer my congratulations for the work developed here.
I have shelly 1 working in the garage, with a magnetic sensor, to indicate that it is closed.
When I close the gate, I get two notifications "the garage was open" and "the garage was close", shouldn't I get a notification?
When I open it, only a correct notification appears.
I have the right "move team" with the 28-second garage.
Any ideas, what can it be?
Thanks
[image]

SilvioMigu, 2020-11-16:
I think that the "moviment time" when it is closing the garage, notifies open, and immediately afterwards the magnetic sensor, notifies that it is closed.
I don't know if they use active notifications, because if I turn off notifications, it works well. I am simply not notified if it was opened or closed.

rojer, 2020-11-17:
can you enable debugging in the web UI, perform movement and post the log here? it should help clarify what events are seen by firmware.

SilvioMigu, 2020-11-17:
Last night at home, I moved the magnetic sensor a little bit.
Now, it does not report any notifications when it closes.
But it shows in the home app, that it is closed.
I made a video to show
https://youtu.be/dbVkAppdxJg

SilvioMigu, 2020-11-17:
Right now as I am at work, I cannot show the Log

However, here on the phone, I tried to open and stop. And then to close again, to my surprise, I receive the notification that the garage has been closed.
Conclusion, if I open everything and close it again, I don't get notification, if I just open half and close it again, I get notification
https://youtu.be/nMr6wCYL6Bo

SilvioMigu, 2020-11-17:
@rojer , your work is incredible, it is working well, as I always wanted.
It's just a matter of fine-tuning notifications 😉

jsiegenthaler, 2020-11-17:
Just remember that if your iPhone goes into DoNotDisturb at night time then notifications will not be shown. That's caught me out a few times...

SilvioMigu, 2020-11-18:
Yes, it has happened to me with other situations, as I have active DoNotDisturb at night.
But in this case it is even during the day that I was not being notified.

SilvioMigu, 2020-11-18:
I changed the state and saved it, and changed it back to normal close, and it went back to notify when it closes.

It seems that he needed to be stirred and refreshed, and he notified again 🙂
[image]

closed 2020-12-04 by SilvioMigu