mongoose-os-apps/shelly-homekit #278: Activation switch mode

lucaspinelli85, 2020-11-16:
Hi, in original firmware there is a "activation switch" mode, that use for motion sensor when any inputs turns "ON" and resets auto off times (instead, in normal mode, when auto off is running, the input turn "OFF" and stop auto off times).
Could you add it in HomeKit firmware?
In my case instead I need also the auto off timer start when input turn "OFF" (not start when input turn "ON"), pratically when toggle is ON, switch is ON, but when toggle go OFF the switch KEEP ON for an auto off timer, after tath it go OFF. It remain ON after toggle off.
Could you add both version? "Activation switch"(input reset timer) and "Keep on" (remain ON after toggle off).
Thanks!

lucaspinelli85, 2020-11-16:
[image]

lucaspinelli85, 2020-11-16:
[image]

rojer, 2020-11-17:
thank you for the detailed description!
this makes sense and is easy to do.

lucaspinelli85, 2020-11-17:
Thanks for helpful and hope it went add soon both the option!

lucaspinelli85, 2020-11-18:
> thank you for the detailed description!
> this makes sense and is easy to do.

When do you think it could be add in the firmware?

closed 2020-11-25 by rojer

rojer, 2020-11-25:
please help test 2.6.0-beta - https://github.com/mongoose-os-apps/shelly-homekit/issues/291